OctoPrint/OctoPrint #3804: [Request] Better handling of no `users.yaml` file

cp2004, 2020-11-10:
**Is your feature request related to a problem? Please describe.**
As Access Control is mandatory, OctoPrint requires a `users.yaml` file to be present so someone can log in. Not a problem on first run, since the wizard sets this up for you. However, should this file go missing (or someone fiddles with it) it means you cannot log in at all.

**Describe the solution you'd like**
Some kind of prompt that there is no users file. Not sure what would be the most secure, since a popup to 'make a new user' would be kind of insecure. Maybe it would not be much of an issue, but should be considered.

**Describe alternatives you've considered**
Getting upset when I break things, crying and then realising it was a mistake because I put the config.yaml but not the users.yaml file in.

**Additional context**
If this is the only issue I can think of, this must be a very good release candidate!

foosel, 2020-11-12:
> However, should this file go missing (or someone fiddles with it) it means you cannot log in at all.

Hm... the way things currently are implemented, it will actually already make the access control wizard pop up if the file is missing, as then the user manager will not be marked as having been customized and the wizard will signal it's needed - just tested against 1.5.0rc1 and that works as expected.

How did you break things so that that didn't work? I might have to add some additional logic to detect corrupted user storage.

foosel, 2020-11-12:
I think I know what happened. Missing file got handled correctly, but broken file (e.g. no user dict inside) didn't. Fixed now and ready for 1.5.0rc2

cp2004, 2020-11-12:
That *could* have been it. Came across this during my ranting at wizards in general a couple of days ago - it is likely related since I was managing to mess with configs - this broke first and I thought it was probably best it was handled since AC is mandatory. 👍

foosel, 2020-11-17:
1.5.0rc2 is out

closed 2020-11-17 by foosel