OctoPrint/OctoPrint #3805: [BUG]: 1.5.0

jneilliii, 2020-11-10:
#### What were you doing?

Testing 1.5.0rc1...when selecting files in the file list an api error is occurring, causing issues with PrusaSlicerThumbnails.

1. Install PrusaSlicer Thumbnails
2. After restart/reload, upload a file that includes a thumbnail (attached for convenience)
3. In plugin's settings enable `Show thumbnail in state panel.`
4. Select a file, which runs js `OctoPrint.files.get('local',data)` as part of a knockout subscribe binding `self.printerStateViewModel.filepath.subscribe(function(data)`
5. error generated in logs (attached) and metadata is not loaded to get thumbnail property.

[LampCylinder.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/5520712/LampCylinder.gcode.txt)

#### What did you expect to happen?

File data to be retrieved and thumbnail to show in state panel.

#### What happened instead?

Error and no thumbnail.

#### Did the same happen when running OctoPrint in safe mode?

Not entirely sure, but can test shortly. 

#### Version of OctoPrint

1.5.0rc1

#### Operating System running OctoPrint

any, tested in both windows dev environment and octopi 0.18.

#### Printer model & used firmware incl. version

virtual printer

#### Browser and version of browser, operating system running browser

Google Chrome Version 86.0.4240.193 (Official Build) (64-bit) on windows 10.

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5520695/octoprint.log)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

[developer_console.log](https://github.com/OctoPrint/OctoPrint/files/5520702/developer_console.log)

#### Screenshot(s)/video(s) showing the problem:

[image]

thumbnail should be just below the first `<hr>`.

I have read the FAQ.

jneilliii, 2020-11-10:
I can confirm, simply running `OctoPrint.files.get('local','LampCylinder.gcode')` with all plugins disabled causes the same error.

foosel, 2020-11-11:
I fear that got introduced by the more aggressive caching on the file api down into the file manager that I now do. Have to investigate. A `dict` check doesn't appear to be the right way, as `children` now seem to be a list in general.

foosel, 2020-11-11:
Fixed by the above commit, ready for 1.5.0rc2

foosel, 2020-11-17:
1.5.0rc2 is out

commit bed0f8d1 (2020-11-11): 🐛 Children is a dict, not a list

Closes #3805
closed 2020-11-17 by foosel