OctoPrint/OctoPrint #3810: [1.5.0 RC1] Action Command not working

schnello, 2020-11-11:
Hello.

1.5.0 installed at a Pi.

First:
Looks like there is a new default action command:
action:start ... starts the print
[image]

This information is missing here:
[image]

But the main poblem is that my cumston commands are not working.

[image]

[image]

cp2004, 2020-11-11:
Please can you fill in the full issue template in the contribution guidelines - there are important questions to do with safe mode, OS etc. It helps analyse issues far faster if all of the requested information is present. You will notice that that is also what the bot's asking, not just the log.

cp2004, 2020-11-11:
Actually, just looked at the partial log you uploaded. This is an issue with the action command plugin, since it does not include any `*args` or `**kwargs` in the hook. Reach out to the author of the plugin with the issue please, there is nothing we can do here:
https://github.com/benlye/OctoPrint-ActionCommandsPlugin

closed 2020-11-11 by cp2004