OctoPrint/OctoPrint #3814: [1.5.0rc1] Error on translation upload

jneilliii, 2020-11-14:
#### What were you doing?

Uploading a translation file, attached.

[OctoPrint.1.5.0rc1.-.Pack.Langue.FR.zip](https://github.com/OctoPrint/OctoPrint/files/5540089/OctoPrint.1.5.0rc1.-.Pack.Langue.FR.zip)

#### What did you expect to happen?

File to upload and translation to be available.

#### What happened instead?

Nothing

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.5.0rc1

#### Operating System running OctoPrint

octopi 0.18 nightly

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 86.0.4240.198 (Official Build) (64-bit) on Windows 10

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5540091/octoprint.log)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

Just a 500 Server error warning.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-11-14:
Analysed, Python 2/3 issue - in Py2, `filter()` returned a list. In Py3, it returns an iterable. The problem:
https://github.com/OctoPrint/OctoPrint/blob/b7a45b79bf86339f0b28177ac1eaebb56d00ea37/src/octoprint/server/api/languages.py#L143-L149
In https://github.com/tjjfvi/OctoPrint-IFTTT/pull/35 I just converted it to a list again, will test that this works in this case and prepare a PR.

cp2004, 2020-11-14:
Interestingly, I don't believe this is a regression. See my comments in #3815, the history of this line was confusing. I'm leaving it up to @foosel as to whether this should be included in 1.5.0 now it has started rc.

foosel, 2020-11-16:
Bug yes, regression no, but I still included it in 1.5.0 as the change is minimal enough and the fix valuable enough. Thanks @cp2004!

foosel, 2020-11-17:
1.5.0rc2 is out

PR #3815 (2020-11-16): Fix uploading a language pack
In Python 3, `filter()` returns an iterable rather than a list. Whilst this is better for performance, our list is not long and it means we can call `len()` on the list.

Tested using reproduction steps in #3814 and verified that language packs can now be installed properly.

Closes #3814

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [x] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talki [... cut]
commit ab5d19bf (2020-11-16): :bug: Fix uploading a language pack

In Python 3, `filter()` returns an iterable rather than a list. Whilst this is better for performance, our list is not long and it means we can call `len()` on the list.

Tested using reproduction steps in #3814 and verified that language packs can now be installed properly.

Closes #3814
closed 2020-11-17 by foosel