OctoPrint/OctoPrint #3819: [Request] file search bar recognise space as _

MadeforMaking, 2020-11-16:
#### What were you doing?

Trying to search for gcode to print that was successful, I have hide successful prints enabled
Rather than manually type _ I'd like the search algorithm to automatically change a space to _

#### What did you expect to happen?
The queried file(s) to be displayed in the files section as a space could mean an underscore. Some filenames have underscores removed but if you search for them the way they are displayed they will not appear as you need an underscore.

#### What happened instead?
File(s) where not displayed as they are spaced with underscores "_"

See .gif at end

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
Windows : 1909
Octopi : 0.15.1

#### Printer model & used firmware incl. version
**2 instances of octopi on 1 raspberry pi**
Prusa i3 MK2.5s : 3.9.0-3421
Prusa i3 MK3 MMU2s : 3.9.0-3421

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
[Here](https://pastebin.com/W9kMC0F5)

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2020-11-16:
This came a bit too late for inclusion in 1.5.0, but the above commit will be part of 1.6.0.

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel