OctoPrint/OctoPrint #3820: File Operations freezes (deleting file, moving file, creating directory)

empakoso, 2020-11-17:
#### What were you doing?
I was deleting a print from the files list.
The delete button was indicating an endless 'progress wheel'
had to stop and restart the daemon from ssh to allow the delete to complete
restarting OctoPrint did not help at first before trying the above stop/restart.
same hangup also presented itself when a new folder was attempted to be created and file tried to move to another folder.
[2020-11-16 OctoPrint Logs - FO freezing.zip](https://github.com/OctoPrint/OctoPrint/files/5550742/2020-11-16.OctoPrint.Logs.-.FO.freezing.zip)

#### What did you expect to happen?
delete the file immediately and not hangup on aprogress wheel. 

#### What happened instead?
nothing was working not even the systemd service I created in a python script which I've been running alongside the PSU plugin for the last couple years.

#### Did the same happen when running OctoPrint in safe mode?
safe mode was not attempted.

#### Version of OctoPrint
OctoPrint 1.4.2 Python 3.7.3 OctoPi 0.17.0
Just updated/upgraded Raspbian yesterday

#### Operating System running OctoPrint
OctoPi Linux

#### Printer model & used firmware incl. version
Creality Ender3 Pro

#### Browser and version of browser, operating system running browser
Win10 lastest version with Chrome browser

#### Link to octoprint.log
included (attached)

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2020-11-17:
It would be a good idea to re-attempt this running OctoPrint in safe mode to verify the issue is not plugin related.

empakoso, 2020-11-17:
Will attempt later today.  Thank you for the advice.  Did not realize what safe-mode did.

empakoso, 2020-11-17:
OK, in safe mode everything deletes beautifully and fast.  I will attempt to uninstall plugins now.

closed 2020-11-17 by empakoso