OctoPrint/OctoPrint #3821: [1.5.0rc1] Path to FFMPEG: /usr/bin/avconf missing

JohnOCFII, 2020-11-17:
#### What were you doing?

I started with a new installation of OctoPi 0.18.0 (the RC).  I upgraded the default 1.4.2 OctoPrint to 1.5.0rc1.  I then performed a restore of OctoPrint 1.5.0rc1 from my previous installation on an older nightly build of OctoPi 0.18.

As I reviewed configuration settings, I noticed in `Settings --> Webcam & Timelapse` there is a setting for `Path to FFMPEG`. This field is pre-populated with `/usr/bin/avconv`. That item does not exist on this fresh installation.  

#### What did you expect to happen?

I expected either `avconv` to exist at the specified path, or that the field would be pre-populated with `/usr/bin/ffmpeg`

#### What happened instead?

Field needed to be manually updated to `/usr/bin/ffmpeg`

#### Did the same happen when running OctoPrint in safe mode?

n/a

#### Version of OctoPrint

OctoPrint 1.5.0rc1 

#### Operating System running OctoPrint

OctoPi 0.18.0 (RC)

#### Printer model & used firmware incl. version

Prusa MK3S w/MMU2S

#### Browser and version of browser, operating system running browser

Safari Version 14.0.1 (15610.2.11.51.10, 15610)
macOS Catalina 10.15.7

#### Link to octoprint.log

https://gist.github.com/JohnOCFII/af96d79e4768bb90f0dcf9e9f958b481

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a
[image]

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-11-17:
That's an Octo**Pi** issue (the ffmpeg path comes preconfigured on the image, not by OctoPrint).

foosel, 2020-11-17:
Created an issue on the OctoPi repo, please follow that.

closed 2020-11-17 by foosel

jneilliii, 2020-11-17:
@foosel shouldn't the restore overwrite the previously wrong value with the correct one?

cp2004, 2020-11-17:
I think in this case it may be a previous *incorrect* value has overwritten the correct one, since this is correct in OctoPi's codebase.

Not entirely sure how it would make it there without having been changed previously.

foosel, 2020-11-17:
> I think in this case it may be a previous _incorrect_ value has overwritten the correct one

This is precisely the issue here. The backup contains the value that was correct on the former OctoPi image, but is wrong on the new one. It overwrites the correct one on the new image. 

As pointed out in the OctoPi ticket, I think I'll need to think about allowing something like environment specific overlay files that OctoPi could then use in order to pre-configure this kind of stuff, without it making it into backups unless differing from the (overlaid) default. I'll have to think a bit about this - the settings implementation already allows this, the question is how to implement it specifically. Definitely not something for 1.5.0 though.