OctoPrint/OctoPrint #3823: [RC Feedback] Feedback on the 1.5.0rc2 Release Candidate

foosel, 2020-11-17:
Please provide *general* feedback on your experience with the 1.5.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

  * 🐛 #3699 M876 not being force-sent while printing

### Third party issues

  * 🐛 https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/75 PrettyGcode - [PR available](https://github.com/Kragrathea/OctoPrint-PrettyGCode/pull/76) - no reaction yet from maintainer @Kragrathea
  * 🐛 https://github.com/BillyBlaze/OctoPrint-FullScreen/issues/32 OctoPrint-Fullscreen

### Issues needing analysis

kmanley57, 2020-11-17:
Gina, Just to give feedback at least once or twice in 5 years. The rc2 seems to be working pretty well, but the initial resend notice was disconcerting the first couple times it pops up on server-printer startup.

Guilouz, 2020-11-17:
Another issue with language pack on this new RC.

Now I can upload a language pack but but it is not taken into consideration.

I have two raspberry pi 4,

Onthe first, I deleted old language pack (with problem with first RC), upload ok the new one on this new RC but language is not in the list of available languages

[image]

[image]

On the second pi, I left the old language pack and uploaded the new one over it and all is ok

[image]

[image]

I deleted the cookies and tested with Chrome and Safari, no changes.

varazir, 2020-11-17:
Don't know if it's plugin issue of RC issue, I have a plugin called Domoticz (0.1.3) and it just gives me a error 

Domoticz Error
Status unknown for 192.168.0.10:80 Double check IP Address\Hostname in Domoticz Settings.

It started with RC1, 
[plugin_domoticz_debug_20201117.log](https://github.com/OctoPrint/OctoPrint/files/5555138/plugin_domoticz_debug_20201117.log)
[octoprint_20201117.log](https://github.com/OctoPrint/OctoPrint/files/5555139/octoprint_20201117.log)

foosel, 2020-11-17:
@kmanley57 on rc2 it should actually only be popping up after at least 100 lines have been transmitted, so no longer on printer connection 🤔 Do you see it right after connecting? If so, please enable `serial.log`, reproduce and share that, I need to take a look.

@Guilouz Please open a full bug report.

@varazir Please report this on the plugin's tracker, I cannot debug this as I'm not familiar with the plugin's code and don't even know what this error means specifically.

cp2004, 2020-11-17:
Oops, wrong button sorry! 😢

foosel, 2020-11-17:
@cp2004 you just gave me a bit of a start 🤣

jneilliii, 2020-11-17:
oh Domoticz is one of mine, didn't add that one to the list before did I? I'll look into it and provide RC.

jneilliii, 2020-11-17:
Unable to reproduce domoticz issue @varazir please open issue on its github repo.

oliof, 2020-11-17:
Upgrade went fine on my installation. After the issue I reported in 1.5.0rc1, I removed the MattaCloud plugin and the Pi was rocksteady for the few days between the rc1 issue and updating. It still is up and happy. I will try again with the MattaCloud plugin and report a bug there if I can reproduce the instability issues when installing it.

Prutsium, 2020-11-17:
I have the feeling RC2 is a bit Sluggish as i use an Bed level Gcode with Pause Resume and this feels slower getting out of it's pause.
Will keep an eye on it and would take a look at the CPU loads at those moments.

Further .... is Octofarm still working with RC2? I have 2 Pi installs and one came immediate online and the other wouldnt come online. Cant reboot the Pi yet as it's printing right now.

jneilliii, 2020-11-18:
#3825 - Unable to install plugins in Windows with latest rc.

JohnOCFII, 2020-11-18:
All good so far:

- Upgraded (from 1.4.2) with no issues
- Uploaded new GCODE
- Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- The modified Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Telegram plugin worked well
- Firmware Updater worked well

Lantoit, 2020-11-18:
Hi @foosel 
Since now no problem with RC1 and RC2. The only strange behavior is installing Bed-Visualizer 0.1.15, the installation appear ok but the plugin doesn't appear!
In attach the plugin log.
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint/files/5558674/plugin_pluginmanager_console.log)
Leo

foosel, 2020-11-18:
@Lantoit that one has already been fixed by @jneilliii in the current 1.0.0 RC as far as I am aware.

@Prutsium at least on my test install of octofarm everything looks solid

foosel, 2020-11-18:
1.5.0rc3 is out to fix #3825, new feedback ticket is #3827

closed 2020-11-18 by foosel