OctoPrint/OctoPrint #3824: Systemd service file modification that allows easy port change[Request]

Eewec, 2020-11-17:
Spent a while trying figure out how to change the port binding with the systemd file. From my understanding it's either init or systemd services and not both? Anywho, a simple addition to the Execstart line allows you to set the port number:

ExecStart=/home/pi/OctoPrint/venv/bin/octoprint --port=5000

It would be easy enough to add the above in the default file which would mean the next person isn't hunting for an age on how to change it.

cp2004, 2020-11-19:
I *personally* don't think its smart to add this to the (only provided as example) service files, since it is just one more thing to change *in the event* that the default port needs to be changed - this would override that. Ideally, these files wouldn't be included in the codebase but maintained elsewhere I would imagine (but I've not been around here long enough to know the full history).

It's well documented in both the [configuration docs](https://docs.octoprint.org/en/master/configuration/config_yaml.html#server) and on the CLI itself (specifically `octoprint serve --help`) how to change the port number, and you could apply the argument 'why not set the basedir there' or 'why not set the host there' too, which would become unnecessary.

Eewec, 2020-11-20:
Huh, I now feel daft. I've actually looked at that document beginning of last month. No idea how I missed that. I'll have to look at why I didn't use the config.yaml file to set things... I have a feeling it didn't work due to the virtualenvwrapper being bugged to **** and back so ditched it trying to find something that did work and never looked back once I found out about (and got fixed) the bug with virtualenvwrapper.

Eewec, 2020-11-20:
OK, yeah config.yaml now works fine to set the port, so I'm going to assume (correctly or not) that when I tried that before one of the many other issues I had was causing it to be ignored (like it installing in the wrong place instead of inside the virtualenv in the first place. Seriously, v5 was a mess. Glad that's been taken off the mirror site.)

closed 2020-11-20 by Eewec