OctoPrint/OctoPrint #3825: [BUG] 1.5.0rc2 Windows Unable to Install Plugins

jneilliii, 2020-11-18:
#### What were you doing?

Testing 1.5.0rc2 on widows and installing plugins. Just to verify I completed created a new python 2.7 venv and am able to reproduce.

1. Create python 2.7 venv
2. Install OctoPrint 1.4.2 via `pip install octoprint` in active venv
3. Go through initial install wizard
4. Upgrade OctoPrint to 1.5.0rc2 via `pip install https://github.com/OctoPrint/OctoPrint/archive/1.5.0rc2.zip` in active venv
5. Get below error, doesn't matter which plugin from plugin manager

#### What did you expect to happen?

Plugin to install

#### What happened instead?

No Install

#### Did the same happen when running OctoPrint in safe mode?

N/A, can't install plugins from scratch so no other plugins are loaded.

#### Version of OctoPrint

1.5.0.rc2

#### Operating System running OctoPrint

Windows 10 64 bit

#### Printer model & used firmware incl. version

Virtual

#### Browser and version of browser, operating system running browser

Google Chrome Version 86.0.4240.198 (Official Build) (64-bit)

#### Link to octoprint.log

[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint/files/5556930/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5556931/octoprint.log)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No errors

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

jneilliii, 2020-11-18:
It appears this issue existed in 1.5.0rc1 as well. I have reverted my install to 1.4.2 and plugin installs work fine.

cp2004, 2020-11-18:
Only slightly helpful, but I remember installing plugins successfully quite recently, so it's the later side of changes that might have caused this. Confirmed bug, probable regression.

foosel, 2020-11-18:
Consider me mightily puzzled by this, as the error is something that I'd expect to raise regardless of Python version and underlying OS 😶

foosel, 2020-11-18:
Oh, never mind, the issue actually appears to be yet another case of unicode strings making it into the subprocess environment on Windows:

```
Exception in thread Thread-27:
Traceback (most recent call last):
  File "c:\Python27\Lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\Python27\Lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "D:\Code\OctoPrint\OctoPrint\src\octoprint\util\fixes.py", line 28, in decorated_function
    return f(*args, **kwargs)
  File "D:\Code\OctoPrint\devenv27\lib\site-packages\sarge\__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "D:\Code\OctoPrint\OctoPrint\src\octoprint\util\fixes.py", line 28, in decorated_function
    return f(*args, **kwargs)
  File "D:\Code\OctoPrint\devenv27\lib\site-packages\sarge\__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "D:\Code\OctoPrint\OctoPrint\src\octoprint\util\fixes.py", line 28, in decorated_function
    return f(*args, **kwargs)
  File "D:\Code\OctoPrint\devenv27\lib\site-packages\sarge\__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
  File "c:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "c:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
TypeError: environment can only contain strings
```

I wonder how that crept back in, I remember fighting against that for 1.4.0...

foosel, 2020-11-18:
Explanation was turning

``` python
kwargs = dict(env=dict(PYTHONWARNINGS="ignore:DEPRECATION::pip._internal.cli.base_command"))
```

([source](https://github.com/OctoPrint/OctoPrint/blob/559c9b4e527eda242ed0ad3d389bf64c9aef18e5/src/octoprint/plugins/pluginmanager/__init__.py#L1029))

into 

``` python 
kwargs = {
    "env": {
        "PYTHONWARNINGS": "ignore:DEPRECATION::pip._internal.cli.base_command"
    }
}
```

([source](https://github.com/OctoPrint/OctoPrint/blob/9f5781eff4d8b21e5fa6166fab8f4c1e4429710e/src/octoprint/plugins/pluginmanager/__init__.py#L1373-L1377))

which turned the *key* into unicode, which then didn't get fixed by 

``` python
# if we are running under windows, make sure there are no unicode strings in the env
if get_os() == "windows" and "env" in kwargs:
    kwargs["env"] = dict(
        (k, to_native_str(v)) for k, v in kwargs["env"].items()
    )
```

([source](https://github.com/OctoPrint/OctoPrint/blob/rc/maintenance/src/octoprint/util/commandline.py#L177-L179))

Fix is turning that into

```
# if we are running under windows, make sure there are no unicode strings in the env
if get_os() == "windows" and "env" in kwargs:
    kwargs["env"] = dict(
        (to_native_str(k), to_native_str(v)) for k, v in kwargs["env"].items()
    )
```

([source](https://github.com/OctoPrint/OctoPrint/blob/staging/maintenance/src/octoprint/util/commandline.py#L177-L181))

Ready for 1.5.0rc3.

foosel, 2020-11-18:
1.5.0rc3 is out

closed 2020-11-18 by foosel

Sophist-UK, 2020-11-22:
I am unclear - is windows install working in 1.5.0rc3 or not?

cp2004, 2020-11-22:
Yes, it does.

Sophist-UK, 2020-11-22:
@cp2004 Charlie - Thanks - I will give it a go in a few minutes.

commit 4768a7eb (2020-11-18): 🐛 Value AND key might need str-conversion

Closes #3825