OctoPrint/OctoPrint #3827: [RC Feedback] Feedback on the 1.5.0rc3 Release Candidate

foosel, 2020-11-18:
Please provide *general* feedback on your experience with the 1.5.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

  * 🐛 https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/75 PrettyGcode - [PR available](https://github.com/Kragrathea/OctoPrint-PrettyGCode/pull/76) - no reaction yet from maintainer @Kragrathea
  * 🐛 https://github.com/BillyBlaze/OctoPrint-FullScreen/issues/32 OctoPrint-Fullscreen

### Issues needing analysis

CRCinAU, 2020-11-19:
Just updated to rc3 from rc1 - I'd like to flag this as an issue I keep seeing on RC1 that I don't seen mentioned in the release notes...

I've been doing a lot of tweaking to my printer (hello to a 500W 240v AC heatbed!) and I've noticed that a lot of the time the Terminal tab just stops updating. It seems that messages I type in are not displayed in the window - but they do get to the printer.

The results of things such as a G29 don't show until I reload the page and hit all my hide checkboxes again - then I can see what was output.

Will experiment to see if this issue continues with RC3 - and if it matters, its on the RC of OctoPi as well...

mild-lemon, 2020-11-19:
Works fine for me, no quirks encountered yet.

cp2004, 2020-11-19:
@CRCinAU If you think there is a bug, please open a full bug report, definitely with the browser error console. Please also try in safe mode, to check it is not a plugin issue. This is all in the template when you open a bug report.

cp2004, 2020-11-19:
Note that I'm unable to reproduce this, but proper reproduction steps & logs in a bug report would make sure I am looking at the right thing.

foosel, 2020-11-19:
Sounds like a network issue to be honest, might be that the websocket gets stuck. Or maybe something crashes in the backend while pumping data. Impossible to debug without logs, do as Charlie said, please open a ticket, and be sure to also include the browser's error console and network tab in addition to octoprint.log

zeroflow, 2020-11-19:
Works good for me.
Known issue of Webcam-Fullscreen Plugin also seen.
Print successful.

OctoPi on a Pi 3B+
Prusa MK3S + MMU2S

ctgreybeard, 2020-11-19:
I'm in the "Working Fine ..." camp. Most notably (for me) the Telegram plugin is working great.

Pavulon87, 2020-11-20:
Has anyone noticed a different behavior with filament runout? Previously, I only used this as a notification.
Now, as soon as it detects `action:filament_runout T0`, it pauses printing.

> 2020-11-20 09:10:45,750 - octoprint.plugins.action_command_notification - INFO - Got a notification: Printing...
2020-11-20 09:10:47,396 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Pausing"
2020-11-20 09:10:54,555 - octoprint.util.comm - INFO - Communication timeout while printing, trying to trigger response from printer.
2020-11-20 09:11:14,570 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Shape-Box_0.2mm_PLA_10m.gcode, owner: Konrad, user: None
2020-11-20 09:11:14,865 - octoprint.plugins.action_command_notification - INFO - Got a notification: Runout detected
2020-11-20 09:11:14,871 - octoprint.util.comm - INFO - Changing monitoring state from "Pausing" to "Paused"

Same behavior in safe mode.
Any option to disable pausing?

Pavulon87, 2020-11-20:
Nevermind. I see now:

>elif action_name == "pause":
		self._log("Pausing on request of the printer...")
		self.setPause(
				True, tags={"trigger:serial.action_command.pause"}
		)

AndKe, 2020-11-20:
Fullscreen plugin stopped working with 1.5.0 RC1 ,2,3
https://github.com/BillyBlaze/OctoPrint-FullScreen/issues/32

jneilliii, 2020-11-21:
Icons broken in TouchUI due to FontAwesome upgrade.

https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/449

BillyBlaze, 2020-11-22:
> Fullscreen plugin stopped working with 1.5.0 RC1 ,2,3
> [BillyBlaze/OctoPrint-FullScreen#32](https://github.com/BillyBlaze/OctoPrint-FullScreen/issues/32)

Fixed in 0.0.6 release

BillyBlaze, 2020-11-22:
> Icons broken in TouchUI due to FontAwesome upgrade.
> 
> [BillyBlaze/OctoPrint-TouchUI#449](https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/449)

TouchUI issues has been fixed; not yet released

AndKe, 2020-11-23:
Every first job sent from Slic3r fail this way
[image]

retry, and it succeeds.

foosel, 2020-11-23:
Could be that the name resolution takes too long, making the connection fail. Try if switching to the IP works. If connecting via the browser works, this is rather something for slicer devs to figure out.

AndKe, 2020-11-23:
@foosel  At the same time octoprint is open, at all times (I turn on the power there)
So the name is resolved and cached way before.
Also - the error is 500: connection refused  (an active decision has been made)
Finally: the same old Slic3r - build commit 54a31ee Works perfectly when talking (using another profile) to another Octoprint - not upgraded to a beta.   This spawned together with me upgrading one to 1.5.0rc3  ... no so .. it would be wrong to blame >6months old slic3r that works with another octopi :)

cp2004, 2020-11-23:
>  the error is 500: connection refused

HTTP 500 is an internal server error, not connection refused - going to need logs and full bug report in that case.

Guilouz, 2020-11-23:
I have a message "Rate limit exceeded while checking for update, please try again later" when going to Settings / Updates

[image]

cp2004, 2020-11-23:
> I have a message "Rate limit exceeded while checking for update, please try again later" when going to Settings / Updates

That's because the rate limit was exceeded 🙂 
The limit is 60 requests per hour, and each plugin takes 1 request. Also multiple devices on the same network can add to this iirc.

This has always been a problem, but now it is reported. Please read the release notes.

It is also possible to add a Github account token to `config.yaml` to increase the cap (to ~5000 iirc) - see also #3777 *which is also linked from the release notes*

Guilouz, 2020-11-23:
@cp2004 It's the first time I have this, since I updated to RC3

And I have another big issue, I have two pi and I have issue on both. When I start them, I have access to UI but few minutes after, hosts are unreachable, impossible to join them, need to restart pi to access it again to then switch back to inaccessible.
 
Any idea to fix this ? Issue since RC3, no problem with RC2

cp2004, 2020-11-23:
Logs, logs, logs please. I have no idea why your instance is going offline, because my crystal ball stopped working yesterday. But note that these logs should be in the form of a bug report.

If you think it is a bug, then report it as such. If not - please head to the [community forums](http://community.octoprint.org) for support.

The only change from RC2 -> RC3 was with plugin installs under windows if I remember correctly, so that is *unlikely* to be the issue but again that'll need you to open a bug report. You could try downgrading to RC2 or the stable release to check this is the issue.

AndKe, 2020-11-23:
my 500-error shows nothing of interest in the normal debug log.
What options should I choose to log to see everything related to the API access?

foosel, 2020-11-23:
@AndKe I'm unclear on whether this error even ever makes it *to* OctoPrint, the message format looks more like it is something inside slic3r. So a log from that would probably be helpful, and a full octoprint.log as well in a full bug report please.

@Guilouz can you still ping the pi then? If so, bug report please with logs, testing in safe mode and so on. If not: likely a network issue, not something i can fix in userspace software like OctoPrint. As Charlie pointed out, the only change between rc2 and rc3 was related to plugins under Windows and Python 2.

All in all I'd like to remind you that the goal of RCs is to get *actionable* information on issues with it. That means full blown bug reports if something is wrong, not a comment that something "is not working". I put this into every release announcement, changelog and feedback ticket. Please don't make me have to beg for log files all the time 😕

foosel, 2020-11-24:
I was originally aiming for a stable release of 1.5.0 for tomorrow, considering the so far VERY positive look in the stats:

[image]

However, that will have to wait until next week now (I don't release after Wednesday - gets too close to the weekend) due to still waiting for *proper* bug reports here on the problems seen by @AndKe and @Guilouz. If I don't get them by Thursday, I'll have to move forward without analysis, so *please* give me something to work with here.

AndKe, 2020-11-24:
@foosel  please define "full octoprint log" - there is no such option. I am already logging "octoprint" at level "DEBUG"  (and see nothing related to the failed attempt.)

cp2004, 2020-11-24:
@AndKe Full octoprint.log means upload the whole file. Sometimes people only upload some random parts of the log file, not the entire thing. And I can't find anything on the internet about Slic3r logs. 

If you want to get a log of *every* HTTP request, run OctoPrint using `octoprint serve --debug` command.

Either open a full bug report and we will discuss and analyse it there, whether you think you have all the information or not, not on this feedback ticket. Be sure to answer all the questions in the template.

foosel, 2020-11-24:
> I am already logging "octoprint" at level "DEBUG" (and see nothing related to the failed attempt.)

And for some reason you still have not *shared* it for me to confirm that there actually is nothing in there, and to check what else might be in there hunting at what could be going on here. I ask for logs so I can get *myself* a picture of what's happening. You are free to read them yourself, but that doesn't mean I don't still want to see them.

But without a slic3r log or maybe whatever it might spew to the command line when run from there on reproduction, I fear we are at a dead end anyhow. You confirmed that the browser frontend works, it is likely that the first request from slic3r never even reached OctoPrint and the error code is misleading and not an http error at all, so everything points to a network problem or a name resolution problem. The latter could be ruled out by testing whether using the IP instead of the .local domain works, but you never reported back on that suggestion either. 

From where I stand, this is nothing I can do anything about as it seems to happen in slic3r and/or your network from what information I've gotten so far. I believe when you say this didn't happen earlier, but as things are now I don't see anything that puts the ball into OctoPrint's court or would at least allow me to analyse anything. My hands are currently tied.

AndKe, 2020-11-24:
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5592846/octoprint.log)
@foosel  This is the log I checked yesterday, saw nothing strange.
As long it is only me who report this issue, you are right to nor care about is as a showstopper for 1.5.0 stable.
I ignored it myself at first (it's trivial to click "send to printer"  again)  - hence it's not really annoying.

cp2004, 2020-11-24:
>octoprint.log
@foosel This is the log I checked yesterday, saw nothing strange.
As long it is only me who report this issue, you are right to nor care about is as a showstopper for 1.5.0 stable.
I ignored it myself at first (it's trivial to click "send to printer" again) - hence it's not really annoying.

Do you know what a bug report is? I apologise if what comes next sounds rude, its getting *really* frustrating now.
* This is a feedback ticket - for quick comments about the release.
* This is not a support ticket.
* If you find a problem, you are asked to raise a bug report.
* You (@AndKe) were asked to provide a bug report 4 posts ago.
* I have spent too long replying already that you should post a bug report.
* >As long it is only me who report this issue, you are right to nor care about is

  This is not how reporting issues works, if there is a problem in the release candidate that **did not happen on previous releases** then it should be fixed.
* You haven't posted a bug report yet as asked.
* Discussion on this issue would be easier in a bug report, not here.
* I can't even waste my *own time* downloading, installing and testing Slic3r, since I know next to nothing about your situation.
* You've ignored every other question so far, so I present a list:
   * Try if switching to IP works
   * Run using `octoprint serve --debug` to log all HTTP requests
   * Does it still work on OctoPrint 1.4.2 (**on the same instance - downgrade it**)

In other news, I just tried this with Slic3r and it chucks out 500 error if the address can't be reached.

DodgeDeBoulet, 2020-11-27:
WebCamTab plugin is no longer working (and no, this is not because of the Python 3 issue; I've installed the patch that allows the plugin to load successfully). After loading, the tab displays but no video output from the webcam is presented.

I've attached octoprint.log @ DEBUG logging level, but setting the WebCamTab plugin to debug level does not produce a separate log.

I have opened [issue #13 in the author's repository](https://github.com/malnvenshorn/OctoPrint-WebcamTab/issues/13) but have so far received no response.

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5608088/octoprint.log)

JohnOCFII, 2020-11-27:
Similar to my RC2 feedback -- all is looking good for me with RC3:

- Upgraded (from rc2) with no issues
- Uploaded new GCODE
- Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
- Printed GCODE on Prusa MK3S/wMMU2 in multi-extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- The modified Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Telegram plugin worked well
- Firmware Updater worked well

AndKe, 2020-11-27:
@cp2004 @foosel  Yes, you have all right to be rude. 
I am busy with lots of other stuff, and did not take time to peoperly check everyrhing.
It turn out this could be some odd DNS thing, because as Foosel guessed, changing to IP rather than DNS name solved it.
It is very strange, as the computer has the session up, and nothing else indicate a delay/DNS trouble.
Coincidently - not only Octoprint were upgraded on this netwark, but also the Fortigate. (firewall)
 This is clearly not an OctoPrint issue. 
thank you for what you do (including putting up with users)

FormerLurker, 2020-11-28:
I'm getting the following error trying to run via a pycharm debugger:

```
File "C:\Users\Brad\source\repos\Octolapse_v0_4_OcotprintV1_3_12\OctoPrint\src\octoprint\plugins\forcelogin\__init__.py", line 129, in check_login_required
AttributeError: 'module' object has no attribute 'is_blueprint_protected'
```

I was prompted to create an access protection account (this is a debug instance, so I never had one before), which I did.  After this all subsequent attempts to navigate to localhost:5000 give me the above error, even after clearing the cache and signing on again.  I verified that the venv has the current source (```2020-11-28 12:42:27,453 - octoprint.startup - INFO - Starting OctoPrint 1.5.0rc3```).  Unsure what to do here.  Is this a bug or something else?

FormerLurker, 2020-11-28:
So, something changed that broke Octolapse in a way that I'm having trouble even figuring out what's going on.  Help icons are rendering multiple times (3), I get errors in foreach bindings for empty arrays (should not be rendered in that case), and the settings screens are only showing the name and descriptions.  I will dig in and report what I find.

foosel, 2020-11-28:
@FormerLurker I was just about to say it looks like you have left overs there somehow which are causing problems. Might also be what's causing you issues now. Try if a `python setup.py clean` helps.

FormerLurker, 2020-11-29:
@foosel, thanks for the time!  I'll try to remember that next time I upgrade :)

Regarding the other issues, it looks like they were triggered by the knockout upgrade.  Been dealing with the issues one at a time with the help of a few friends (Thanks @jneilliii  and @cp2004 !!).  So the new version HATES self closing tags.  These were errors, to be sure (I'm used to XHTML where they are valid), but the errors caused by say using <textarea \> or <a\> are mysterious and incredible.  Stuff was duplicated, omitted, rearranged, and worse.  Beware of this.  I'm combing through my jinja2 templates now trying to find any other errant self closing tags.

cp2004, 2020-11-29:
Unable to reproduce, but in some cases wizard dialogs refuse to close apparently. I have 2 reports of this, and I know there were people with OctoEverywhere that *I think (can't actually remember)* had this issue.

Internal server error aparrently, waiting on a log: https://github.com/cp2004/OctoPrint-EEPROM-Marlin/issues/13#issuecomment-735300396

cohaolain, 2020-11-29:
Here's all the relevant log entries, related to the 500 mentioned by @cp2004 :
```
2020-11-28 22:41:10,222 - octoprint - ERROR - Exception on /api/settings [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1539, in decorated_view
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 387, in setSettings
    response = _saveSettings(data)
  File "/usr/local/lib/python3.8/site-packages/octoprint/server/api/settings.py", line 933, in _saveSettings
    "gcode", name, script.replace("\r\n", "\n").replace("\r", "\n")
AttributeError: 'NoneType' object has no attribute 'replace'
2020-11-28 22:41:10,224 - tornado.access - ERROR - 500 POST /api/settings (172.23.0.1) 5.32ms
```

foosel, 2020-11-30:
On first glance that doesn't look like it is a regression. For some reason, a settings save is triggered here that contains a GCODE script set to `null`/`None`. The code in question hasn't been touched since 2015 (apart from the `black` reformatting that looks clean). The settings save call on the click of "finish" in the wizard also hasn't seen a chance since 2015. 

Something here is causing the local settings data to be wonky. I'll try to reproduce, but so far haven't seen it in my tests at all. It might be specific to a certain plugin constellation.

foosel, 2020-11-30:
1.5.0 has just been released. The 500 settings error was not deemed a regression and thus not included in that, but I added a fix for that above that will definitely be in 1.6.0, and if push comes to shove and this becomes a reproducible common issue there's always the option to roll it out as a hotfix.

closed 2020-11-30 by foosel