OctoPrint/OctoPrint #3831: Blocking error during printing - "octoprint.util.comm - ERROR - Something crashed inside the serial connection loop"

lucarp78, 2020-11-23:
Hi,
I was printing the 3d benchy multi color with canvas hub s connected to palette 2s pro and to my printer (creality ender 5 plus) and it stopped twice after many hours of job.
I prepared the STL file with Ultimaker Cura and Chroma, the file was different in the second attempt because I changed the flow of the initial layer.

Here the first log: 

2020-11-22 13:50:47,687 - octoprint.plugins.palette2 - INFO - SPLICE START
2020-11-22 13:50:47,998 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-22 13:50:48,000 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-22 13:51:18,028 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-22 13:51:18,030 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-22 13:51:19,738 - octoprint.plugins.palette2 - INFO - Omega Read Thread: reading: O97 U25 D1 D0069
2020-11-22 13:51:19,741 - octoprint.plugins.palette2 - INFO - SPLICE END
2020-11-22 13:51:48,062 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-22 13:51:48,064 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-22 13:52:18,097 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-22 13:52:18,099 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-22 13:52:24,638 - octoprint.plugins.palette2 - INFO - Omega Read Thread: reading: O97 U26 D00003A94
2020-11-22 13:52:24,639 - octoprint.plugins.palette2 - INFO - 14996mm used
2020-11-22 13:52:24,643 - octoprint.plugins.palette2 - INFO - Updating UI: filamentLength
2020-11-22 13:52:48,133 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-22 13:52:48,137 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-22 13:52:56,654 - octoprint.util.comm - ERROR - Exception while processing line
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4242, in getNext
    processed = self._process(line, offsets, current_tool)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4251, in _process
    return process_gcode_line(line, offsets=offsets, current_tool=current_tool)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4513, in process_gcode_line
    line = apply_temperature_offsets(line, offsets, current_tool=current_tool)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4462, in apply_temperature_offsets
    match = _temp_command_regex.match(line)
MemoryError
2020-11-22 13:52:56,656 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1836, in _monitor
    self._handle_ok()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2323, in _handle_ok
    self._continue_sending()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2491, in _continue_sending
    elif job_active and self._send_from_job():
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2998, in _send_from_job
    line, pos, lineno = self._get_next_from_job()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2967, in _get_next_from_job
    line, pos, lineno = self._currentFile.getNext()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4248, in getNext
    raise e
MemoryError
2020-11-22 13:52:56,685 - octoprint.server - **CRITICAL - Now that is embarrassing... Something really really went wrong here. Please report this including the stacktrace below in OctoPrint's bugtracker. Thanks!**
2020-11-22 13:52:56,687 - octoprint.server - ERROR - Stacktrace follows:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/__init__.py", line 940, in run
    ioloop.start()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/ioloop.py", line 1048, in start
    event_pairs = self._impl.poll(poll_timeout)
MemoryError
2020-11-22 13:52:56,690 - octoprint.server - INFO - Shutting down...
2020-11-22 13:52:56,692 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: See octoprint.log for details)"
2020-11-22 13:52:56,717 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-11-22 13:52:56,756 - octoprint.plugins.palette2 - INFO - Palette 2 Led Thread Error

and the second one:
2020-11-23 20:29:25,717 - octoprint.plugins.palette2 - INFO - SPLICE START
2020-11-23 20:29:27,763 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-23 20:29:27,766 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-23 20:29:57,760 - octoprint.plugins.palette2 - INFO - Omega Read Thread: reading: O97 U25 D1 D00aa
2020-11-23 20:29:57,762 - octoprint.plugins.palette2 - INFO - SPLICE END
2020-11-23 20:29:57,798 - octoprint.plugins.canvas - INFO - Connecting shadow client...
2020-11-23 20:29:57,800 - octoprint.plugins.canvas - INFO - Could not connect shadow client
2020-11-23 20:30:21,323 - octoprint.plugins.palette2 - INFO - Omega Read Thread: reading: O97 U26 D00005B51
2020-11-23 20:30:21,325 - octoprint.plugins.palette2 - INFO - 23377mm used
2020-11-23 20:30:21,328 - octoprint.plugins.palette2 - INFO - Updating UI: filamentLength
2020-11-23 20:30:24,326 - octoprint.util.comm - ERROR - Exception while processing line
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4242, in getNext
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4251, in _process
    return process_gcode_line(line, offsets=offsets, current_tool=current_tool)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4513, in process_gcode_line
    line = apply_temperature_offsets(line, offsets, current_tool=current_tool)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4462, in apply_temperature_offsets
    match = _temp_command_regex.match(line)
MemoryError
**2020-11-23 20:30:24,330 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:**
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1836, in _monitor
    self._handle_ok()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2323, in _handle_ok
    self._continue_sending()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2491, in _continue_sending
    elif job_active and self._send_from_job():
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2998, in _send_from_job
    line, pos, lineno = self._get_next_from_job()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2967, in _get_next_from_job
    line, pos, lineno = self._currentFile.getNext()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4248, in getNext
    raise e
MemoryError
2020-11-23 20:30:24,400 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: See octoprint.log for details)"
2020-11-23 20:30:24,448 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-11-23 20:30:24,503 - octoprint.plugins.palette2 - INFO - Updating UI: actualPrintStarted
2020-11-23 20:30:24,524 - octoprint.plugins.palette2 - INFO - Palette 2 Led Thread Error

Attached the complete log.
[octoprint.log.zip](https://github.com/OctoPrint/OctoPrint/files/5585996/octoprint.log.zip)

cp2004, 2020-11-23:
You need to test this in safe mode, to check the memory error is not coming from the plugins.

How large is the file you are trying to print?

lucarp78, 2020-11-23:
The file is large 9.4 MB ... if I start the canvas hub in safe mode it will not be able to communicate with the palette 2s pro and print.

foosel, 2020-11-24:
The problem is, it's very likely that this issue is in fact caused by one of your third party plugins, hence the request to please test in safe mode. So if you could test with palette out of the equation, that would help. Otherwise I'd ask you to please get in touch with Palette first to have them check that it is definitely NOT their plugin causing memory to run out here, as in both cases of errors you shared here there was activity from their plugins right before, including something that hints at some kind of error that could mean connection attempts are not getting cleaned up. I don't have the hardware to test myself, and I must insist on an error to reproduce in safe mode first before I can start analyzing it, otherwise chances are high it's a bug that's not in core OctoPrint, thus completely outside of my means to mitigate and any kind of analysis thus a waste of resources that's unfair to the general user base of OctoPrint.

lucarp78, 2020-11-24:
ok, I will try to load in safe mode.

closed 2020-12-08 by GitIssueBot