OctoPrint/OctoPrint #3832: cancelling print doesnt turn off extruder

Dloranger, 2020-11-25:
#### What were you doing?
Running a print as normal, visually observe a print defect, and hit the cancel print button.

observe the temperature graph, see the hot bed is disabled, but hotend is holding temperature still.

This is not 100% repeatable, uncertain what the exact conditions are to reproduce it

#### What did you expect to happen?
Hot end to be disabled
#### What happened instead?
hotend was momentarily disabled, but then reasserted to full printing temperature
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
------------------------------------------------------------------------------
OctoPrint version : 1.4.2
OctoPi version    : 0.15.1
------------------------------------------------------------------------------

pi@octopi-diamond:~ $ uname -a
Linux octopi-diamond 4.14.34-

#### Printer model & used firmware incl. version
Custom Anet A8, Tango mainboard, full firmware files 
https://github.com/bigtreetech/Tango-3D-Printer-Motherboard/issues/4#issuecomment-468336950

#### Browser and version of browser, operating system running browser
Chrome running on an archaic android tablet & windows 10 laptop 64 bit

#### Link to octoprint.log

file is too big to open, 1.2GB

#### Link to contents of terminal tab or serial.log

will need to capture, will update when available

#### Link to contents of Javascript console in the browser

will need to capture, will update when available

#### Screenshot(s)/video(s) showing the problem:

will need to capture, will update when available

I have read the FAQ.

cp2004, 2020-11-25:
> file is too big to open, 1.2GB
This is suspiciously big, my development logs get large and that's 20MB. Sounds *either* like a *ton* of errors, or a plugin filling the log with errors. Maybe delete the existing log file, reproduce the issue and then it should be small enough to upload?

Also: Please format logs using the 'preformatted text' button, looks like `< >`, to make it easier to read.

Dloranger, 2020-11-25:
the logs were left on for ages now, I think that is the likely culprit for the file size.  I forgot to disable them last time I was debugging an issue

foosel, 2020-11-25:
Logs should rotate each day, so that doesn't make sense and sounds really fishy.

Delete logs, enable safe mode and serial.log, reproduce and share please.

foosel, 2020-11-25:
Actually, never mind, just saw your serial.log.

OctoPrint is telling your printer to set the temperature to off (N24791). Your printer ignores that command (target in following temperature reports is still set). That's a firmware issues, not an issue with OctoPrint.

You could try changing the cancel code to contain an explicit M104 S0, maybe your firmware doesn't like the tool parameter. But that's about it. OctoPrint can't do anything else but tell the printer to change the target temperature, which it does here. If the firmware ignores that command, there's nothing OctoPrint can do.

closed 2020-11-25 by foosel

Dloranger, 2020-11-25:
I think that may be partially incorrect..

Send: N24791 M104 T0 S0*40

it specifically is calling out T0 there, T1 is the active extruder at the moment

Dloranger, 2020-11-25:
here is another snip that explicitly shows its on T1

```
Recv: echo:Active Extruder: 1
Recv: ok
Send: N7605 G92 E0*115
Recv: X:126.85 Y:123.09 Z:3.00 E:0.00 Count X:12685 Y:12009 Z:1200
Recv: ok
[...]
Send: N7607 M109 S200*92
[...]
Changing monitoring state from "Printing" to "Cancelling"
[...]
Recv: ok
Send: N7608 M108*19
Recv: ok
Send: N7609 M400*31
Recv: ok
Send: N7610 M114*23
Recv: X:126.85 Y:123.09 Z:3.00 E:0.00 Count X:12685 Y:12009 Z:1200
Recv: ok
Send: N7611 M84*46
Recv: ok
Send: N7612 M104 T0 S0*19
Recv: ok
Send: N7613 M140 S0*86
Recv: ok
Send: N7614 M106 S0*83
Recv: ok
Changing monitoring state from "Cancelling" to "Operational"
```

foosel, 2020-11-25:
According to your temperature report, there's only one heater (T = T0), and that is being shut off. Is this a shared nozzle setup (should only have on heater but multiple extruders feeding into it)? How many extruders are configured in the printer profile in OctoPrint? What is your cancel gcode snippet?

Dloranger, 2020-11-25:
This is indeed a shared nozzle setup, 1 heater, 3 feeders.  

[image]

**After print job is cancelled** _(all others are blank)_
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0

foosel, 2020-11-25:
In that case you have three options here.

1. modify the firmware so that it behaves like other shared nozzle firmware out there and will turn off the heater regardless of currently selected extruder when receiving an M104 T0 S0

2. modify the cancel script so it contains an explicit M104 S0 instead of the disable_hotends script include (worth a try, but not sure if it'll work with your firmware)

3. modify the cancel script and replace the disable_hotends script include with explicit calls of M104 S0 for T0 through T2

I cannot modify the bundled snippet as that would break existing setups or there that all behave differently than yours, but you can just change it yourself, it's a modifiable script after all.

Dloranger, 2020-11-25:
I am looking to recompile the firmware anyways, any idea what setting values to check to make sure the M104 is honored as expected? I don't think I changed anything related to this, but anything is possible

foosel, 2020-11-25:
Sorry, I never so far configured any firmware for a shared nozzle setup myself due to lack of the hardware and don't feel qualified to give advice here. The only thing I can tell you is how it should behave from the host's perspective 😅