OctoPrint/OctoPrint #3834: [Request] Please add new action command.

AndKe, 2020-11-25:
**Is your feature request related to a problem? Please describe.**
the printer is in >100m away from my workstation..
I go to the printer, remove previous print, and want to restart another batch.
or:
I go to printer, remove some mess, refill filament, and want to start job over, stay and see it start well.

**Describe the solution you'd like**
I would like to be able to restart last print job from the printer

**Describe alternatives you've considered**
https://docs.octoprint.org/en/master/features/action_commands.html
Add action command "printlast" that starts the last job, (be it a successful, or failed one)

**Additional context**
The command should simply repeat the "print" action on whatever job was the last one the user clicked "print" on.

foosel, 2020-11-25:
1.5.0 will already come with support for the `start` action that starts the currently selected print job (which will be the same one again if one already finished)

https://docs.octoprint.org/en/maintenance/features/action_commands.html

So - already implemented.

closed 2020-11-25 by foosel

AndKe, 2020-11-25:
fantastic.. just fantastic.  :)