OctoPrint/OctoPrint #3840: [Request] Better to use octoprint.org or github.com for connectivity check?

deosrc, 2020-11-30:
**Is your feature request related to a problem? Please describe.**

Having just gone to the OctoPrint UI to install a plugin, I found OctoPrint claiming that it was offline. After some investigation I found this was because the connectivity check uses Cloudflares DNS (1.1.1.1, port 53).

Like many others using PiHole, I have configured my firewall to deny requests on port 53 outside of my network. This prevents devices using their own DNS settings to bypass local network configurations (a practice used to prevent DNS ad blocking). This had the side-effect of OctoPrint thinking it was offline when it actually had full connectivity.

**Describe the solution you'd like**

Given that most of the OctoPrint functionality which requires an internet connection uses `octoprint.org` or `github.com` (e.g. plugin installs, updates), would it not be better to use one of these for the connectivity check?

**Describe alternatives you've considered**

As a minimum, I would think there should be an indicator in the header bar or notification when the page is first loaded to show that OctoPrint thinks it is offline. There doesn't seem to be an indication at the moment which means that users could be missing out on vital security updates without realising.

**Additional context**

#2474

foosel, 2020-11-30:
The problem with that is that the general connectivity check needs to a) use an IP (to not interpret issues of name resolution as complete loss of connectivity) and b) needs to be reliably reachable.

I cannot guarantee that the IP of `octoprint.org` will not change at one point, leaving outdated instances stranded. I cannot guarantee that the IP of `github.com` will not change at one point, leaving outdated instances stranded. I also cannot guarantee the level of availability that 1.1.1.1, 1.1.0.0, 8.8.8.8, 8.8.4.4, 9.9.9.9 and similar public DNS have.

Hence the choice to use something as default that is not owned by Google (as that was raising concerns for many), reachable by IP long-term and with a high availability. Thus: a public well-established DNS server.

I'm aware that some people deny external traffic on port 53, the general majority however doesn't and due to the nature of how the internet works it's a good candidate to not get firewalled. Port 80 or 443 would be better, but see the requirements above again as to why that is problematic. The setting is configurable to allow people in your situation to change it so it fits their setup.

So, I'm not so sure if I really want to change this here, due to the aforementioned reasons, but I agree that a notification about the offline state would be a really good idea!

deosrc, 2020-11-30:
Seems fair enough. I think the main issue is the lack of notice so I'd be happy for the resolution to just be the notification.

Just to check, is the default now `1.1.1.1` ? The text still indicates the default is Google but I couldn't recall if I'd changed the value at some point.

> If you don't know what to set here, the default values (using Google's DNS server) should work.

foosel, 2020-11-30:
Now you made me check, and damn, I misremembered myself. I *wanted* to switch it to 1.1.1.1 due to the aforementioned concerns, but so far didn't.

lasmaty07, 2021-01-20:
Well... this gave me one hell of a headache. 
Turns out the name resolution configured on the server tab in octoprint **IS NOT** using the DNS IP set on the HOST IP field.

### My case:
I had configured static IP, Gateway and DNS via the `/etc/dhcpcd.conf` file. the DNS was set to my local Pi-Hole running on another rpi, with no secondary DNS IP (I know, my bad). So then my Pi-Hole suddenly died (faulty sd) and octoprint said it had no internet connection. The conection to the HOST IP Server (8.8.8.8:53 by default) showed ok. But name resolution was not OK. 

### Request:
Please add a disclaimer on the server tab saying that name resolution does not use the IP set on HOST IP field. I cant figure out if this only happens when the `/etc/dhcpcd.conf` is edited or not. but anyway.

foosel, 2021-01-20:
What made you think it was supposed to be used for name resolution in the first place? It is a connectivity check that utilizes a DNS endpoint by default simply because that is readily and highly available, that doesn't mean it is a DNS configuration or it would explicitly say so (and would also be very much out of scope for OctoPrint). You can configure any IP address and port there.

You are the first person I heard of that assumed this would actually manipulate DNS resolution in any way, I'm a bit worried that adding a statement like you are requesting here would more confuse than help to be honest.

foosel, 2021-03-23:
Implemented and ready for 1.6.0.

[image]

Icon can also be clicked and links to https://faq.octoprint.org/connectivity-offline

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel