OctoPrint/OctoPrint #3841: upgrade to Octoprint 1.5

Drummerboy2, 2020-11-30:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Drummerboy2, 2020-11-30:
I did not download any logs or messages yet. First I wanted to find out if there is an issue. I upgraded to v 1.5 and immediately my Prusa thumbnail does not show any thumbnails and if I click on the icon nothing comes up. I changed nothing else.
Thanks Brian

jneilliii, 2020-11-30:
Related to the upgrade, but is an issue with the plugin, not octoprint. There is already one issue opened on it's repo for the state panel not getting the thumbnails in 1.5.0.

closed 2020-11-30 by foosel