OctoPrint/OctoPrint #3842: 1.5.0 update broke Octolapse and webcam performance

badpixelvfx, 2020-12-01:
#### What were you doing?

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2020-12-01:
there is an updated version to OctoLapse that fixes issue relative to OctoPrint 1.5.0. This is not an issue with OctoPrint but needs to be reported to the plugin's repo, not here. For your convenience, copy/paste the URL below in Plugin Manager > Get More > ...from URL and clicking install if you want to run the updated version.

```
https://github.com/FormerLurker/Octolapse/archive/v0.4.1rc1.zip
```

cp2004, 2020-12-01:
> Link to octoprint.log
> Did not collect

Since this is a 3rd party issue and has been solved by the author, this can be closed.

closed 2020-12-01 by cp2004