OctoPrint/OctoPrint #3843: 1.5.0 UI degradation: all icons in touchUI menu show as X after update to 1.5.0

akrilov, 2020-12-01:
#### What were you doing?

Updating footprint to 1.5.0
Right after update desktop version of octoprint works fine, but touchUI menu icons all disappeared and replaced by X in a square. Following update of TouchUI did not fix the issue.

browser.user_agent: Mozilla/5.0 (Linux; Android 5.1.1; KFSUWI) AppleWebKit/537.36 (KHTML, like Gecko) Silk/86.2.31 like Chrome/86.0.4240.185 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 917016576
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.0
printer.firmware: Marlin Ver 1.70.1 BL (Creality3D)

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-01:
Fixed in latest release of TouchUI

cp2004, 2020-12-01:
At least, we've been told it is. If not, contact the author.

closed 2020-12-01 by cp2004