OctoPrint/OctoPrint #3844: Files API doesn't return contents of child folder (1.5.0)

UnchartedBull, 2020-12-01:
#### What were you doing?

- Create a subfolder (`test`) in the OctoPrint UI
- Open the subfolder (`test`)
- Upload a file
- Create a subsubfolder (`test2`)
- Execute `GET http://localhost:5000/api/files/local/test`
- Children array is empty and doesn't include either the `test2` folder nor the uploaded file

The GET URL is taken from `refs.ressource` from the response of `GET http://localhost:5000/api/files/local`, which does work by the way.

#### What did you expect to happen?
The children of the subfolder to be listed correctly.

#### What happened instead?
Children array is empty. Maybe I missed something completely obvious but the requests are correct according to the API docs.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.5.0 (happened on rc2 and rc3 as well)

#### Operating System running OctoPrint
Docker, Raspian

#### Printer model & used firmware incl. version
n.a.

#### Browser and version of browser, operating system running browser
Postman, OctoDash (latest Angular)

#### Link to octoprint.log
https://pastebin.com/X60TVLV3

(Sorry for the messy log, tried to create a cleaner one, but that didn't inlcude any error message whatsoever, if you search for `/test` you'll eventually get to the error message).

#### Link to contents of terminal tab or serial.log
n.a.

#### Link to contents of Javascript console in the browser
n.a.

#### Screenshot(s)/video(s) showing the problem:
Response for top-level folder:

```
{
    "files": [
        {
            "children": [],
            "display": "test",
            "name": "test",
            "origin": "local",
            "path": "test",
            "refs": {
                "resource": "http://localhost:5000/api/files/local/test"
            },
            "size": 3912363,
            "type": "folder",
            "typePath": [
                "folder"
            ]
        },
        {
            "DisplayLayerProgress": {
                "totalLayerCountWithoutOffset": "44"
            },
            "date": 1606832919,
            "display": "Body1_0.3mm_PET_ENDER3_1h21m.gcode",
            "gcodeAnalysis": {
                "dimensions": {
                    "depth": 180.0,
                    "height": 13.2,
                    "width": 199.674
                },
                "estimatedPrintTime": 3275.9477617088114,
                "filament": {
                    "tool0": {
                        "length": 6578.786159999465,
                        "volume": 15.8238351128099
                    }
                },
                "printingArea": {
                    "maxX": 201.674,
                    "maxY": 190.0,
                    "maxZ": 13.45,
                    "minX": 2.0,
                    "minY": 10.0,
                    "minZ": 0.25
                }
            },
            "hash": "14c09b682a0662f78cd16597a70b259996ee94e8",
            "name": "Body1_0.3mm_PET_ENDER3_1h21m.gcode",
            "origin": "local",
            "path": "Body1_0.3mm_PET_ENDER3_1h21m.gcode",
            "refs": {
                "download": "http://localhost:5000/downloads/files/local/Body1_0.3mm_PET_ENDER3_1h21m.gcode",
                "resource": "http://localhost:5000/api/files/local/Body1_0.3mm_PET_ENDER3_1h21m.gcode"
            },
            "size": 1304121,
            "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/Body1_0.3mm_PET_ENDER3_1h21m.png?20201201142839",
            "thumbnail_src": "prusaslicerthumbnails",
            "type": "machinecode",
            "typePath": [
                "machinecode",
                "gcode"
            ]
        }
    ],
    "free": 56012165120,
    "total": 62725623808
}
```

Response for `test` sub-folder:
```
{
    "children": [],
    "display": "test",
    "name": "test",
    "origin": "local",
    "path": "test",
    "refs": {
        "resource": "http://localhost:5000/api/files/local/test"
    },
    "size": 3912363,
    "type": "folder",
    "typePath": [
        "folder"
    ]
}
```

I have read the FAQ.

foosel, 2020-12-01:
> (happened on rc2 and rc3 as well)

Would have been lovely to have gotten a report for this during the RC phase then, and not just now when it's out :/

Can reproduce, probably crept in during the cache optimization, will have to check if that merits a hotfix or not. Currently leaning towards not tbh.

UnchartedBull, 2020-12-01:
Yeah sorry, I just tested it today and tried all different version (from rc2 on). The issue in OctoDash was opened before that, but I didn't had the time to upgrade my dev instance and test it myself. Didn't want to bother you with a possible issue in OctoDash. Sorry!

foosel, 2020-12-03:
If in doubt, I'd rather get a report with "it might be on my end" and a "was on my end, never mind" later than not hear anything and think all is good :)

foosel, 2020-12-03:
So, I have a fix for this on the `staging/hotfix` branch, since I might have to push 1.5.1 after all due to an issue with mdns I'm currently investigating in #3852. So IF there's a hotfix release (currently highly likely), it'll be part of it, otherwise this will have to wait until 1.6.0.

closed 2020-12-04 by foosel

UnchartedBull, 2020-12-04:
Whoa that was a fast release! Thanks for everything :)

I'll make sure to let you know of any issues in the OctoDash repository during RC phase now if they pop up, sorry again!

foosel, 2020-12-04:
All good. That's why we communicate :)

commit 1536ed6d (2020-12-03): 🐛 Fix folder listing on files API

Children were not being included without the recursive parameter when it
should in fact only have been grandchildren and later generations.

Closes #3844