OctoPrint/OctoPrint #3845: Adding new Terminal filters providing faulty defaults.

LMS0815, 2020-12-02:
If you are adding a new terminal filter the default entry is faulty as it is missing the escape characters `\`

This is the provided filter:
[image]

This how it should look:
`(Send:\s+(N\d+\s+)?M105)|(Recv:\s+(ok\s+)?.*(B|T\d*):\d+)`

Same issues on Windows and Linux:

```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.66 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 902402048
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.0
printer.firmware: Marlin 2.0.5_DW6.2 TM3D SSMDBU

```

```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.66 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 2701
env.hardware.ram: 34224439296
env.os.bits: 64
env.os.id: windows
env.os.platform: win32
env.python.pip: 20.3
env.python.version: 2.7.14
env.python.virtualenv: false
octoprint.safe_mode: false
octoprint.version: 1.5.0

```

foosel, 2020-12-02:
The system information does not contain logs, and you also deleted some other parts from the bug reporting template. In this case however we can forgo those as I can reproduce.

LMS0815, 2020-12-02:
Thank you - its a minor issue anyway.

Kindly consider adding a `reset` to defaults button to that page :) 
just to get rid of old settings and/or to get the newly added default settings (as 1.5.0   is providing updates to older versions).

cp2004, 2020-12-27:
#3913 fixes this. Seems to be as a result of Prettier - it changed that line & removed the escape characters. Fixed with @LMS0815's provided example, verified it works to suppress things.

foosel, 2021-01-11:
PR merged 🙌, ready for 1.6.0

PR #3913 (2021-01-11): Fix default regexp for new terminal filters
Closes #3845

This was prettier's doing I suspect. Watch out that it doesn't do this one again.

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [x] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the maintainers, e.g. through a
    Brainstorming ticket
  * [x] Your PR targets OctoPrint's devel branch if it's a completely
    new feature, or maintenance if [... cut]
commit e8b0277d (2021-01-11): :bug: Fix default regexp for new terminal filters

Closes #3845

This was prettier's doing I suspect. Watch out that it doesn't do this one again.
closed 2021-04-27 by foosel