OctoPrint/OctoPrint #3846: [Request] Polling if Prusa printer is powered on

brjhaverkamp, 2020-12-02:
I am trying to make octoprint automatically detect that my prusa i3 Mk3 printer powers up.
Portlistener doesn't work, as the USB port doesn't get terminated when powering the Prusa down.
This is documented in 
https://community.octoprint.org/t/autodetect-printer-on-power-on-from-running-octoprint/12088/3
and 
https://community.octoprint.org/t/usb-not-resetting-when-powering-off-printer-prusa-i3-mk3/7112/2
 I would prefer a software solution. It should be simple enough to poll the printer every 30 or 60 seconds and see if it replies. This is already done while the printer is connected. So when I switch the printer off, octoprint automatically detects the connection is lost. I just would like it to work the other way around as well! 

The alternative that is mentioned in the post is to modify a USB cable to disconnect the 5V op put a Diode in. But it is not clear if that works. One post suggests it doesn't. And a software solution would be cleaner.

Please consider this option for a next version.
And thank you for a great piece of software and the hard work!

Mit freundlichen Grussen

Bert Haverkamp

cp2004, 2020-12-03:
Just to be clear, this is something that should be in a plugin, not part of core OctoPrint (unless it is impossible to do currently via a plugin, in which case we can make it possible). Since adding printer-specific code is not something that works well to core OctoPrint, a plugin would be a great candidate - since it doesn't apply to all users.

>It should be simple enough to poll the printer every 30 or 60 seconds and see if it replies.

Also, it is not as simple as you might think, due to all the different firmware variants, firmware bugs, hardware faults, bad cables, interferance, etc. - you just have to read through the forums, this would likely break other things. *That said*, a 'Prusa AutoDetect' plugin would be fine, because it is clearly only intended to be used with a Prusa, so gets around this limitation.

foosel, 2020-12-03:
Agreed, this is not something I'd implement in core. Too high a risk to break stuff for everyone else.

brjhaverkamp, 2020-12-04:
Hi all, Thanks for your comments. 

@foosel, could you explain where the risk is? Just to be sure, I am not looking for an "autoconnect", more than an "auto-reconnect".. I would like octoprint just be a bit more patient (24x7 patient:-) to see if the printer comes back on. But I might be overlooking a usecase here.
I would like octoprint just to keep on trying with the M105 handshake, after it fails. Currently there is the timeout and octoprint will put the printer on disconnected. But if it then keeps sending regular M105s till the printer comes back on, and then put the printer back in operational state, would that work?
Currently there are already 2 tickboxes below the connect button. Maybe a 3rd, "Reconnect printer when switched on" or something similar could help? This should not be printer specific, right? It would work with all printers (assuming that M105 handshake is supported by all printers)

jneilliii, 2020-12-04:
Isn't that what portlister plugin does?

foosel, 2020-12-04:
The risk is that the connection is *gone*. I cannot poll `M105` and check if I get a reply without an open connection. So I would *have* to reconnect to figure out if the printer is back, but for these reconnects I must not generate warnings on fail because that would flood users with senseless messages. Any reconnect outside of the user's control also carries the risk of interrupting ongoing sd prints or leveling routines or whatnot due to the printer controller rebooting on serial connect.

So, long story short, what you want here will boil down to having to implement an automatic reconnect after error scenarios indistinguishable from severe connection issues. And that is just asking for trouble. So: plugin. 

@jneilliii Nope, portlister can only work if the port vanishes and then pops back up. The issue with the prusa machines is that the port stays there, but it simply... dies? You get an exception as if someone had pulled the plug, and reconnection attempts fail (absolutely no responses) until you power the printer back on.

brjhaverkamp, 2020-12-04:
Hi Gina, Thanks for the explanation. That makes sense. On all other machines the connection is really gone. So yes, in my case the solution would be a plugin that does that continuous m105 polling and informs octoprint that the printer is back "alive" and do the connection. Is that all possible from a plugin? 
I assume the re-connection part is possible, as that would be identical to portlistener. I feel a "Prusalistener" plugin is needed here(not sure if I am timewise/experise-wise able to write it though..)

I am wondering if this only an issue with the Prusa i3 mk3(this specific brand/type)? From the number of mentions it feels like I am really stuck in a rare corner case of the octoprint usage..:-)

JasonKeirstead, 2021-01-05:
+1 request for this. I have this exact same problem with my Creately Ender 3. See full description on my original issue #3927

cp2004, 2021-01-05:
Ah I understand it now, you could fix that issue by removing the 5V line (either through taping or cutting the wire or a power-blocker-adapter).

It's a well known issue, and can often cause actual problems (see the forums) such as strange noises (as my ender 5 does).

Try this, see if it makes PortLister work:

https://community.octoprint.org/t/put-tape-on-the-5v-pin-why-and-how/13574?u=charlie_powell

foosel, 2021-01-05:
I will not implement auto reconnect of any kind into OctoPrint - as pointed out that could cause various issues and break things for people, too high risk.

Besides, for the more general situation you are facing @JasonKeirstead there already exists a plugin that does pretty much what you are asking me to implement in core, in the shape of portlister. As already pointed out by @cp2004 in the other ticket, it would make more sense to figure out why that isn't working in your case and fix this in the plugin rather than create a competing (and risky) implementation in core OctoPrint. 

I'm leaving this open as a plugin idea for a specific plugin for Prusa machines for which portlister cannot work (as explained).

JasonKeirstead, 2021-01-05:
@cp2004 Actually I have already cut the 5V line. This isn't related to that, IMO it is related to how OctoPrint does not handle the serial port disappearing from windows elegantly.

JasonKeirstead, 2021-01-05:
If it helps clarify the issue - this is the exact error stack you get when this happens in Windows. 

OctoPrint won't reconnect after this no matter what you put in the connection attempts (I have 604800 set in mine)

2021-01-04 16:28:44,893 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\comm.py", line 3831, in _readline
    ret = self._serial.readline()
  File "c:\octoprint\venv\lib\site-packages\octoprint\util\comm.py", line 6455, in readline
    c = self.read(1)
  File "c:\octoprint\venv\lib\site-packages\serial\serialwin32.py", line 295, in read
    raise SerialException("GetOverlappedResult failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))
2021-01-04 16:28:44,894 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2021-01-04 16:28:44,904 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: SerialException: 'GetOverlappedResult failed (PermissionError(13, 'Access is denied.', None, 5))' @ comm.py:_readline:3831)"

WoLpH, 2022-01-27:
I have a different proposal that might be suitable for most (all?) people and could be suitable for inclusion into octoprint.

Whenever a print is sent to Octoprint through the API (such as when prusaslicer tries to send and print an upload) it could try to automatically connect to the previously connected printer.

That doesn't require any polling or unsafe stuff, it just expects that when you try to print that your previously connected printer might be available. And if it causes issues, it could still be configurable.

jneilliii, 2022-01-27:
A plugin could be made to do that, PSUControl does exactly this to power on and connect to printer when files are uploaded with the auto start print option.

WoLpH, 2022-01-27:
Yes, a plugin could. But I would argue that it's a good feature for the octoprint core. I personally don't see any downside at least and it could add quite a bit of usability if you ask me.

Instead of wondering why your print is not starting when sending it from your slicer, it will just start.

foosel, 2022-01-27:
An endpoint to upload files should upload files, period. It should not go try to connect to a printer. There is another endpoint for that, and also one for querying state, so if a "connect on upload of not connected" is what is needed/expected for the workflow, it is on the client (read, the slicer here) to implement the necessary API calls. Or it could be done with a plugin. But I will not add arbitrary side effects to established API endpoints in core.

WoLpH, 2022-01-27:
Yes, I fully agree that the upload endpoint should not connect to the printer.

However, the slicer also sends a print command and I personally think it would not be unreasonable to connect to a printer when a print command is received.

Let's consider the alternative that the slicer also has to tell octoprint to connect to the printer, in that case the logic of whether or not to connect to the printer is now part of the slicer instead. That seems the wrong place to me as all printer management should be handed off to octoprint instead (as it's directly connected and has more knowledge about the printer and it's type/settings).

foosel, 2022-01-27:
I don't agree. A print start command also shouldn't have the side effect of connecting to the printer if not connected but actually report back if it is not possible to print because of current state, be it disconnected, already printing, error, whatever. And that is what OctoPrint does. That is a matter of how the whole API is designed and also used by OctoPrint itself, separation of concerns. OctoPrint's UI doesn't even offer to start a print if the printer is not connected. Any other front-end (which the slicer in this case acts as) can do the exact same thing. I don't see why I should weaken my API contracts and introduce side effects into currently well separated endpoints when everything is already there to implement the exact behaviour you are expecting here from the slicer, by the slicer, proven by the fact that OctoPrint's own front-end relies on the very same API. Side effects btw that might also cause issues with existing clients out there. 

But again, this is about changing behaviour on the API level in core. There's nothing keeping anyone from building a plugin that hooks into the upload event and connects to the printer if not already connected, or even overrides the API endpoint's behaviour in that way. It's a core change like this won't happen.

WoLpH, 2022-01-27:
Thank you very much for the detailed reply! In that case I completely understand and agree your rationale. I was under the assumption that OctoPrint was silently accepting the print call and failing after the fact. But this indeed sounds like a mistake in the slicer implementation.

Perhaps I'll just write a plugin to handle it in that case, I doubt that prusaslicer will fix the behaviour any time soon given my previous experiences the prusaslicer developers. I'm not in a hurry though, I use my printer every few weeks/months so I run into this every time but it's not that much of a bother.

JasonKeirstead, 2022-01-27:
The way I ended up solving this was switching to Linux.

OctoPrint on Linux with PortListener handles the serial port disappearing and reappearing. On Windows it does not without a manual reconnect (see comment above)

foosel, 2022-01-27:
> @foosel RE "OctoPrint's UI doesn't even offer to start a print if the printer is not connected." - the problem is there is a bug in the Windows implementation of this. When a serial port disappears completely (as is the case if you unplug a serial connected printer from the mains to turn it off like you have to with my Ender 3), Octoprint does not detect it coming back. It goes into an error state requiring manual intervention. See above stack trace.

My comments were directed at the request to add side effects to API endpoints to force a reconnect to the printer, not the original issue on hand. 

In this thread we are currently looking at two points:

  1. Prusa serial ports just stopping to reply or half-disconnecting when powering off the printer, instead of actually behaving like an unplugged cable. Known issue, printer vendor specific so cannot be remedied by core adjustment without alienating users of different brands, thus plugin territory. Apparently worse in Windows, reconnect required in either case.
  2. Prusa slicer sending a file to print to an OctoPrint instance that's currently not connected to a printer not correctly handling that situation. Request to solve this through API side effects of the endpoints used by Prusa Slicer, denied by me quoting that clients should use the API as needed to implement workflows instead of changing API behaviour.

foosel, 2022-01-27:
> I was under the assumption that OctoPrint was silently accepting the print call and failing after the fact.

I just realized that there indeed is one situation in which this CAN happen (upload API endpoint with print parameter set, instead of upload, then select, then print), so I'll rectify this in 1.8.0.

JasonKeirstead, 2022-01-27:
Thanks... what I am saying though is #1 above is not restricted to Prusa. It will happen any time anyone is controlling a printer externally (such as via IoT devices).. it happens any time power is cut to the printer without cable being unplugged. And it seems worse in windows. It is not tied to Prusa.

jneilliii, 2022-01-27:
> I just realized that there indeed is one situation in which this CAN happen (upload API endpoint with print parameter set, instead of upload, then select, then print), so I'll rectify this in 1.8.0.

Please elaborate, this could impact several of my plugins, including Tasmota.

foosel, 2022-01-27:
@JasonKeirstead That a printer, regardless of manufacturer, disconnects usually when power is cut is normal, not a Windows or model specific bug, and actually not even a bug but completely expected behaviour, and I'm not seeing how what you are observing is any different from this. The "Permission denied" in your stack trace is a bit weird, but that rather indicates that something else was already getting a hold of the port. Windows is a finicky beast in that regard, and that's one of the reasons I recommend running OctoPrint under Linux. In any case, I'm not sure what you are asking from me to be fixed here and to be frank also start to think it isn't related to this ticket. Automatic reconnect is something I will not and can not do as it can have severe consequences. Imagine starting a 24h print from your printer's SD, OctoPrint disconnected, then you absentmindedly reattach your printer to OctoPrint, OctoPrint does an automatic reconnect on its reconnect cycle, your printer resets (because you have a model which resets) and boom, now your print is toast. Power or cable cut, printer gone, that's how things are, there's no use in trying to cover up that fact, and any such attempt will only introduce problems. 

And for the record, all that Port Lister does is that it scans the device tree in a way that's not possible to do on Windows and if it finds a new serial device pretty much hits the "refresh ports" button for you and optionally also connects if a specific device is found. In my experience that actually did NOT work with Prusa printers (see above) because those don't really disconnect but just stop responding on power cut.

@jneilliii Only changes will be on the upload API endpoint, and behaviour on the API endpoint will stay the same to stay backwards compatible, but two new response fields are introduced, `select` and `print`, which mirror the request fields and will contain what was ACTUALLY done after evaluating permissions and preconditions. The cleaner solution would be to respond with 403 or 409 respectively, but that would change API behaviour here and cause issues with existing clients. Pushing a commit in a bit that you then can check out.

jneilliii, 2022-01-27:
Ok, as long as the upload api still accepts files and those parameters from the api are still passed to the upload event callbacks I should be good. I was afraid you were going to not allow the upload with error response, but you obviously said that wouldn't be the case.

foosel, 2022-01-27:
Ah... parameters from API in the events would change in the current draft, BUT if you say that causes issues for you (and I can see how that can be), I'll change that.

foosel, 2022-01-27:
@jneilliii See 8be58b105e058b6136cb0703c3b370c394d2bbb7 - should hopefully not cause issues for you?

jneilliii, 2022-01-27:
I'll pull the updates tonight and verify, but looks like that would be ok for me as I'm triggering the select and print on the connect event after flagging it in the upload event. Might want to get feedback from @fieldOfView for his Cura plugin too, but if I remember correctly Aldo is actually using the API to connect ahead of upload, including power on, etc.

vehystrix, 2024-09-13:
So for whoever finds this issue or issue #3927 on google looking for a way to have octoprint continuously poll their serial port and connect when it's available, I had the same issue.

I wrote a plugin that polls a specific serial port for every x seconds, and if the port is available and open, it will tell octoprint to connect the printer to it.
Maybe this will work for you guys as well
https://github.com/vehystrix/OctoPrint-PortRetry

jneilliii, 2024-09-13:
You can do this with the portlister plugin I thought. You just have to save the port and set to automatically connect in the connection panel.

jneilliii, 2024-09-13:
> So for whoever finds this issue or issue #3927 on google looking for a way to have octoprint continuously poll their serial port and connect when it's available, I had the same issue.
> 
> I wrote a plugin that polls a specific serial port for every x seconds, and if the port is available and open, it will tell octoprint to connect the printer to it. Maybe this will work for you guys as well [vehystrix/OctoPrint-PortRetry](https://github.com/vehystrix/OctoPrint-PortRetry)

Any plans to register the plugin on the plugin repo? https://plugins.octoprint.org/help/registering/

vehystrix, 2024-09-14:
I registered the plugin (well I created the PR)