OctoPrint/OctoPrint #3847: M600 filament change results in line number error on Prusa MK3

Crowlord, 2020-12-02:
#### What were you doing?

General printing with a filament runout. 

#### What did you expect to happen?

M600 filament runout triggered by printer. Extruder to park and filament eject. New filament load and print continue. 

#### What happened instead?

M600 issued Extruder moves to park and printer asks for intervention. Octoprint experiences a comms error. Paste of relevant section below

Recv: fsensor_checkpoint_print
Recv: Resend: 345756
Recv: ok
Send: N345756 G1 X26.077 Y82.082 E0.03958*87
Recv: ok
Recv: fsensor_update - M600
Send: N345757 G1 X25.422 Y82.700 E0.03048*84
Recv: echo:Enqueing to the front: "M600"
Recv: Error:Line Number is not Last Line Number+1, Last Line: 345755
Recv: Resend: 345756
Recv: ok

#### Did the same happen when running OctoPrint in safe mode?

Unable to replicate reliably as this seems to be an intermittent occurrence. 

I have had 2 distinct occurences of this but this is the first i could capture. 

#### Version of OctoPrint

1.5.0

#### Operating System running OctoPrint
OctoPi 0.17

#### Printer model & used firmware incl. version
Prusa Mk3s 3.9.1

#### Browser and version of browser, operating system running browser
Windows 10 
Chrome 87.0.4280.66 (Official Build) (64-bit)

#### Link to octoprint.log
https://pastebin.com/Lwnj10TP

#### Link to contents of terminal tab or serial.log

Didn't capture full terminal tab, Now overwritten. 

Partial  https://pastebin.com/QKi9erVA

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-12-02:
Resend requests also land in `octoprint.log`, so maybe we can reconstruct the full picture of what happened here through that. Alas, a log rotation happened right during the situation it seems. Can you share the earlier `octoprint.log` as well (probably `octoprint.log.2020-12-02` or maybe `octoprint.log.2020-12-01`)?

Crowlord, 2020-12-02:
This is the previous one its octoprint.log.2020-12-02

https://pastebin.com/8wGw47Ke

foosel, 2020-12-08:
That sadly doesn't seem to contain anything on that print 🤔 There are roughly 10min missing, first one you shared started at 15:32, last one you shared ends at 15:22. The interesting bits are probably the missing ones.

Looking closer at the line number errors in the first log however, it seems to be a firmware issue.

See this annotated excerpt:

```
| Last lines in terminal:
| Send: N345762 G1 X22.473 Y83.465 E0.01172*90
| Recv: ok
| Send: N345763 G1 X22.146 Y83.162 E0.01509*82
| Recv: ok
| Send: N345764 G1 X21.600 Y82.234 E0.03646*88
| Recv: fsensor_checkpoint_print
| Recv: Resend: 345756                                                  <- firmware requests 345756 again
| Recv: ok                                                              <- this ok belongs to the resend request
| Send: N345756 G1 X26.077 Y82.082 E0.03958*87                          <- OctoPrint sends requested line
| Recv: ok                                                              <- firmware acknowledges 345756
| Recv: fsensor_update - M600
| Send: N345757 G1 X25.422 Y82.700 E0.03048*84                          <- OctoPrint sends next line
| Recv: echo:Enqueing to the front: "M600"                              <- firmware enqueues another line (M600)
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 345755  <- firmware is confused
| Recv: Resend: 345756                                                  <- and requests 345756 AGAIN,
                                                                           however OctoPrint ignores that since
                                                                           it had already send up to line 345764
                                                                           so this resend request is likely from
                                                                           the FIRST time it sent 345757 outside
                                                                           of our logged range
| Recv: ok                                                              <- this ok belongs to the resend request
| Send: N345758 G1 X24.767 Y83.100 E0.02599*87                          <- OctoPrint sends next line
| Recv: Error:No Line Number with checksum, Last Line: 345755           <- firmware is not amused
| Send: N345759 G1 X24.111 Y83.372 E0.02400*87                          <- OctoPrint sends the next line, however
                                                                           without an ok - this is weird
| Recv: Resend: 345756                                                  <- firmware requests a resend yet again
```

From my perspective, either the firmware is sending too many `ok`s, causing things to get out of sync (otherwise OctoPrint would not be sending new lines without an `ok` as visible with 345759 above, or it gets REALLY confused about what line number should be the current one. The initial resend request visible there is already really fishy since OctoPrint had already have gotten enough `ok`s to send lines up to 345764, yet now the firmware suddenly decides that 345756 wasn't properly sent, and right after injecting an `M600` into the command queue. My money is on this `M600` being injected without a line number, or a wrong line number, the firmware thinking it came from OctoPrint/serial and thus requesting a resend of **it**, triggering this whole resend request cascade, which at the end makes OctoPrint realize something is going very wrong here and it is running in circles with the firmware, and thus aborts the print as all its attempts to satisfy the firmware's resend requests only trigger the same one again and again and thus things are stuck in an infinite loop:

```
2020-12-02 15:33:04,583 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 345756, current line = 345765
| Last lines in terminal:
| Recv: Error:No Line Number with checksum, Last Line: 345755
| Send: N345758 G1 X24.767 Y83.100 E0.02599*87
| Recv: Resend: 345756
| Recv: ok
| Send: N345756 G1 X26.077 Y82.082 E0.03958*87
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 345755
| Recv: Resend: 345756
| Recv: ok
| Send: N345757 G1 X25.422 Y82.700 E0.03048*84
| Recv: Error:No Line Number with checksum, Last Line: 345755
| Send: N345758 G1 X24.767 Y83.100 E0.02599*87
| Recv: Resend: 345756
| Recv: ok
| Send: N345756 G1 X26.077 Y82.082 E0.03958*87
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 345755
| Recv: Resend: 345756
| Recv: ok
| Send: N345757 G1 X25.422 Y82.700 E0.03048*84
| Recv: Error:No Line Number with checksum, Last Line: 345755
| Recv: Resend: 345756
2020-12-02 15:33:04,627 - octoprint.util.comm - WARNING - Printer keeps requesting line 345756 again and again, communication stuck
2020-12-02 15:33:04,627 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: Printer keeps requesting line 345756 again and again, communication stuck"
```

Without a full `serial.log` of a complete occurrence of this problem I cannot be certain of this, but the current information strongly suggests this to be a firmware bug.

I'm therefore closing this for now and would advise you to report this to your firmware vendor if not already done. If you manage to trigger it again *with* a `serial.log`, be free to share that here and I'll take another look and if need be reopen. 

You could also see if increasing `serial.maxConsecutiveResends` in config.yaml to a larger value (default is 10) helps to maybe get things back in sync, try 20.

``` yaml
serial:
  maxConsecutiveResends: 20
```

or via command line (on OctoPi run `source ~/oprint/bin/active` first): `octoprint config set --int serial.maxConsecutiveResends 20`

closed 2020-12-08 by foosel