OctoPrint/OctoPrint #3849: Plugin Manager Not working - Can't install any plugin

Matrixma, 2020-12-02:
#### What were you doing?

Trying to configure a new camera on octolapse, but all the buttons and text box on the forms were as dicribed in this issue: https://github.com/OctoPrint/OctoPrint/issues/3842 

I unnistall the octolapse plugin to reinstall it and can't install none of the plugins.
When I press the button "Install", open a box with the title "Installing plugin..." and the text: "Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip..." (see picture below)

Ande it happens with all the plugins.

I did the update to 1.5.0 today and that happened after the update, before the update everything was working great.

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
octoprint.version: 1.5.0

#### Operating System running OctoPrint
env.plugins.pi_support.octopi_version: 0.17.0

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.66 Safari/537.36

#### Link to octoprint.log
https://pastebin.com/h5AZ6tkg

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
https://prnt.sc/vunm7n

#### Screenshot(s)/video(s) showing the problem:
[image]
https://prnt.sc/vunmot

I have read the FAQ.

foosel, 2020-12-03:
Before I can look into this, you first need to take care of this:

```
2020-12-02 22:32:25,432 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

Problems not being able to install stuff, losing connectivity and a multitude of other issues are all possible side effects of a Pi running in undervoltage mode. Which is why OctoPrint will also have displayed you a big red warning icon in the navbar.

See [this FAQ entry](https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256?u=foosel).

Fix the undervoltage, see if you can still reproduce the issue, if so see if you can still reproduce **in safe mode**, and only if SO provide a new `octoprint.log` and also `plugin_pluginmanager_console.log`. Otherwise close this.

foodbandlt, 2020-12-09:
Hey, just wanted to chime in on this.  Setup a new Octopi install on a RasPi 3B and ran into this same issue.  Log files looked the same, undervoltage and also 

`WARNING - Could not delete temporary directory /tmp/tmpwrFSBs: [Errno 21] Is a directory: '/tmp/tmpwrFSBs'` 

Looks like that's where it downloads the plugin repo?  I SSH'd and went into the tmp folder and was able to delete it with `rm -r /tmp/tmpwrFSBs`  (sub actual folder name in of course) without any problem, so not sure what Octoprint's problem is.  Seems to be the only error appearing in the log related to this.

Solved by restarting the Pi several times.  No config or hardware changes made.  Still looking for literally any power supply that doesn't trigger undervoltage, but it's suddenly working now.  No "Could not delete temporary directory" error anymore, still have undervoltage errors though.  Deleting the tmp folders manually still resulted in the same issue after, got the same error the very next time I tried to install a plugin. 

During troubleshooting I also checked all of the points mentioned [on this FAQ post](https://community.octoprint.org/t/i-cant-install-any-plugins-from-the-repository-it-is-unreachable/178) and everything seemed to be in place.  Wasn't working until I restarted the Pi several times though.

I can't say that this was 1.5.1's fault of course, but my 1.4.2 install on another pi doesn't experience this problem.  Kind of afraid to update that one now for fear of breaking plugins lol.

And... then I restarted again and now I can't install plugins again lol!  I'll try to find a power supply that doesn't cause undervoltage just to see if that helps, but after that restart there's no undervoltage warnings in the logs, only another `Could not delete temporary directory` right when I tried installing a plugin.  The only thing in these tmp folders is ZIP of the repo for the plugin.  Tried installing the plugin a few more times without restarting and it worked on the 5th try.  This seems to random, so perhaps it is an undervoltage issue?

foosel, 2020-12-09:
Can't say it is, can't say it isn't. Experience over the years has shown though that once the pi is reporting undervoltage, the system should be considered very unstable and the problem needs to be fixed before or makes sense to investigate any kind of weirdness observed, as chances are high it's due to flakiness.

I've had good experiences with the official power supply.

MikePfaff, 2020-12-12:
I am experiencing the same or similar problem on a **Raspberry Pi 3 Model B Rev 1.2** running a completely **fresh install of OctoPi 0.17.0** (freshly flashed IMG to SD card). After the fresh install i did update to **OctoPrint 1.5.2** (through the web interface) and then reboot the Raspi. I then had to try out various power sources with reboots inbetween to get rid of the undervoltage problem report. The python version reported in the webinterface is **Python 2.7.16**.

I then tried to install the **Bed Visualizer** plugin through the webinterface, but that always gets stuck with the message
> Installing plugin "Bed Level Visualizer" from https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/master.zip...

in the webinterface. I also tried to install the **NavBar Temp** plugin, but that gets stuck in the same way, just a different URL.

As soon as i try to start a plugin install, the following entry appears in **octoprint.log**:
> 2020-12-12 19:46:53,622 - octoprint.util - WARNING - Could not delete temporary directory /tmp/tmp_5L4kx: [Errno 21] Is a directory: '/tmp/tmp_5L4kx'

On subsequent tries, the error stays the same, but the reported path to the tmp directory is different every time. The reported tmp directory exists and contains the plugin ZIP file. The plugin ZIP file is complete and exactly the same (verified with sha512sum) as the master.zip of that plugin on GitHub. 

I **don't have any reported undervoltage problems** (so far) running on a **Rigol DP832 lab bench power supply** (Output voltage set at 5V, max. current at 3.2A / Overvoltage protection at 5.5V / Overcurrent protection at 3.3A). I thus suspect that the issue is not related to any power supply problems.

The following command line installation **workaround**, when logged in via SSH, works for me:

> cd ~/oprint/
> source bin/activate
> pip install "INSERT_PLUGIN_INSTALL_URL_HERE"
> sudo service octoprint restart

On a sidenote:
I actually first encountered the issue on that same Raspi 3B which had an older OctoPi image which i manually (apt-get update|upgrade|dist-upgrade|etc.) upgraded to Raspbian "buster" and where i then installed OctoPrint 1.5.2 with python3 freshly (but over the old OctoPrint instance) as described in https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian-or-raspberry-pi-os/2337 . After i ran into the plugin install issue, i decided to start again with a completely fresh OctoPi image to rule out any problems caused by the old image and/or the manual upgrade. But it looks like this was not the case.

cp2004, 2020-12-12:
@MikePfaff could you upload the octoprint.log & plugin_pluginmanager_console.log so that we can review? Thanks.

Only explanation for this so far has been undervoltage, and I would like to check if there is something new in your environment that could be causing it.

MikePfaff, 2020-12-12:
@cp2004 When i was about to prepare the octoprint.log for upload i actually discovered that i once again had undervoltage problems reported. So i've had it with this mystery and took a 2mm^2 (14 AWG) cable to directly connect the power supply to the RPi GPIO pins. On the RPi side i had to cut away about 50% of the diameter (resulting in about 1mm^2 actual copper connection) so that i could crimp on the dupont connectors. The cables are so thick, i cannot even fit the connector into the housing.

[image]

I then started out with a fresh OctoPi image (fresh flash of IMG to SD card) and can now report that **plugin installation was successful**.

This surely was an interesting issue to track down (the term Heisenbug comes to mind ;) ), but i still wonder why it always fails at the same task (plugin installation)? I would really have expected power issues to cause "random" problems and not be so reproducible. While I tried to enable logging on the power supply, its maximum logging resolution is only 1 sample per second, so the reported maximum current draw of 0.5A among the sample set is not surprising (and might actually be much higher inbetween).

Also, i still wonder why everything worked well in previous versions of OctoPrint and what is happening differently (e.g. what code causes a spike in current or power draw) in newer versions that suddenly causes those problems to appear. But of course i totally understand that from a developer point of view the case here is clear: power problems = user error due to crappy equipment. 

To summarize it also for other users (and feel free to link to this):
> **Undervoltages are a real issue. Save yourself the time and headache and get the best power supply and the thickest USB/power cable that you can find.**

cp2004, 2020-12-13:
So I can explain a couple of the points here, and am 90% sure we have come across a bug in the process.

OctoPrint is supposed to actively prevent plugin installation or updates on a throttled system. This is probably why it fails for you, since it never starts in the first place. I'll come on to why later. Installing things while a system is throttled has caused a number of issues in the past, so now this is actively prevented to save problems further down the road. So yes, power issues cause random problems but we try and make them less random by preventing the causes.

The change in 1.5.0 is that plugin installation is now done async, to solve issues with requests timing out in a very long plugin install (think compiling numpy for bed level vis, and others). This is where I reckon the bug has come in, the undervoltage state is not being reported back to the frontend properly, and something in the server is then preventing plugin install. So to the user on the web UI, it looks as if it is hanging. This bit needs fixing.

Hopefully that clears up some of the mystery around this, time to go and dig through to find out where the undervoltage warning is not being evaluated properly.

cp2004, 2020-12-27:
Hopefully fixed by PR #3912. The various error handling was not being properly reported to the UI - in OctoPrint 1.6.0 (pending PR acceptance) this will be fixed.

cp2004, 2021-01-14:
There are still a lot of issues with this. Is it worth a hotfix release?

foosel, 2021-01-15:
Might actually be, yeah... Will think long and hard about it over the weekend. Not going to push one today, have learned the hard way that releases just before the weekend are a bad idea unless absolutely necessary ;)

closed 2021-01-19 by foosel

Nathan-Blocks, 2021-09-29:
where do you find whether or not you are getting undervoltage errors?

jneilliii, 2021-09-29:
> where do you find whether or not you are getting undervoltage errors?

octoprint.log will show those errors.

Nathan-Blocks, 2021-10-06:
I looked there and I couldn't find any undervoltage errors but then when I replaced the power supply it downloaded plugins no problem

PR #3912 (2021-01-11): Fix plugin install errors not being reported to the frontend
When plugin install went async (4cf2a673c8860304472904cf11d13e2187bc17fd) not all of the responses were caught. This PR changes the rest of the responses to become asynchronous as well.

Closes #3849

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [x] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the maintainers, e.g. through a
    Brai [... cut]
commit b807eb2d (2021-01-11): :bug: Fix plugin install errors not being reported to the UI

When plugin install went async (4cf2a673c8860304472904cf11d13e2187bc17fd) not all of the responses were caught. This PR changes the rest of the responses to become asynchronous as well.

Closes #3849
commit cacc329a (2021-01-19): :bug: Fix plugin install errors not being reported to the UI

When plugin install went async (4cf2a673c8860304472904cf11d13e2187bc17fd) not all of the responses were caught. This PR changes the rest of the responses to become asynchronous as well.

Closes #3849