OctoPrint/OctoPrint #3850: When UI Customizer enabled, screen is blank.

holocronology, 2020-12-03:
#### What were you doing?
After login, screen is blank, info bar at top still present.

When UI Customizer plugin is active, screen is blank. All other 3rd party plugins deactivated.

Safe Mode

#### What did you expect to happen? Normal screen.

#### What happened instead? White Screen.

#### Did the same happen when running OctoPrint in safe mode? Yes.

#### Version of OctoPrint octoprint.version: 1.5.0

#### Operating System running OctoPrint  linux

#### Printer model & used firmware incl. version na

#### Browser and version of browser, operating system running browser browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2020-12-03:
This is an issue with that plugin and should be reported on that plugin's repository not here. I have received reports of the same thing when it is installed in conjunction with my Bed Level Visualizer plugin and the plugin author is investigating on his end what might be causing that.

cp2004, 2020-12-03:
As @jneilliii said, this is a 3rd party issue, there is nothing we can do about that here. And in the future, please fill out the template when you create a bug report.

closed 2020-12-03 by cp2004

holocronology, 2020-12-03:
Okay, thanks. I assumed since it was installed as part of the update that it was part of the base.

cp2004, 2020-12-03:
> I assumed since it was installed as part of the update

Wasn't related, it was probably coincidence since it was released at a similar time on the repository, you got the 'new plugin available' and 'octoprint update' notices at the same time.

holocronology, 2020-12-04:
Right, now that you mention it. Odd thing is that even after deleting all the other plug-ins, it still misbehaves in the same manner.