OctoPrint/OctoPrint #3851: After version 1.5.0 update the web interface does not display on ipad 2

huseman21, 2020-12-03:
#### What were you doing?

1. Go to '...'IP address of octopi
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
the web page load.
#### What happened instead?
just a blank white page.  It does make the connection.  it does  work on my pc.
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-03:
The issue template is supposed to be filled in **fully**. If this information is not provided within the next 2 weeks then this issue will have to be closed. Please fill it in completely.

cp2004, 2020-12-16:
Since the rest of the information requested in the report has not been provided, I'm going to close this.

@huseman21 if this is still an issue, feel free to provide the requested information and we will decide if it needs re-opening.

closed 2020-12-16 by cp2004