OctoPrint/OctoPrint #3852: [1.5.0] Connection via octopi.local no longer works, IP works

foosel, 2020-12-03:
### Problem

Reported on the forums [here](https://community.octoprint.org/t/cannot-connect-with-local-name-only-with-ip-address-after-1-5-0-update/26907/22).

What we currently know:

  * The symptom is that the `.local` domain resolves to 127.0.0.1
  * It is related to something in 1.5.0 as the issue goes away when rolling back to 1.4.2.
  * For Spybyte it can be fixed by disabling the discovery plugin (which was switched from `pybonjour` to `python-zeroconf` in 1.5.0 and takes care of *service* discovery (not hostname resolution)). For audryhome, disabling said plugin however does not change behaviour
  * For audryhome, there's also a network error during initialization of the discovery plugin and a negative connection check, hinting at some issue there
    ```
    2020-12-01 13:44:28,476 - octoprint.vendor.zeroconf - WARNING - Exception occurred:
    Traceback (most recent call last):
      File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/vendor/zeroconf.py", line 2237, in send
        bytes_sent = s.sendto(packet, 0, (addr, port))
    error: [Errno 101] Network is unreachable
    2020-12-01 13:46:00,663 - octoprint.server.heartbeat - INFO - Server heartbeat <3
    2020-12-01 13:46:07,061 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from online to offline
    2020-12-01 13:46:07,062 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
    2020-12-01 13:46:07,063 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
    2020-12-01 13:54:12,221 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.42
    2020-12-01 13:34:09,097 - octoprint.startup - INFO - ******************************************************************************
    2020-12-01 13:34:09,099 - octoprint.startup - INFO - Starting OctoPrint 1.5.0
    2020-12-01 13:34:09,100 - octoprint.startup - INFO - ******************************************************************************
    2020-12-01 13:34:15,914 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: offline
    2020-12-01 13:34:15,915 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
    2020-12-01 13:34:15,918 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
    ```
  * The issue is unrelated to Python 2 vs 3 (which is good to know because Python 2 uses a vendored version of `python-zeroconf`) 
  * Both reporters run OctoPi 0.17.0

Further weirdness [reported by Spybyte](https://community.octoprint.org/t/cannot-connect-with-local-name-only-with-ip-address-after-1-5-0-update/26907/28?u=foosel):

> another thing I just found out. My OctoPi's name is mk3s.local. If the plugin is disabled, I can ping the server. When I enable the plugin mk3s.local will be resolved to 127.0.0.1. So far nothing new. But with the activated plugin I have a "new" hostname mk3s-2.local which has the proper IP.
> 
> So with disabled plugin:
> 
> ping mk3s.local -> proper IP
> ping mk3s-2.local -> host not found
> 
> with enabled plugin:
> 
> ping mk3s.local -> 127.0.0.1
> ping mk3s-2.local -> proper IP
> 
> Maybe this helps.
> 
> Can someone please verify this?

[audryhome](https://community.octoprint.org/t/cannot-connect-with-local-name-only-with-ip-address-after-1-5-0-update/26907/36?u=foosel) could solve the issue by switching to wired connection, resetting the wifi credentials, then reconnecting on wifi:

> I also apparently solved the problem by connecting directly with LAN (dhcp), then editing the /etc/wpa_supplicant/wpa_supplicant.conf to erase the wifi passwd, then rebooting (same outcome) then reestablishing the password then rebooting again in wifi and IT SEEMS TO WORKS NOW.
> 
> Would be interested if you can do the same and check the outcome.....
> 
> Will test all the day with a octoprint.log monitoring through tail -f ~.octoprint/logs/.... and will report if any issue.

### Background information

In 1.5.0, the discovery plugin switched from using the abandoned and py3 incompatible `pybonjour` to the maintained [`python-zeroconf`](https://pypi.org/project/zeroconf/). For py2, an older version of that dependency is vendor-bundled as the current version is py3-only.

Spybyte, 2020-12-03:
I found a solution, that worked at least in my case. On the Pi my LAN configuration is as follows:

```
pi@mk3s:~ $ ifconfig 
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:27:9e:2d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 4465  bytes 9124866 (8.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4465  bytes 9124866 (8.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.85  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fd5e:854e:e8b4:1:6479:4557:94d5:bb20  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::3e5f:a481:61fb:58c3  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:72:cb:78  txqueuelen 1000  (Ethernet)
        RX packets 26906  bytes 6127709 (5.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6695  bytes 6473416 (6.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
So my Wifi connections is last in this list.

In config.yaml I added the wifi interface explicitly:
```
discovery:
     interfaces: wlan0

```
It might be, that there is something wrong with the order of the interfaces in the discovery plugin.

cmburg, 2020-12-03:
I'm not sure if it's right here but I try:
when I load octopi.local on a webbrowser it says "server not found" or "server offline" even when I was able to connect to it a few minutes prior.
Testing the connection with prusaslicer (2.2.0 and 2.3.0 alpha 3) I often get 

> Could not connect to OctoPrint: Couldn't connect to server:
Failed to connect to octopi.local port 80: Verbindungsaufbau abgelehnt
[Error 7]
Note: OctoPrint version at least 1.1.0 is required.

ssh'ing to it refuses connection (seems to be something in between on-/ offline since I got 'no route to host during a reboot of the pi')

however: simply refreshing the page (firefox), hitting try again (slicer/ ssh) seem to work fine.

I am running pfsense as a router automatically assigning .local as a domain name and IIRC it says something about bonjour issues there.

foosel, 2020-12-03:
> I am running pfsense as a router automatically assigning .local as a domain name and IIRC it says something about bonjour issues there.

In that case you are probably looking at a fight between your router's DNS and bonjour (which is an implication of multicast DNS). I'd strongly suggest to not use `.local` as local subdomain for your LAN due to exactly this kind of namespace clash. Personally I use `.lan` to avoid issues.

foosel, 2020-12-03:
@Spybyte quick follow-up question to that...

> ```yaml
> discovery:
>      interfaces: wlan0
> ```

Really like this, not like this?

``` yaml
discovery:
    interfaces:
    - wlan0
```

The former should basically nuke zeroconf service registration - a list is expected here, you gave it a single string, so what will happen is it will iterate over each character (`w`, `l`, `a`, `n`, `0`), try to fetch the network config for that "interface", fail, continue, and in fact never do anything. So basically, this is the same as disabling the discovery plugin altogether, at least with regards to zeroconf service announcements.

But that means, whatever is going wrong there on your end is in fact rooted in the zeroconf service registration. I'll try to see if maybe there has been something about this upstream. 

Would you in principle be open to running a development branch to test something out for me? I might be able to add some more logging that could shed some more light on this, or maybe experiment with some arguments for the zeroconf library. I sadly still cannot reproduce, so this would help a ton.

foosel, 2020-12-03:
As a follow-up to that, it would be interesting to know how behaviour is with 

```yaml
discovery:
    interfaces:
    - wlan0
```

foosel, 2020-12-03:
Oooooh, interesting! I think I can reproduce by setting

``` yaml
discovery:
    interfaces:
    - lo
```

so forcing zeroconf registration on the loopback device only:

```
gina@nuc-ubuntu:~$ dig -p 5353 @224.0.0.251 octopia-2.local

; <<>> DiG 9.16.1-Ubuntu <<>> -p 5353 @224.0.0.251 octopia-2.local
; (1 server found)
;; global options: +cmd
;; Got answer:
;; WARNING: .local is reserved for Multicast DNS
;; You are currently testing what happens when an mDNS query is leaked to DNS
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 63305
;; flags: qr aa; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;octopia-2.local.               IN      A

;; ANSWER SECTION:
octopiA-2.local.        10      IN      A       192.168.1.184

;; Query time: 224 msec
;; SERVER: 192.168.1.184#5353(224.0.0.251)
;; WHEN: Do Dez 03 17:09:02 CET 2020
;; MSG SIZE  rcvd: 59

gina@nuc-ubuntu:~$ dig -p 5353 @224.0.0.251 octopia.local

; <<>> DiG 9.16.1-Ubuntu <<>> -p 5353 @224.0.0.251 octopia.local
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached

gina@nuc-ubuntu:~$ ping octopia.local
PING octopia.local (127.0.0.1) 56(84) bytes of data.
64 bytes from localhost (127.0.0.1): icmp_seq=1 ttl=64 time=0.041 ms
64 bytes from localhost (127.0.0.1): icmp_seq=2 ttl=64 time=0.035 ms
^C
--- octopia.local ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1011ms
rtt min/avg/max/mdev = 0.035/0.038/0.041/0.003 ms
```

Which means, it should in theory work with 

```yaml
discovery:
    interfaces:
    - wlan0
```

Now I'm *really* keen on hearing if that solves it.

Spybyte, 2020-12-03:
@foosel as soon as my current print is finished I will try "- wlan0". I was not aware that "wlan0" how I used it is indeed a character array. 

And yes, of course I'm open to install a dev version or to help you figure out that problem by doing what ever is needed.

jneilliii, 2020-12-03:
This is the type of cooperation/collaboration that makes this community great, we appreciate your interaction @Spybyte.

foosel, 2020-12-03:
@Spybyte awesome! If it works, if you could try the `staging/hotfix` branch without the config changes that would be splendid. As far as I understand what might be going on here, that should hopefully fix it.

Spybyte, 2020-12-04:
Hey, 
today I will install the branch and give feedback.
For now  I already tested the proper syntax
```
discovery:
    interfaces:
    - wlan0
```
and it worked on my machine.

Spybyte, 2020-12-04:
Good new. At least in my case the hot fix worked!
```
------------------------------------------------------------------------------
OctoPrint version : 1.5.1.dev4+g82dd94d51
OctoPi version    : 0.17.0
------------------------------------------------------------------------------
```
```
PING mk3s.local (192.168.1.85): 56 data bytes
64 bytes from 192.168.1.85: icmp_seq=0 ttl=64 time=9.984 ms
64 bytes from 192.168.1.85: icmp_seq=1 ttl=64 time=9.207 ms

```
I also tested it with PrusaSlicer. Perfect 👍

foosel, 2020-12-04:
Perfect, that'll have to do then. I'd like to get the hotfix out today to still make it before the usually print and update heavy weekend (even though it freaks me out to be releasing code on a Friday) and it doesn't look like anyone else is stepping up to verify. So, buckle up, 1.5.1 incoming, let's hope for the best 😅

foosel, 2020-12-04:
1.5.1 is released.

closed 2020-12-04 by foosel

SirSean26, 2020-12-04:
Please excuse my noob-ery. 

I experienced this issue where the octopi.local was not showing up after update. I'm trying to update to 1.5.1; I found my IP but was not able to pull up the octopi interface to reach it. I currently have it set up so that my raspberrypi boots directly into the desktop UI. 

Is there a way to (1) prevent the desktop UI from auto-loading, and then (2) run a command to update to 1.5.1?