OctoPrint/OctoPrint #3855: Deleting existing filament temperature settings does not work.

MartinMajewski, 2020-12-04:
Version: 1.5.1

Issue description:

In "Settings" → "Temperatures" → "Preset" Section.
When deleting an existing temperature preset and saving the dialog, the changes are not applied.

Adding a new material, or changing existing materials works. As soon as new material is added, it cannot be removed anymore.

The "bug" was possibly introduced after 1.4.2, as it worked there. I've upgraded directly to 1.5.1 today and discovered this behavior.

I will provide logs shortly!

I have read the FAQ ;-)

foosel, 2020-12-04:
@OctoPrint/octoprint-triage I've told @MartinMajewski to open an abridged bug report, he's reported this in length via chat.

foosel, 2020-12-04:
[image]

[image]

[image]

foosel, 2020-12-04:
Can reproduce. Weirdly enough I can only reproduce if I try to delete every custom added temperature.

foosel, 2020-12-07:
Fixed by the above commit, and this ticket actually turned out to be a general issue with how array settings are being processed in the frontend, see #3867 for details. I'll look into providing a quick workaround plugin for this.

foosel, 2020-12-07:
This plugin should work as a fix for now: https://github.com/OctoPrint/OctoPrint-FixSettingsDeepMerge

Install via plugin manager, get more, from this URL: https://github.com/OctoPrint/OctoPrint-FixSettingsDeepMerge/archive/master.zip

foosel, 2020-12-10:
About to release 1.5.2 after all (third party dependency released a breaking update), so this fix gets a hotfix release after all.

commit ade3a772 (2020-12-07): 🐛 Replace _.merge with object only deepMerge

It turns out _.merge will also merge arrays, which in turn will re-add
anything we've just removed from a setting that happens to be an
array, thus as blocked commands or temperature presets.

Fixes #3855
Fixes #3867
commit 12d6de51 (2020-12-10): 🐛 Replace _.merge with object only deepMerge

It turns out _.merge will also merge arrays, which in turn will re-add
anything we've just removed from a setting that happens to be an
array, thus as blocked commands or temperature presets.

Fixes #3855
Fixes #3867
closed 2020-12-10 by foosel