OctoPrint/OctoPrint #3857: Streaming speed reduction while printing files stored on raspberry when switching from 1.4.2 to 1.5.1 results in printer to stutter while printing

caesar1111, 2020-12-04:
What changes were mad to the working installation?
- upgrade from 1.4.2 (stable) to 1.5.1 (stable)

How does is the problem exposed?
- when printing on higher speeds (>100mm/s) the print process starts to stutter when a fast gcode sequence is sent to the printer.
- as a testprint I used a christmas tree in vase mode at 100mm/s
[image] left tree is printed with 1.5.1 (stable) right tree is printed on the same setup having rolled back the installation to 1.4.2 (stable) 

What hardware is used?
- printer: SK-Go² (BTT SKR 1.4 Turbo board with BTT S42B closed loop steppers for X&Y and TCM2209 driven Nema 17 for the extruder and Z
- octoprint server: raspberry pi 3B+ with 7inch raspberry touch display and PiCam

Which addons are used?
- see included octoprint backup
- also the TouchUI addons are installed to display the octoprint on the 7inch touchscreen
[octoprint-backup-20201201-191119.zip](https://github.com/OctoPrint/OctoPrint/files/5645779/octoprint-backup-20201201-191119.zip)

What testing was already conducted:
- Baudrate was increased to 250000 => no effect on the stuttering 
- plain installation of octoprint, backup of the config and updating to 1.5.1 => no effect on the stuttering
- replacing complete hardware with another pi 3B+ without display and Picam => no effect on the stuttering
- copying the file on the SD card and print directly form the SD card  => stuttering stopped
- print speed was reduced to 60mm/s => stuttering stopped
- rollback to 1.4.2 while keeping print speed at 100mm/s => stuttering stopped

Current conclusion:
- updating to 1.5.1 while suing older hardware (pi 3B+) will result in stuttering while printing complex geometry at high speed

I hope I was able to summ up the issue for you to provide a staring point to hunt down the bug.
Just contact me, if you need any more details

Regards

cp2004, 2020-12-04:
Interesting one. The only changes in the comm layer were resend ratio tracking, which has an incredibly small impact since it just counts the number of lines, and only calculates something new on a resend request.

While you may find it unrelated, please provide **all the information requested in the template** - maybe your log file is flooded with errors, causing performance issues, we just don't know. The printed hours for 1.5.0 are already in the 100s of years I would think (RC phase was 3.4 years alone) so I don't believe this can be a widespread issue. Likely some specific configuration, or firmware that is not playing nice with whatever you have setup.

robinm8, 2020-12-05:
I noticed the same issue on my instance when I updated to 1.5.0 on my RPi 3B. I was still running OctoPi with Python 2.

The following worked for me to "fix" the issue on my instance: 
1. Migrated OctoPi to Python 3 via https://octoprint.org/blog/2020/09/10/upgrade-to-py3/
2. Taped the 3D printer USB power +5V line (Prevent 3D printer board powered by both the RPi and the 3D printer power supply)
3. Overclocked the RPi 3B CPU
    - force_turbo=1
    - boot_delay=1
    - arm_freq=1375
    - core_freq=525
    - over_voltage=3
    - disable_splash=1

After that, my OctoPrint instance was snappy to load and this issue was no longer observed for me yet. (YMMV)

cp2004, 2020-12-06:
@caesar1111 - can you complete the requested information?

In addition, I'd just like to point out:
> **Link to contents of Javascript console in the browser**
not applicable since the problem is not occurring with the same browser on 1.4.2

This does not make it not applicable, since OctoPrint has updated. The javascript has changed, and therefore the log will. If you can, please add this. Maybe the root cause is something in the frontend causing the server unnecessary load.

And @robinm8 - was it *all* of those things combined, or just one thing that fixed the issue? Need to know from @caesar1111 which Python version they were running, to rule that one out (see the information requested...). Might also be useful if you still have the logs from when this *was* an issue, or if you can undo those steps 1 by 1 (Python can be done if you haven't deleted the backup) and test this please.

Overclocking it solving things *hints* at general performance issues, however I have not been able to observe any myself yet, having completed multiple prints with curves this weekend, that often cause stuttering. I have in the past seen this, when something like the touchscreen is loading, so I know my setup is not immune to stuttering issues from external factors.

Lisheng2016, 2020-12-07:
Same issue on both of my pi3B+.The problem started with version 1.5.0.Once i rolled back to v1.4.2,the problem was gone.

I'm still on python2.Will test if python3 version still stutters.

caesar1111, 2020-12-07:
@cp2004 
OK, here we go with the missing information:
### python version:
python version 2 as provided by octopi 0.17

#### Browser and version of browser, operating system running browser
Firefox 83.0
on 
Windows 10

#### Link to contents of Javascript console in the browser
https://gist.github.com/caesar1111/6c4ba495aad7c10ad3303b5a6d85b453

### Link to contents of terminal tab or serial.log

#### Link to octoprint.log
https://gist.github.com/caesar1111/564ed91264f827dbc65891c9f54f4b0b

#### additional octopi logs
https://gist.github.com/caesar1111/a943be000e7af8a462026415a9871a81
https://gist.github.com/caesar1111/aac5b0e3c2fff4d0f5c257828518ae0b
https://gist.github.com/caesar1111/f3a09494c473fb14509f8f66b7b13a62

#### gcode file for testprint
[SK-Go2_PLA_0h27min_5g_LEDChristmastree_120mms.zip](https://github.com/OctoPrint/OctoPrint/files/5652810/SK-Go2_PLA_0h27min_5g_LEDChristmastree_120mms.zip)

#### video of printer stuttering

[printing.zip](https://github.com/OctoPrint/OctoPrint/files/5652830/printing.zip)

#### picture of print results with 

[image]

.......let me know if you are using any other information.

PS:
I was also testing further impacts of my printer hardware (I am running closed loop S42Bs for X and Y). I physically changed the mode of the closed loop boards to open loop, making it a standard A4988 stepper driver setup => no effect, still stuttering. 
Also I was rigging my CR-10s Pro with a spare 3B+ pi, without the 7inch display and at 100mm/s I have no stuttering at this printer.....

foosel, 2020-12-07:
@caesar1111 

```
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

Your Pi is throtteling due to undervoltage. Its power supply is insufficient. You need to fix that, that might very well be already what's causing this.

> without the 7inch display and at 100mm/s I have no stuttering at this printer.....

So the one stuttering *has* a display? With what running on it?

Lisheng2016, 2020-12-08:
> Same issue on both of my pi3B+.The problem started with version 1.5.0.Once i rolled back to v1.4.2,the problem was gone.
> 
> I'm still on python2.Will test if python3 version still stutters.

just upgraded to python3 and uninstalled octopod plugin.The problem is gone.I don't know which one actually solved it.Try upgrade python version first if you are experiencing same issue first.Even if that was no the cause,python2 has reached its EOF and it's time to upgrade.

caesar1111, 2020-12-08:
@foosel 
Thks for the lead. This was really the problem! 
So to answer your question:
Yes, my setup includes a raspberry 3B+ with the 7 inch screen connected, both supplied with a 5V 3A power supply:
[image]

This touch display is used to run the TouchUI interface:
[image]

I was observing one or two flashes of the the under voltage spark with 1.4.2, but only while booting up the pi and not while printing and therefore no stuttering while printing. So I thing the setups was somehow just below the tipping point. Now with 1.5.x it seems just to be above the tipping point and also shows the under voltage spark while printing if having to process rapidly lines of gcode.
That also explains the reason it is not occurring on my CR-10s Pro, which also runs a pi 3B+ but no display while being powered by a dedicated Meanwell RS25-5.

My solution is: hook up the pi/display to a dedicated Meanwell RS25-5. 

I just hope that this also will help other with the stuttering issue.....

And one final question out of curiosity:  When I copied the gcode on a SD Card, inserted the card in the printer, selected the file in the file dialog in octorprint and printed it, I had no issues.....
Does this mean octoprint is just "remotly" controlling the print and the data stream is directly form the SD card to the printer controller?
Or is octoprint reading the file remotely form the SD card reader of the controller instead of reading it from its own SD card? If thats the case, it would mean that the reading of local SD card information of the raspberry is using way more power than receiving the information via USB......

Thanks again for the fast support!

jneilliii, 2020-12-08:
> Does this mean octoprint is just "remotly" controlling the print and the data stream is directly form the SD card to the printer controller?

That is correct.

foosel, 2020-12-11:
Since all of the people chiming in here can no longer reproduce this, I'm closing it.

closed 2020-12-11 by foosel

caesar1111, 2020-12-11:
@foosel Hi, thks for the support. 

So even if it is a hardware issue and not a bug in octoprint, the community might be informed that due to the little bit more of power consumption when running the newer versions their installed power supplies will no longer be sufficient. 
In my case  running a raspberry pi 3B+ with standard raspberry 7inch touchdisplay installing the Meanwell RS25-5 (capable of 5A) and cranking up the voltage to a max of 5.8V I still have observing undervoltage. I had to hook up a second power supply to the display to get rid of my undervoltage

foosel, 2020-12-11:
> the community might be informed that due to the little bit more of power consumption when running the newer versions their installed power supplies will no longer be sufficient

I mean, that's precisely why OctoPrint shows you a flashing warning symbol if there is an undervoltage issue, unless you've decided to disable the Pi Support plugin because "the undervoltage warning lies" (yes, I've heard exactly that from people). And this will apparently only happen if the power supply was barely cutting it to begin with - if it was suitable from the get go there would not be an issue after an update. I've never seen this happen on any of my own instances, only in the field 🤷🏻‍♀️

GeoCyberwolf, 2020-12-22:
I had this issue in 1.5.2 also. No Pi power supply problems, no undervoltage warnings and the Pi is plenty powerful. It's a Pi 4 4GB powered with a waveshare Power over Ethernet hat.

In 1.5.2 octoprint would cause the printer to stop and stutter during the print causing a degredation in print quality. When this happens, the temperature graph in octoprint drops to zero while the printer stutters.

After downgrading to 1.5.1 I do not have this issue so I think this is a bug introduced in 1.5.2.

cp2004, 2020-12-22:
@GeoCyberwolf I don't think it *can* be a bug introduced in 1.5.2, since if you read the [release notes](https://github.com/OctoPrint/OctoPrint/releases/tag/1.5.2) or compare changes there are only two changes:

* Pinned a dependency since it released a broken version that dropped Python 3 support (obviously that won't work, needed pinning quikly)
* Fixing an issue with saving a setting in the frontend. 

Not denying you have an issue, however I am suggesting that it might be coincidence.

This part of your comment:
> When this happens, the temperature graph in octoprint drops to zero while the printer stutters.

Suggests to me that it might be a communication issue with the printer, but to know for sure you'd need to upload all the logs etc. and test this in safe mode, enable serial.log and upload this too.