OctoPrint/OctoPrint #3862: Communication Timout while waiting for user action

AndKe, 2020-12-05:
#### What were you doing?
Marlin 2.0 is configured with advanced pause.
Simulated filamant runout causes error anyway.

#### What did you expect to happen?
I'd expect no timeout being possible while printer is in paused state.

#### What happened instead?
It timed out, failed, before I could change filement

#### Did the same happen when running OctoPrint in safe mode?
yes.

#### Version of OctoPrint
1.5.1 (same with older version)

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
Custom built , Marlin 2.0

#### Browser and version of browser, operating system running browser
Firefox

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log
```
Recv: ok
Send: N134 G1 X108.660 Y124.354 E35.04503*92
Recv: ok
Recv: //action:prompt_end
Recv: //action:prompt_begin FilamentRunout T0
Recv: //action:prompt_show
Recv: //action:paused filament_runout 0
Printer signalled that it paused, switching state...
Changing monitoring state from "Printing" to "Pausing"
Send: N135 G1 X108.061 Y123.859 E35.11567*93
Recv: //action:paused
Printer signalled that it paused, switching state...
Recv:  T:214.34 /215.00 B:55.13 /55.00 @:42 B@:0
Recv: echo:busy: processing
Recv:  T:214.61 /215.00 B:55.00 /55.00 @:39 B@:0
Recv: echo:busy: processing
Recv:  T:214.77 /215.00 B:54.96 /55.00 @:38 B@:0
Recv: X:318.00 Y:290.00 Z:10.35 E:33.05 Count X:25440 Y:23200 Z:3986
Recv: //action:notification Print Paused
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Pausing" to "Paused"
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Paused" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

```

cp2004, 2020-12-05:
This looks to me like a firmware issue, not OctoPrint's. Just tested this with my own printer (Marlin 2.0.7.2) and it continues to respond to commands in the paused state, including when I then press 'resume' in OctoPrint. 

A timeout is still possible while paused, since the printer should continue to respond to commands - for example to purge more filament, or resume, and when paused you should be able to use the buttons in OctoPrint to move the nozzle out of the way, or change the temperature. All of which require the printer to still at least *respond*, and having just tested it I can confirm that mine *does*, so I think a communication timeout is the correct thing for the host to do in this situation.

You could try updating your firmware, or seeking support on the [community forums](https://community.octoprint.org), [discord server](https://discord.octoprint.org) or from the wider 3D Printing community.

The information the issue bot is after is the **template in full**, you've missed sections such as the `octoprint.log`.

foosel, 2020-12-05:
I agree completely with the assessment by @cp2004. OctoPrint is doing the right thing here, the printer stops communicating. It should at least be sending `busy` messages, but it goes completely radio silent. This is a firmware issue.

closed 2020-12-05 by foosel

FotoFieber, 2020-12-10:
Opened an issue in Marlin: https://github.com/MarlinFirmware/Marlin/issues/20431

FotoFieber, 2020-12-17:
Use following define in marlin:
```
// Bad Serial-connections can miss a received command by sending an 'ok'
// Therefore some clients abort after 30 seconds in a timeout.
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
#define NO_TIMEOUTS 1000 // Milliseconds
```