OctoPrint/OctoPrint #3867: As of 1.5.0, the blocked commands setting can't be cleared

lciscon, 2020-12-05:
#### What were you doing?

I was trying to delete the blocked commands (M0,M1) set in the  Firmware & protocol tab of the Serial Connection settings dialog.

#### What did you expect to happen?

I expected that I could delete the commands like I could do in earlier versions (1.4)

#### What happened instead?

If I remove the contents of that setting and hit the Save button, the old contents (M0, M1) reappears right before the dialog closes. If I then re-open the dialog the old contents are in that row (M0, M1).

If I delete the contents, save and close the dialog, reboot Octoprint and then re-open the dialog the old values are still there (M0, M1).

If I CHANGE the contents to a new string with the exact same length (M1, M2), save, close and reopen it retains the new value (M1,M2). And I have verified that this value is actually active.

I verified that the behavior worked correctly in 1.4 (i.e. you can delete the value just fine).

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.5.1

#### Operating System running OctoPrint

Octopi 0.17.0

#### Printer model & used firmware incl. version

Northworks 600x.  Firmware v. 1.6

#### Browser and version of browser, operating system running browser

Chrome Version 87.0.4280.67 

#### Link to octoprint.log

https://gist.github.com/lciscon/0bc5ad953f8ce9db532e8147eb0e7218

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

jneilliii, 2020-12-05:
confirmed, I'm able to reproduce.

jneilliii, 2020-12-05:
I also noticed in an attempt at a workaround that if you just put on item in there the second item from before gets put in.

[image]

jneilliii, 2020-12-05:
@lciscon if you use `SOMETHING1, SOMETHING2` it will stick and give you the desired effect as a temporary workaround.

cp2004, 2020-12-05:
Looks like an issue in the client side - the debugger says data is never sent to the server if they are deleted, however it *is* if they are changed.

cp2004, 2020-12-05:
This actually doesn't work with any of the comma separated lists, maybe other settings too.

Documenting how far I have got narrowing this down.
* The data is **wrong** when it comes out of the viewmodel - it remains ["M0", "M1"] in the observable:
   https://github.com/OctoPrint/OctoPrint/blob/38256815ef6d6dd602e0fac949a17fbf8f30f838/src/octoprint/static/js/app/viewmodels/settings.js#L985
* The data processed from the special mapping is **correct**, from here, returns []:
https://github.com/OctoPrint/OctoPrint/blob/38256815ef6d6dd602e0fac949a17fbf8f30f838/src/octoprint/static/js/app/viewmodels/settings.js#L1179-L1180

* Then when merged, the empty data is overwritten with the previous data here:
 https://github.com/OctoPrint/OctoPrint/blob/38256815ef6d6dd602e0fac949a17fbf8f30f838/src/octoprint/static/js/app/viewmodels/settings.js#L1182

This appears to be as a result of https://github.com/OctoPrint/OctoPrint/commit/9494db95683ba280ea84bbc440f9db502bd02df4, and I can confirm that rolling back this change fixes this behaviour. So now it is down to @foosel - what was that change fixing, and what can be done instead. Bug analysis done 🙂

12345Josparky, 2020-12-06:
Ok so if we do a lot of printing that needs the pause what is the correct thing to do as I get this message now at every pause Not sending M0 to printer, it's configured as a blocked command Do we just let octoprint handle the pause or do we need it sending it to the printer as well. I don't know anything at all about this coding but would like to know do I just leave it as is and put up with the message that comes up now every time. Never seen that message before until I updated to 1.5.1 version

foosel, 2020-12-07:
@cp2004 Wow, thank you **SO** much for the legwork you did here!

I have the feeling this and #3845 are actually two symptoms of the same issue. Now I'll need to jog my memory and try to remember what I observed that made me do that change, and then figure out how to achieve the same functionality without breaking list style settings in the UI.

And maybe also push a fixing plugin if possible (I feel this issue to be a bit too low key to merit a full hotfix tbh)

foosel, 2020-12-07:
[image]

Damn... I overlooked the highlighted part. What's probably happening here is the old array data doesn't get overwritten but instead extended by the new. I didn't expect `merge` to do that for arrays as well. Will need to find a solution that only touches objects, like our `dict_merge` helper on the backend.

foosel, 2020-12-07:
Fixed by the above commit, ready for 1.6.0. Should another hotfix 1.5.2 be need to be pushed, I'll backport this into that. In the meantime I'll whip up a quick plugin to work around this.

cp2004, 2020-12-07:
Good to see I was useful 😄 

Just a bit of tracing through logging to console, then a bit of `git blame` to find the issue. On to the next bug report...

foosel, 2020-12-07:
This plugin should work as a fix for now: https://github.com/OctoPrint/OctoPrint-FixSettingsDeepMerge

Install via plugin manager, get more, from this URL: https://github.com/OctoPrint/OctoPrint-FixSettingsDeepMerge/archive/master.zip

lciscon, 2020-12-07:
Awesome! Thanks everyone!!

12345Josparky, 2020-12-08:
Ok so I know this is for bug reports but can someone answer this for me please or tell me where to get the answer.

Ok so if we do a lot of printing that needs the pause what is the correct thing to do as I get this message now at every pause Not sending M0 to printer, it's configured as a blocked command Do we just let octoprint handle the pause or do we need it sending it to the printer as well. I don't know anything at all about this coding but would like to know do I just leave it as is and put up with the message that comes up now every time. Never seen that message before until I updated to 1.5.1 version

Thanks

foosel, 2020-12-08:
> Never seen that message before until I updated to 1.5.1 version

It was simply silently suppressing it until 1.5.x. It's always done what it does now, just silently. There was confusion by some people over commands getting silently suppressed, so when commands are now suppressed they generate a heads-up for the user in form of a notification. If everything has always worked for you so far, then the behaviour of blocking the command is still correct and you don't need to do anything at all.

OctoPrint by default suppresses `M0` and `M1` and instead pauses on its end since most firmware out there would read `M0`/`M1` as a command to pretty much commit suicide until a button is pressed on the printer - not what you usually want.

Btw, you can also disable these notifications altogether via Settings > Serial Connection > Behaviour > Sanity Checking:

[image]

12345Josparky, 2020-12-08:
Thank you for the better explanation. So probably best for me to just leave it like it is and close the popup then?? If I disable the checking will that not block some other checks that I might need to see?? Can the M0 and M1 commands bee removed like this fix does without harm?? sorry for so many questions just trying to understand all this from a dummy point.

foosel, 2020-12-09:
> Can the M0 and M1 commands bee removed like this fix does without harm??

Removed from the GCODE, yes. Removed from the block list, I wouldn't recommend it, because then your printer will get them sent and the *firmware* will handle those commands, which means hard lock until you press a button (at least according to spec).

> If I disable the checking will that not block some other checks that I might need to see??

Well, any related to blocked commands, yes. Unchecking that will return to the old behaviour before 1.5.x.

> So probably best for me to just leave it like it is and close the popup then??

Yes. If you use `M0`/`M1` to pause your prints to do whatever, you can also inject `@pause` instead, see https://docs.octoprint.org/en/master/features/atcommands.html

foosel, 2020-12-10:
About to release 1.5.2 after all (third party dependency released a breaking update), so this fix gets a hotfix release after all.

closed 2020-12-10 by foosel

12345Josparky, 2020-12-10:
Thanks foosel

commit ade3a772 (2020-12-07): 🐛 Replace _.merge with object only deepMerge

It turns out _.merge will also merge arrays, which in turn will re-add
anything we've just removed from a setting that happens to be an
array, thus as blocked commands or temperature presets.

Fixes #3855
Fixes #3867
commit 12d6de51 (2020-12-10): 🐛 Replace _.merge with object only deepMerge

It turns out _.merge will also merge arrays, which in turn will re-add
anything we've just removed from a setting that happens to be an
array, thus as blocked commands or temperature presets.

Fixes #3855
Fixes #3867