OctoPrint/OctoPrint #3869: Setting SW Update credentials via CLI fails

cp2004, 2020-12-07:
#### What were you doing?

1. Try and configure `plugins.softwareupdate.credentials.github` via the CLI, like so:
    ```
    octoprint config set plugins.softwareupdate.credentials.github my_api_key2
    ```
2. Value remains as the default. Note that it *can* be overwritten like this, if it has been changed manually (in config.yaml)
    - While the default value is `None`, I tried with a couple of different defaults & the same issue was present.

There's not a lot more to it, but we reckon it might be to do with the update check overlays. Very minor issue, since the CLI config is not used often. But a bug nevertheless 🙂 

#### What did you expect to happen?
The value would update & save

#### What happened instead?
Value remained as the default

#### Did the same happen when running OctoPrint in safe mode?
Not applicable via CLI

#### Version of OctoPrint
1.6.0dev (maintenance), also 1.5.0/1

#### Operating System running OctoPrint
Windows 10

#### Printer model & used firmware incl. version
VIRTUAL (N/A)

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log
No logging on the CLI as far as I can see, so I dug through the debugger, but equally didn't find much of any use.
Feels wrong not to submit the octoprint.log, so here we are:
 
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5654706/octoprint.log)

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Will-wastelander, 2020-12-08:
I can confirm that this happens on OctoPi 0.17.0 along with 0.18.0 armhf/arm64.

foosel, 2020-12-08:
Seems to be due to the default being `None` indeed. Probably some path evaluation issue? Digging...

cp2004, 2020-12-08:
Tried changing the defaults & still couldn't get it to work, CLI works with all other paths that are None *apart* from the GitHub credentials.

foosel, 2020-12-08:
Yeah, that was a false observation. Could have sworn that changing the default fixed it, but just now as I was stepping through with the debugger it didn't. However, reason is that the settings defaults aren't getting properly populated in CLI mode, which means that the path can't be found, which means that (without the `force` parameter, which currently isn't exposed on the CLI) OctoPrint will silently refuse to set this value. 

So, actual bug is that the CLI isn't initializing the default overlays from the SettingsPlugins. What I'm wondering though is whether I should attempt to fix that (it could cause issues due to partial, unexpected initialization with some third party plugins) or whether I should just default to `force=True` on the set call on the CLI. Will see what impact has the former and if push comes to shove go with the latter.

foosel, 2020-12-08:
It became `force=True` as firing up the full platform just for a settings editor just feels wrong.

commit 8a7a436e (2020-12-08): 🐛 CLI: Force set values on the config CLI

We don't initialize the settings plugins (and we don't want to change
that either since it's a TON of potentially error prone overhead just
to allow a glorified editing of a yaml file), so we now can force
values even if their paths are unknown to OctoPrint. Can be abused to
clutter config.yaml, but so can plugins.

Closes #3869
closed 2021-04-27 by foosel