OctoPrint/OctoPrint #3870: [Request] -  Confirmation when clicking Disconnect in Web UI

coconutservices, 2020-12-08:
**Is your feature request related to a problem? Please describe.**
Today I errantly brushed the disconnect button in the Web UI while on on my cell phone, and it ended a long running print nwith no recourse.

**Describe the solution you'd like**
The same prompt that appears when you click cancel for the print. 

**Describe alternatives you've considered**
None really. Seems like the same safeguard would apply to either action.

**Additional context**
I made up a few new swear words when this happened. ;)

jneilliii, 2020-12-08:
You mean like this one that's already there?

[image]

coconutservices, 2020-12-08:
Well I can tell you today i clicked disconnect and did not get any
confirmation at all it just disconnected. I watched it stop dead in its
tracks.

So in guess this is a bug report and not a feature request.

coconutservices, 2020-12-08:
would you like a video? I have tried this repeatedly on my cellphone browser and have never seen that prompt, it just disconnects.

foosel, 2020-12-08:
Cannot reproduce problem, neither on desktop nor mobile. I'm seeing a confirmation in both cases:

[image]

[image]

If your mobile interface looks different than this, you are not using the default UI (in which case you need to take that up with whoever provides the one you are using), or a version older than 1.3.7 (in which case you REALLY should update as that would mean your version is older than 2 years, and this would be a duplicate of #2287).

coconutservices, 2020-12-08:
I have the regular ui as shown and am using the most current version. I can
demonstrate that the UI does not give a confirmation dialog. After the
current print ends I will be happy to send that. I have upgraded this on at
least 2 occasions, I would suspect then that this maybe an artifact of
upgrading?
 I can update the OS as well.

[image: image.png]

foosel, 2020-12-08:
Ok... waiting for a video then. Also, since this is turning into a bug report, please try in safe mode and fill out [the full template](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report). Note that it will only show that dialog by printing or paused, not while idle (but your description sounds like it didn't show it while printing).

foosel, 2021-01-25:
Still no video or bug template, closing. If this is still an issue, feel free to open a *full* bug report.

closed 2021-01-25 by foosel