OctoPrint/OctoPrint #3871: PrintDone event fires before GCode scripts have finished.

shadycuz, 2020-12-08:
#### What were you doing?

I have a plugin that snaps a picture of the PrintDone event. A user reported that the picture is taken before finishing gcode is done running. The code is from Octoprints GCode Scripts and not from a slicer. Similar to https://github.com/OctoPrint/OctoPrint/issues/2699

Original issue by user https://github.com/taxilian/OctoPrint-Twilio/issues/14

#### What did you expect to happen?

That PrintDone would mean that all printer activity related to the current job has finished. 

#### What happened instead?

The printer continued to move after sending the PrintDone event.

#### Did the same happen when running OctoPrint in safe mode?

I honestly didn't try.

#### Version of OctoPrint

I have noticed this in many versions going back a while. I'm not sure how to test this on a virtual printer but in the logs I see:

```
2020-12-08 04:27:55,181 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2020-12-08 04:27:55,189 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: test.gcode, owner: shadycuz
2020-12-08 04:27:55,323 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
```
Seems like PrintDone is called while "Finishing" and not at the end of or after.
 
#### Operating System running OctoPrint
Ubuntu 20 something something on WSL

#### Printer model & used firmware incl. version

Virtual

#### Browser and version of browser, operating system running browser

Chrome
#### Link to octoprint.log

```
2020-12-08 04:27:55,181 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2020-12-08 04:27:55,189 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: test.gcode, owner: shadycuz
2020-12-08 04:27:55,323 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
```

#### Link to contents of terminal tab or serial.log

I can provide if needed

#### Link to contents of Javascript console in the browser

Will provide if needed

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ. (once or twice over the years.)

cp2004, 2020-12-08:
If you run OctoPrint in 'debug' mode `octoprint serve --debug` you can see logging for all the events too, which would allow you to pinpoint this. Otherwise we'll try too.

shadycuz, 2020-12-08:
@cp2004 

I'm not sure what we are looking for, its the same output as I gave above, nothing has changed. 

```
2020-12-08 07:19:23,224 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {'state_id': 'FINISHING', 'state_string': 'Finishing'})
...
2020-12-08 07:19:23,233 - octoprint.events.fire - DEBUG - Firing event: PrintDone (Payload: {'name': 'test.gcode', 'path': 'test.gcode', 'origin': 'local', 'size': 56, 'owner': 'shadycuz', 'time': 0.03050320001784712})
...
2020-12-08 07:19:23,241 - octoprint.plugin - DEBUG - Calling on_event on smsnotifier
...
2020-12-08 07:19:24,817 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {'state_id': 'OPERATIONAL', 'state_string': 'Operational'})
```

This is the intended behavior of octoprint I think. I'm asking we change the intended behavior because I think it's misleading. The PrintDone event does not mean that the printer is finished, its that the end of the gcode file was reached. Since you can add additional gcode with octoprint the printer might actually have more things to do. If the printer is still doing things for the current job, then its not really done. 

I doubt you will want to change the current event as there is probably lots of plugins that rely on the current behavior, maybe its possible we add events for the additional start and end gcode.

It's also possible in my plugin I keep track of previous events? Not sure this would work?

```python
 def on_event(self, event, payload):
        
        if event != "PrinterStateChanged":
            return

        self.prev_state = self.cur_state
        self.cur_state = payload['state_id']
   
        if self.prev_state == 'FINISHING' and self.cur_state == 'OPERATIONAL':
            # Printer is actually finished, take a picture and send the notification.
```

cp2004, 2020-12-08:
@shadycuz debug logging was just to confirm that this was in fact the issue, that the PrintDone event was definitely sent before the script. There's events `PrintCancelling` and `PrintCancelled`, similarly for pausing - maybe it makes sense to bring PrintFinishing or something in to make this more consistent.

jneilliii, 2020-12-08:
Yeah, switching the current PrintDone to PrintFinishing and then making the PrintDone happen after the post scripts makes sense to me and I don't think would impact any plugins.

shadycuz, 2020-12-08:
I can submit a PR if @foosel is okay with one of these approaches.

foosel, 2020-12-08:
I'm not 100% sure about this not impacting any plugins - changes like these are always ticking bombs. Still, I'd be open to a PR and then see if it generates any "WAIT WHAT?!" from the plugin developer community ;)

shadycuz, 2020-12-08:
I wasn't going to change the existing event, just add new ones for the gcode scripts.

@foosel

kantlivelong, 2020-12-16:
IMO moving the PRINT_DONE event after the gcode script makes sense. Being that events are fired async I wouldn't ever rely on receiving the event before "afterPrintDone". Because I cannot rely on it firing beforehand there is nothing I could reliably do that would change the result operations performed by "afterPrintDone".

shadycuz, 2020-12-16:
@kantlivelong but changing the current event is not going to be backwards compatible.  I think its best to just create a new event for the pre and post gcode steps. 

Probably a "start" and "end" event for each step. I do plan on working on this "eventually". Possibly after Christmas. If you want to start working on this then I can help on a branch in your fork if you're okay with that.

shadycuz, 2020-12-22:
@foosel 

Today I started looking into implementing new events for the [predefined gcode](https://docs.octoprint.org/en/master/features/gcode_scripts.html#sec-features-gcode-scripts-predefined) that is able to be set in the settings. I first started by finding where these things are [called](https://github.com/OctoPrint/OctoPrint/blob/bf8ff6358556aaa073da2d0646f193c3a1fef48f/src/octoprint/printer/standard.py#L1606-L1612). I noticed that [Printer.script](https://github.com/OctoPrint/OctoPrint/blob/bf8ff6358556aaa073da2d0646f193c3a1fef48f/src/octoprint/printer/standard.py#L427-L443) was probably the best place to put the new logic. 

So my plan was something like this inside the script function:

```python
def script(...):
    // error handing at the top of the script...

    eventManager().fire(f"{name}Start", context['event'])

    // rest of the functions logic

    eventManager().fire(f"{name}End", context['event'])
```

So this is pretty much all that is needed besides a tiny bit of logic handling if the context is `None`. This means all gcode scripts would have `Start` and `End` events.

I wanted to check with you because I had two main concerns. 

1. This is the first time that I know of that events have been dynamically generated. All these events in the docs are [statically named](https://docs.octoprint.org/en/master/events/index.html#available-events) and even in the event class they are all [class variables](https://github.com/OctoPrint/OctoPrint/blob/bf8ff6358556aaa073da2d0646f193c3a1fef48f/src/octoprint/events.py#L38-L46).
2. My next concern is about "registering" for events. I don't know what that means. I have an event mixin based plugin but I don't remember registering for the event my plugin listens for. I just do `if event != 'PrintDone': return` and that works fine. I read the code and it doesn't seem that `fire` method checks to see if the event is a class variable so it seems like if I fire `AnyRandomEvent` name then plugins with the event mixin SHOULD get it. I could probably test this theory out in 10 mins but I thought I would ask because of my first concern. 

In total, I see 5 calls to `script` in the print object. I'm not sure if there are more calls elsewhere. So at minimum, we would be adding start and end events for:
1. beforePrintStarted
2. afterPrintDone
3. afterPrintCancelled
4. afterPrintPaused
5. beforePrintResumed

Let me know your thoughts on the above and if you are okay with a dynamically generated event for each script custom gcode script. I can send a PR in the next 2 weeks. 

Thanks

cp2004, 2020-12-22:
@shadycuz Gina is on holiday until January 10th*ish*, so you might be waiting. In the meantime, I'll give you my thoughts 🙂 

That seems like quite a few events to me, but if it is useful then I suppose it will happen, and they will get fired. I'm not entirely sure about the overhead of sending events, but they are done asynchronously so I don't think it will block sending the gcode scripts. *Just had a look, doesn't seem like too much from code reading.*

---

> This is the first time that I know of that events have been dynamically generated. All these events in the docs are statically named and even in the event class they are all class variables.

I think that events should try and remain static ideally, rather than being dynamically generated (see documenting below). This means that anyone can import the events class without needing to worry about the events being adjusted/renamed easily. Maybe I'm weighing this too highly, and I can't really say either way.

In addition, you should keep events consistent - this means that they start with a capital and use CamelCase. As I read your code snippet, will the event be created as `beforePrintStartedStart` - I don't think this is very clear/concise personally.

They would also have to be documented, which means you would have to type them all out anyway. If they are scattered throughout the code, this makes documentation quite a lot more difficult.

My suggested solution to keeping them static but *referencing* them dynamically is using `getattr()` to find the class name. Something like:
```python
event = gettattr(octoprint.events.Events, name + "Start")
eventManager().fire(event, context['event'])
``` 

---

> My next concern is about "registering" for events. [...]

There are two ways that events can be 'handled'. First is as you describe, using `on_event()` in an `EventHandlerPlugin`. The second, is subscribing to events, such as:

```python
def event_callback():
    print("Event fired!")

octoprint.events.eventManager().subscribe(
    "PrintStarted",
    event_callback
)
```
Then the event callback will be called every time the event is fired. This should work for any kind of event, whether it is static/defined by a plugin/dynamic.

That seems like a lot of words, but hopefully it gets the point across! Let me know if you have more questions.

shadycuz, 2020-12-23:
@cp2004 

Well the name scheme is easy to fix. As far as the names of these events being goofy... well we can try to work that out but it going to be a bit rough. For example `beforePrintStarted` is the name of the GCODE that is running. So if we need an event for when that gcode starts and ends it would have to be something like:

- `StartBeforePrintStarted`
- `StartBeforePrintStartedGcode` (This seems okayish)
- `BeforePrintStartedStart`
- `BeforePrintStartedStartGcode`
- `BeforePrintStartedInProgress` (This is nice too)
- `GcodeBeforePrintStartedRunning`
- `BeforePrintStartedGcodeRunning` (Also nice)  

I think of all these I probably like `BeforePrintStartedGcodeRunning`. I still don't think it would make sense to hardcode these into the class var for events. This is because the script function also handles [user-defined GCODE](https://docs.octoprint.org/en/master/features/gcode_scripts.html#id6) which we would not know ahead of time. Same with documentation. I would just add a single event that shows the name scheme `${ScriptName}GcodeRunning` and payload to the events docs.

I will start with appending `GcodeRunning` and `GcodeFinished` to the event names for now. I'll shoot over a PR with the dynamic stuff since most of the logic is the same even if we need to change it so the events are class vars.

Thanks

shadycuz, 2020-12-24:
I also have the draft PR opened.

closed 2021-01-18 by shadycuz