OctoPrint/OctoPrint #3872: [Request] User accounts that cannot be logged into, or per-API key permissions

timur-tabi, 2020-12-08:
I'd like to see one of the following implemented:

1. User accounts can only be accessed via the API key, and cannot actually be logged into. This would eliminate the need to specify a password, because there's no way to log into the account with that userid.  This would allow to you create users with specific limited permissions, and then the API keys for that user would be also be restricted to those permissions.  This would let you create different API keys for each application that you use (e.g. slicers that can upload the gcode to Octoprint directly), and that application would be restricted to whatever permissions of the API key.  We can do this today, but we have the side-effect that the user can also be logged into, which creates a security hole.

2. Allow us to specify a subset of permissions on a per-API key basis.  So when you create an API key under a user, you can specify the permissions of that API key, and any application using that key would be restricted to the API's permissions rather than the user's permissions.

timur-tabi, 2023-07-29:
This bug is 2.5 years old.  It seems to me like item 2 has been implemented -- I have a user with limited permissions that is used only by the slicer.

I would still like it if there were a way to prevent this user from being logged in via the GUI, for extra protection.

cp2004, 2023-07-29:
It's not a bug, it's a request for someone to come along and implement it :)