OctoPrint/OctoPrint #3873: Uploading duplicate file name fails without warning

KenLucke, 2020-12-08:
#### What were you doing?
Uploading a model which had been changed slightly, but the filename remained identical.

1. Upload a model
2. Attempt to upload another model with identical filename

#### What did you expect to happen?
Should have warned that file was a duplicate when first attempting to upload.  Ideally, would give the option to Overwrite, Rename, or Cancel.

#### What happened insteadl
Went through entire upload/saving process, then failed without a warning or notice (atthough the previous file did flash in the file list).

#### Did the same happen when running OctoPrint in safe mode?
Currently printing, cannot restart in Safe Mode.

#### Version of OctoPrint
```browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:83.0) Gecko/20100101 Firefox/83.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959975936
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x50000
env.python.pip: 20.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.1
printer.firmware: Marlin 2.0.7.5_NIC NICW VEADBB (Nov 23 2020
```

#### Link to octoprint.log
Will log if need be next time, have not cleared the logs lately, so it would be huge

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-09:
> **Link to octoprint.log***
Will log if need be next time, have not cleared the logs lately, so it would be huge

I have a freewheel mode on my mouse's scroll wheel, do not worry - if it is huge, I will still look at it. Also, they rotate themselves each day, so the work is done for you. Not optional with bug reports, when something fails we need logs.

Also, this does not 'fail' for me, it simply overwrites the old one, with the new file content. If this is the case for you as well, then this will turn to a feature request rather than a bug report.

foosel, 2020-12-09:
It might be silently failing if you are trying to overwrite the file that is currently being printed, because that would nuke the running job. Could be that I missed a dedicated message for that case. But yeah, log file needed indeed.

KenLucke, 2020-12-09:
It is, indeed, overwriting rather than failing (I know if it overwrote while printing, it would kill the print, so I was not doing that, I was in non-printing mode at the time).

So, I guess it turns into a feature request to add a warning dialog that you are about to overwrite the existing file, and give options to cancel or rename.

Do you still need logs, and if so, which one(s) other than the main OP log?

foosel, 2021-03-25:
1.6.0 will ship with a "file name collision" dialog triggered on upload, which allows to cancel upload, upload with a new name (with a collision free suggestion made) and to force overwrite:

[image]

This can also be disabled to restore the old behaviour (silent overwrite) through the settings:

[image]

jneilliii, 2021-03-25:
How will this effect API upload calls?

jneilliii, 2021-03-25:
Never mind, looks like it will only effect API uploads if the `noOverwrite` variable is sent as well, which means all previous upload methods will work as before?

foosel, 2021-03-25:
Yes. Backwards compatible as always, and primarily a pure frontend change (the noOverwrite parameter is just to avoid race conditions if a file gets uploaded while the rename dialog is still open that collides with the rename).

KenLucke, 2021-03-26:
> 1.6.0 will ship with a "file name collision" dialog triggered on upload, which allows to cancel upload, upload with a new name (with a collision free suggestion made) and to force overwrite:

Awesomeness.  THANK YOU!

Alssund, 2021-04-09:
Cura still overwrite when i upload via a application keay API! I have the "Confirm before overwriting a file by upload" checked

cp2004, 2021-04-09:
@Alssund this is something the Cura plugin would have to implement, so far it is only implemented in the regular OctoPrint UI. You would need to ask nicely for a new feature in the Cura-OctoPrint plugin:

https://github.com/fieldOfView/Cura-OctoPrintPlugin/

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel