OctoPrint/OctoPrint #3874: Won't Print/Load/Delete/Download/Move if File is named....

ZhuDaHai, 2020-12-09:
Yup, this is a Wumpus Stumper: If file is named "Display Adapter - PETG Hevo.gcode", it won't print/load/delete/download/move the file. But if I rename it to something else, like "Test.gcode", everything works.

#### What were you doing?
Trying to print a gcode file to my 3D Printer:
1. Uploaded the file named "Display Adapter - PETG Hevo.gcode"
2. Clicked on the Print Icon: nothing happened
3. Clicked on the Load Icon: nothing happened
4. Clicked on the Download Icon: nothing happened
5. Clicked on the Delete Icon: nothing happened
6. Created a folder called "Test" and then clicked on the Move icon and got"

> Unable to move file or folder Display Adapter - PETG Hevo.gcode to /Test

7. Renamed file to "Display Adapter 2 - PETG Hevo.gcode", still failed to Print/Load/Download/Delete/Move
8. Renamed file to "Test.gcode" and voila! Prints! (assuming all the other icons work, but its printing right now, so haven't tested it

Steps enumerated above ^

#### What did you expect to happen?
Problem Child/File is attached.

> Do you expect me to talk?
> No, Mr Bond. I expect you to die 
> 

I expected to be able to Print/Load/Delete/Download/Move - none of which happened.

[Display Adapter - PETG Hevo.zip](https://github.com/OctoPrint/OctoPrint/files/5664286/Display.Adapter.-.PETG.Hevo.zip)

#### What happened instead?
Nothing. Its like its there in name only. can't even get rid of it!

#### Did the same happen when running OctoPrint in safe mode?
Yes

Fails in Safe Mode the same way.

#### Version of OctoPrint

OctoPrint 1.5.1
Python 2.7.16
OctoPi 0.17.0

#### Operating System running OctoPrint
OctoPi 0.17.0

#### Printer model & used firmware incl. version
HyperCube Evolution
Smoothieware 2018 version

#### Browser and version of browser, operating system running browser
Windows 10 Pro 64 bit - current with updates
Google Chrome 87.0.4280.88

#### Link to octoprint.log
Attached
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5664276/octoprint.log)

#### Link to contents of terminal tab or serial.log
[serial.log](https://github.com/OctoPrint/OctoPrint/files/5664296/serial.log)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-09:
Do you have any chrome extensions enabled?

Just tested this myself, and for some reason the request was being blocked by uBlock Origin, but where I don't have this (or disabled it) moving the file worked just fine. Maybe something is being matched in it's filters?

Relevant error from browser console is this (which was helpfully missed out in the template 😉), and it shows up in uBlock when I try (so it's definitely that causing the issue).

```javascript
jquery.min.js:2 POST http://localhost:5000/api/files/local/test/Display_Adapter_-_PETG_Hevo.gcode net::ERR_BLOCKED_BY_CLIENT
```

ZhuDaHai, 2020-12-10:
Went through my extensions in Chrome and found it was AdBlock.
I would have NEVER guessed that OctoPrint was using the browser to move/delete/load/print files ("download" I can understand). As to why it does this instead of internal Python commands, I have no idea:

> Montgomery Scott: Aye, sir. The more they overthink the plumbing, the easier it is to stop up the drain.

closed 2020-12-10 by ZhuDaHai

foosel, 2020-12-10:
It doesn't use the browser, it sends a request with the corresponding command to the server. That request contains the file name in the path. If the browser now blocks the request, the server never knows it's supposed to do something.