OctoPrint/OctoPrint #3875: additional_state_data is broken with invalid JSON

eyal0, 2020-12-09:
#### What were you doing?

Trying to use the additional_state_data hook to send a dict with a value that is `float("inf")`.

1. Create a plugin in the normal way.
2. Add to the hooks: `"octoprint.printer.additional_state_data": __plugin_implementation__.additional_state_data`
3. Add this function:

```python
  def additional_state_data(self, initial, *args, **kwargs):
    return {"x": float("inf")}
```
#### What did you expect to happen?

I expected that the print time left would still be displayed on screen.

#### What happened instead?

Print time left is no longer displayed on screen.  Printer is crippled, unable to print from GUI.

#### Did the same happen when running OctoPrint in safe mode?

I didn't try this because this bug can only happen if plugins are enabled.

#### Version of OctoPrint

OctoPrint 1.5.1 Python 3.7.7

#### Operating System running OctoPrint

Linux

#### Printer model & used firmware incl. version

Virtual printer

#### Browser and version of browser, operating system running browser

Chrome

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-10:
No logs here? Looks like that part of the template was deleted or ignored 🤔 

I haven't got time to reproduce this at the moment & find the error myself. A log would be great, to save myself time in figuring out how to break something you've already done. It will get looked at by someone, but sooner if there are logs.

eyal0, 2020-12-10:
I didn't bother with the log because there's nothing of interest in it.  The bug is that OctoPrint thinks that it is sending valid JSON but it is not.  The web UI is trying to parse that invalid UI and failing.  The web UI doesn't show any error, though, which is a shame because it ought to have parsed the JSON instead of just consuming it.

I will do some more debugging on it.  I think that the solution is that OctoPrint sanitize the JSON before sending it.

cp2004, 2020-12-10:
Ok, the way the bug report was written sounded like there was obvious errors or something. We prefer logs to be included even if you don't think they are relevant, because of the whole system/environment, what events have happened. With exceptions, people start to not provide logs & waste our time.

I understand the problem a *lot* better now with that follow up comment, thanks. Maybe a simple `flask.jsonify` or something should be used before the request goes out.

eyal0, 2020-12-10:
I would like to make it easier for you to debug.  If I sent you a link to a plugin that you can install, would that help?

https://github.com/eyal0/OctoPrint-TimeToFilament/archive/break_json.zip

Install that into your Octoprint.  It ought to make your printer completely not functional.

In your browser (I used Chrome), try hitting "F12" to open up the developer console.  Then go to the "Network" tab and look for the "websocket" transfer, which should be on-going and it never ends.  Double click on it.  Then select the "Messages" tab and you'll see about 2 messages per second.  Each one is *supposed* to be JSON but they aren't and that breaks the webUI.

Here's what it looks like when it works:

[image]

Notice "Network" is selected at the top.  Notice that "websocket" is selected on the left panel.  Notice "Messages" is selected for the websocket.  Notice that I selected a message.  Finally, notice that the contents of the message were recognized as JSON and so the bottom right shows a tree view of the data, with element `a` and sub-element `0`.

Now again but with the broken plugin:

[image]

Same as before but the text in the bottom right is shown as text instead of a tree view because Chrome did not recognize the text as valid JSON.  The text is:

```
a[{"current":{"state":{"text":"Operational","flags":{"operational":true,"printing":false,"cancelling":false,"pausing":false,"resuming":false,"finishing":false,"closedOrError":false,"error":false,"paused":false,"ready":true,"sdReady":true}},"job":{"file":{"name":null,"path":null,"size":null,"origin":null,"date":null},"estimatedPrintTime":null,"lastPrintTime":null,"filament":{"length":null,"volume":null},"user":null},"currentZ":null,"progress":{"completion":null,"filepos":null,"printTime":null,"printTimeLeft":null,"printTimeOrigin":null},"offsets":{},"resends":{"count":0,"transmitted":8,"ratio":0},"plugins":{"TimeToFilament":{"x":Infinity}},"serverTime":1607643217.940848,"temps":[{"time":1607643217,"tool0":{"actual":21.3,"target":0.0},"bed":{"actual":21.3,"target":0.0},"chamber":{"actual":null,"target":null}}],"logs":["Recv: T:21.30/ 0.00 B:21.30/ 0.00 @:64","Recv: Not SD printing"],"messages":["T:21.30/ 0.00 B:21.30/ 0.00 @:64","Not SD printing"],"busyFiles":[]}}]
```

If I remove the leading `a[` and trailing `]` then I can paste it into an online JSON parser like this one: http://json.parser.online.fr/

It parses what it can but also reports an error:

`SyntaxError: Unexpected token I in JSON at position 634`

This corresponds to the bare word `Infinity` in the text, which is not valid JSON.

What's strange is that the python string for infinity is not `Infinity`, it's `inf`.  So it seems like the python side of the `self._emit` function is trying to create something that javascript can read.  It's not strict JSON but `Infinity` is a valid identifier in javascript.  So a JSON parser on the javascript side that isn't too strict could potentially figure out what `Infinity` means and create the right object.  It would be non-standard JSON but it would work!

Barring that, the python should make sure that what it gets from the `additional_state_data` hook is strictly JSON.  Maybe convert from dict to JSON and then back to dict and report errors if there are any.  That would solve the problem, I think, though it wouldn't allow for using "Infinity" like I wanted to.

I hope that helps.

eyal0, 2020-12-11:
I have searched the sockjs code, both sockjs client and tornado sockjs server, and I can't find any mention of "Infinity".  I'm not sure how it's happening but somewhere, the message coming from the server that has a python infinity in it, which is `inf`, is getting converted to `Infinity`.  I don't see where it is happening!

eyal0, 2020-12-11:
Alright, looks like that is built-in to the python json parser.

```
$ python -c 'import json; print(json.dumps({"x": float("inf")}))'
{"x": Infinity}
```

So that's where the "Infinity" is coming from.  The JSON parser on the javascript side is shitting the bed when it gets this.

eyal0, 2020-12-11:
One solution would be to make the parser on the receiving side more forgiving.  Here are some solutions: https://stackoverflow.com/questions/21896792/force-json-stringify-to-emit-nan-infinity-or-js-json-lib-that-does-so

Another possibility is to make the JSON encoding on the sending side more strict.  There is an option to the python json dumps funtion called `allow_nan` that will do it.

I think that might be best.  Modifying the parsing of JSON on the javascript side would be a pain in the ass.  So let's just make it report an error in python!  Easy.  I will make a PR.

cp2004, 2020-12-11:
Thanks for the work on this. The PR will be reviewed as soon as possible (no guarantees as to exactly when, it's busy at the moment), but the solution looks promising to me.

foosel, 2020-12-11:
Fixed by the aforementioned PR and the above commits, ready for 1.6.0

eyal0, 2020-12-11:
Thanks! My solution had problems and yours looks better.

PR #3878 (2020-12-11): Validate websocket JSON before sending
The default json library in python will send infinity values as `Infinity` as described [here](https://docs.python.org/3/library/json.html#json.dump).  Check that the JSON is strictly valid before sending it, otherwise throw an exception.

This fixes #3875

  * [x] Your changes are not possible to do through a plugin and relevant
    to a large audience (ideally all users of OctoPrint)
  * [x] If your changes are large or otherwise disruptive: You have
    made sure your changes don't interfere with current development by
    talking it through with the m [... cut]
commit 7046bb9f (2020-12-11): Validate websocket JSON before sending

The default json library in python will send infinity values as `Infinity` as described [here](https://docs.python.org/3/library/json.html#json.dump).  Check that the JSON is strictly valid before sending it, otherwise throw an exception.

This fixes #3875
commit dc269846 (2020-12-11): 🥅 Validate additional data from plugins at source

That way it won't even enter the state data if it's not JSON compliant,
that's better than no longer being able to send any state reports just
because one plugin is sending broken data.

Closes #3875