OctoPrint/OctoPrint #3879: [Request] Better error message for failed uploads

Stretch96, 2020-12-11:
Currently if an upload fails, it only prompts to ensure that the file ends in a valid extension (eg. `.gcode`)

I've only recently noticed this error message, because I have OctoPrint behind Nginx

It'd be nice to also prompt the user to check any Nginx (or any other Proxy) settings, to ensure there's not an upload size limit

(This was the issue I ran into, as Nginx has quite a small limit by default)

cp2004, 2020-12-12:
So it showed you an error message that was incorrect? Just clarifying as I was unsure what exactly happened here.

Prompting the user to check proxy settings sounds annoying to me, to many people who may have configured their proxy correctly - and all those using OctoPi, some of which probably don't even know haproxy exists and would panic at the sight of it.

We do also have some [reverse proxy configuration examples](https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107?u=charlie_powell) that you might want to take a look at.

Stretch96, 2020-12-13:
I think it's that the warning message, about having the wrong extension, is probably the least likely cause of a failed upload

Maybe the message could include the HTTP response code, which would hold more clues as to why an upload failed