OctoPrint/OctoPrint #3883: Restoring from backup does not work with plugins installed via the plugin manager, marking them as unknown

wlhlm, 2020-12-13:
#### What were you doing?

1. set up fresh installation of OctoPrint 1.5.2
2. install a plugin using the built-in plugin manager via the web interface (for example [Bed Level Visualizer](https://plugins.octoprint.org/plugins/bedlevelvisualizer/))
3. restart OctoPrint to register plugin (not sure if this is strictly necessary)
4. create a OctoPrint backup using the CLI: `/opt/octoprint-venv/bin/python -m octoprint plugins backup:backup --path /var/lib/octoprint/backup.zip`
5. reset OctoPrint installation (for example, reinstall)
6. restore backup using the CLI: `/opt/octoprint-venv/bin/python -m octoprint plugins backup:restore /var/lib/octoprint/backup.zip`

#### What did you expect to happen?

Restoring from backup reinstalls the previously installed plugins (for the example above, just Bed Level Visualizer from the plugin repository).

#### What happened instead?

The backup plugin declares all plugins as "unknown" failing to restore them from the backup zip. (see transcript below)

#### Did the same happen when running OctoPrint in safe mode?

I have not tested this in safe mode as the problem occurs in combination with plugin functionality.

#### Version of OctoPrint

1.5.2

#### Operating System running OctoPrint

Arch Linux ARM (Linux), python 3.8.6, [octoprint-venv](https://aur.archlinux.org/packages/octoprint-venv/) package from AUR (unofficial)

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

[octoprint.log](http://ix.io/2HSW)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

Saving backup:
```
# sudo -u octoprint /opt/octoprint-venv/bin/python -m octoprint plugins backup:backup --path backup.zip
Initializing settings & plugin subsystem...
Creating backup at backup.zip, please wait...
Done.
Backup located at /var/lib/octoprint/backup.zip
```

Restoring backup:
```
# sudo -u octoprint /opt/octoprint-venv/bin/python -m octoprint plugins backup:restore /var/lib/octoprint/backup.zip
Initializing settings & plugin subsystem...
Unpacking backup to /tmp/tmp8_k0ypxh...
Unpacked
Unknown plugins: bedlevelvisualizer
The following plugins were not found in the plugin repository. You'll need to install them manually.
	Bed Visualizer (Homepage: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer)
Renaming /var/lib/octoprint/.octoprint to /var/lib/octoprint/.octoprint.bck...
Moving /tmp/tmp8_k0ypxh/basedir to /var/lib/octoprint/.octoprint...
Writing info file about unknown plugins
Removing temporary unpacked folder
Removing temporary zip
Restarting...
Restored from /tmp/tmpjrh6vng4
```

Contents of `backup.zip/plugins_list.json`:
```js
[{"key": "bedlevelvisualizer", "name": "Bed Visualizer", "url": "https://github.com/jneilliii/OctoPrint-BedLevelVisualizer"}]
```

I have read the FAQ.

jneilliii, 2020-12-13:
Is this just from CLI with a freshly installed OctoPrint that hasn't yet been initialized? Does it do the same if you try to use the web interface to restore on the first setup wizard step?

wlhlm, 2020-12-13:
> Is this just from CLI with a freshly installed OctoPrint that hasn't yet been initialized?

Not entirely sure what you mean with "initialized" but for the steps outlined above, I've installed the plugin through the web interface, so octoprint is running.

> Does it do the same if you try to use the web interface to restore on the first setup wizard step?

Good question. I haven't tried this yet.

cp2004, 2020-12-13:
Are you trying to restore straight into a 'new' install that has been set up, not where the plugin was installed before the backup?

I was wondering if the CLI doesn't go and get the plugin repo (https://plugins.octoprint.org/plugins.json) on it's own, and depends on the server being run at least once before it is saved.

cp2004, 2020-12-13:
Also, we should get @jneilliii on octoprint/octoprint-triage, although that might mean I don't have as much to do around here 😉

wlhlm, 2020-12-13:
> Are you trying to restore straight into a 'new' install that has been set up, not where the plugin was installed before the backup?

No, for the steps outlined above, I just threw away the virtualenv, but kept the OctoPrint state directory (`/var/lib/octoprint/.octoprint` in my case). I'll try to retest the steps by throwing this directory away also.

cp2004, 2020-12-13:
In that case, it should still have the the plugin repository cached, need to try and test this from the CLI myself.

wlhlm, 2020-12-13:
I have slightly clarified my reproduction steps.

> @jneilliii: Does it do the same if you try to use the web interface to restore on the first setup wizard step?

Recreating the virtualenv, throwing away the state directory, starting OctoPrint and then uploading the backup zip (created previously via CLI) in the setup wizard works. Meaning, that it correctly reinstalls the plugin.

> @cp2004: In that case, it should still have the the plugin repository cached

Even when throwing away the OctoPrint state directory after creating the backup and launching `octoprint plugins backup:restore` in a "cold"/empty environment, OctoPrint automatically downloads `.octoprint/data/pluginmanager/plugins.json`. However, it stills fails to identiy the plugin from the backup zip as being in the plugins directory.

foosel, 2021-01-27:
The code looked good. Fetch repo URL from settings, fetch repo from URL, process, done. However... in the CLI environment not all settings from all plugins get thrown into the settings overlay chain, only the basic defaults plus the CLI plugin plus the config.yaml contents. Which means the repo URL was unset and nothing was fetched.

Fixed now, ready for 1.6.0.

wlhlm, 2021-01-28:
Thank you for the fix. ❤️

AlexHunterCodes, 2021-02-09:
Glad to see this is fixed for 1.6.0. Is there a workaround for current versions besides setting up OctoPrint once to manually reinstall all the plugins, and then restoring the backup over the top?

cp2004, 2021-02-09:
@AlexHunterCodes I'm not sure, the settings won't be properly fired up to find the plugin repository URL. I'm not sure it would work to explicitly set the value in `config.yaml` since you would want it to be the same as the default (https://plugins.octoprint.org/plugins.json)  so it would be cleaned up *I think*. You can still upload a backup via the UI and restore that way when the server is loaded.

foosel, 2021-02-09:
Setting it manually (or via the `config` subcommand) in `config.yaml` should work I think. OctoPrint does clean up stuff that is the same as the default settings, but if you run the restore right after, it will have no chance to do so (as the same reason it could not detect the default repo URL from the plugin's default settings prevents the settings initializer on start of the restore context to clean up things too).

So... `octoprint config set plugins.pluginmanager.repository https://plugins.octoprint.org/plugins.json`, followed by the restore command. Untested though.

Otherwise, yeah, setting things up first, then restoring should work. As should directly uploading the backup in the initial setup wizard screen now that I think of it.

AlexHunterCodes, 2021-02-09:
The backup .zip I'm trying to restore is ~1GB and restoring with the initial setup wizard or afterwards in the settings panel wasn't doing anything. See a ~700kB POST request to `/restore` in the developer console with no response, don't see it at all in OctoPrint's debug logs, but OctoPi's HAProxy log shows that OctoPrint responds with an HTTP 400. Found your nginx-style streaming upload implementation  in #455 for large uploads, does the backup restore not use that or is HAProxy mangling the upload?

Stopping the service, setting `plugins.pluginmanager.repository` as above, and then attempting the command line restore again produces the same result on 0.15.3. Looks like the command line restore process still squashes the config. Tried changing the config inside the backup too, although I wasn't expecting that to work.

cp2004, 2021-02-09:
There's a 1GB size limit, #3993 tracks this with an intended 'solution' to warn *prior* to downloading that the backup will fail to upload again.

The config set command might not do anything for the same reason #3869 was created, the fix for that is *also* in 1.6.0.

AlexHunterCodes, 2021-02-09:
Whoops, sorry for not spotting #3993 earlier. Sounds like you've already been through exactly the same process as me, so thanks for pointing that out. #3993 explains my original problem, and this issue explains why the command line workaround also doesn't work.

Have some printing to do, so don't fancy switching to the devel branch for the fix. I'll just restore the plugins manually for now and wait for 1.6.0.

foosel, 2021-02-09:
Damn, I totally forgot about that one @cp2004, thanks for the reminder! And sorry for the false hope then, I was fairly confident it would work.

> so don't fancy switching to the devel branch

Careful. `maintenance`, not `devel`.

jneilliii, 2021-02-09:
> I'll just restore the plugins manually for now and wait for 1.6.0.

You could technically just remove files within the backup that you may not need (ie timelapses and gcode files). or do a backup that doesn't include those at all.

commit f669c3fb (2021-01-27): 🐛 backup: fix plugin repo access on cli

We were trying to fetch the repo url from the settings. However, due to
the settings overlays from other plugins not being available when
operating in a CLI environment, the repository URL was unset and thus
nothing was fetched.

This is now fixed by falling back to the default repo URL from the
pluginmanager, which has been extracted into a constant for that reason.

Fixes #3883
closed 2021-04-27 by foosel