OctoPrint/OctoPrint #3884: [Request] Ability for plugins to provide a themed bootstrap stylesheet

cp2004, 2020-12-13:
**Is your feature request related to a problem? Please describe.**
OctoPrint's bootstrap implementation doesn't allow for proper bootstrap themes.

**Describe the solution you'd like**
Extract the theming of the standard bootstrap lib (currently partly in `octoprint.less`) to something like `theme.less`, as a proper theme. Then, provide plugins with a hook so that they can replace this file. This works differently, since it means instead of fiddling and overriding properties to change the theme, the file can be dropped in & all is well.

Take a look at how https://bootswatch.com/2/ does things - it provides loads of cool different themes with a click of a button.

**Describe alternatives you've considered**
Re-writing OctoPrint's entire UI in a plugin (seriously), however, this is a simpler solution to get some cooler themes that work better than the way that is currently possible, through overriding the stylesheets manually (which can be a massive pain)

**Additional context**
* [Discussion on discord](https://discord.com/channels/704958479194128507/708230829050036236/787759951480487976)
* https://bootswatch.com/2/
* @LazeMSS
* I will look into this over Christmas and see what can be made. I do have a couple of questions/guidance from the plugin side:
   * Would a hook (that returns a link to a stylesheet, similar to the way the recovery/login ones work) be better, or
   * A method on the AssetPlugin mixin that could return urls/various options for this. Unsure which would fit best,
   * OR, a new plugin mixin ThemePlugin that could do this.

Grabbed, by myself, @cp2004 😄

jneilliii, 2020-12-13:
I wonder if you could do this with the current AssetPlugin method overrides maybe since that already supports css as an option in its return?

cp2004, 2020-12-13:
I thought about this, but since it would need to replace the bundled bootstrap theme (& therefore have special handling) I was more hesitant to do it. Maybe adding `get_theme()` or similar to AssetPlugin, something like this:

```python
def get_theme(self):
    return [list of bootstrap theme CSS files]
```
If there is something returned, the bundled BS2 theme file (that I will make, hopefully) will then *not* be included in the UI.

If you can think of a non-complicated way of specifying 'replaces' in the asset plugin `get_assets()`, then that would be cool too. Like the template one does it I guess.

jneilliii, 2020-12-13:
Yeah, either way would make sense, just depends on which would be easier and not cause as many headaches I suppose. I would think mimicking the template replaces method would be a more complicated approach then the get_theme method.

LazeMSS, 2020-12-14:
The get_theme solution seems fine from my point of view.

@cp2004  you/we would probally need to move all the colors/gradients that is hardcoded into some of the base css into that theme file aswell - but as long it detailed described and declared it should be easy to replace with a theme - that way a custom theme could be anyone a css similar to one from: https://bootswatch.com/2/ and then the extra/octoprint added in an extra css file ie:
dark_theme_bs.css  <-- default bootstrap stuff
dark_theme_op.css <-- Octoprint specific stuff

cp2004, 2020-12-15:
I couldn't get Bootstrap 2 less to build successfully (even trying to install decades old nodejs) since it requires an old version of `less`, so I don't think this will be possible.

IMO even if I could get it to build, it would be an awful experience to try and repeat - so I'm not going to do it. Shame, because I liked the idea of *proper* theming. Maybe we will look into a bootstrap upgrade (how much does BS2 > BS5 sound like a breaking change 😨) instead, and come back to this one at a later date.

cp2004, 2020-12-15:
Short lived wishful thinking, but I would like this to be taken into consideration when we can ditch BS2.

closed 2020-12-15 by cp2004