OctoPrint/OctoPrint #3886: Plugin install hangs if URL returns a 404

cp2004, 2020-12-14:
#### What were you doing?

1. Install plugin from URL, eg:
```
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/3.0.0r3.zip
```
which should be 
```
https://github.com/cp2004/OctoPrint-EEPROM-Marlin/archive/3.0.0rc3.zip
```
but there is a typo.

#### What did you expect to happen?
Error was reported to the user

#### What happened instead?
Unhandled exception in plugin manager thread, and install window hangs:
```python
Exception in thread Thread-137:
Traceback (most recent call last):
  File "C:\Program Files\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\charl.TERRANCE\OneDrive\Documents\8. Coding Projects\5. Octoprint\OctoPrint\src\octoprint\plugins\pluginmanager\__init__.py", line 631, in command_install
    path = download_file(url, folder.name)
  File "c:\users\charl.terrance\onedrive\documents\8. coding projects\5. octoprint\octoprint\src\octoprint\util\net.py", line 254, in download_file
    r.raise_for_status()
  File "C:\Users\charl.TERRANCE\OneDrive\Documents\8. Coding Projects\5. Octoprint\OctoPrint\venv3\lib\site-packages\requests\models.py", line 941, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://codeload.github.com/cp2004/OctoPrint-EEPROM-Marlin/zip/3.0.0r3
```

#### Did the same happen when running OctoPrint in safe mode?
Unable to test, since plugin install is not possible under safe mode.

#### Version of OctoPrint
1.5.x (& maintenance branch)

#### Operating System running OctoPrint

Windows 10, and OctoPi 0.17

#### Printer model & used firmware incl. version
VIRTUAL

#### Browser and version of browser, operating system running browser
Edge Version 87.0.664.60 (Official build) (64-bit)

#### Link to octoprint.log
https://gist.github.com/cp2004/6a4b4ff600fadcd6b0f41529baa0ece4

#### Link to contents of terminal tab or serial.log
NA

#### Link to contents of Javascript console in the browser
Nothing of note beyond the usual here (and I would notice, I've stared at that log hundreds of times now)

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

cp2004, 2020-12-14:
In addition (and it might be related) the undervoltage state is not reported to the frontend, maybe it is related to #3849

kantlivelong, 2020-12-16:
Submitted #3889

foosel, 2021-01-11:
Thank for that, merged to `maintenance` and ready for 1.6.0

cp2004, 2021-01-19:
1.5.3 has been released

closed 2021-01-19 by cp2004

foosel, 2021-03-29:
Consider me mightily puzzled whether this actually made it into 1.5.3 just now and thus a milestone set/reset storm.