OctoPrint/OctoPrint #3892: localPipCommand is not respected by Software Update

MichaIng, 2020-12-16:
#### What were you doing?

1. Apply a custom pip command: `octoprint config set server.commands.localPipCommand /path/to/pip`
2. (Re)start OctoPrint
3. Watch octoprint.util.pip INFO log or apply an OctoPrint update

#### What did you expect to happen?
Subsequent octoprint.util.pip logs show and `/path/to/pip` being used for OctoPrint updates.

#### What happened instead?
Subsequent octoprint.util.pip logs show and the default `/usr/bin/python3 -m pip` being for OctoPrint updates.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.5.0, 1.5.1 and 1.5.2

#### Operating System running OctoPrint
Debian Stretch, Buster and Bullseye

#### Printer model & used firmware incl. version
Irrelevant

#### Browser and version of browser, operating system running browser
Opera 74 on Windows 10

#### Link to octoprint.log
<details>

```
2020-12-16 22:34:23,164 - octoprint.startup - INFO - ******************************************************************************
2020-12-16 22:34:23,166 - octoprint.startup - INFO - Starting OctoPrint 1.5.2
2020-12-16 22:34:23,167 - octoprint.startup - INFO - ******************************************************************************
2020-12-16 22:34:23,483 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2020-12-16 22:34:23,483 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2020-12-16 22:34:23,483 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2020-12-16 22:34:23,497 - octoprint.startup - INFO - Blacklist processing done, adding 12 blacklisted plugin versions: roomtemp (any), GcodeEditor (0.1.1), GcodeEditor (0.2.0), GcodeEditor (0.2.1), GcodeEditor (0.2.2), GcodeEditor (0.2.3), GcodeEditor (0.2.4), GcodeEditor (0.2.5), GcodeEditor (0.2.6), GcodeEditor (0.2.8), gcodeleveling (0.1.0), gcodeleveling (0.1.1)
2020-12-16 22:34:23,529 - octoprint.plugin.core - INFO - Loading plugins from /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/plugins, /mnt/dietpi_userdata/octoprint/.octoprint/plugins and installed plugin packages...
2020-12-16 22:34:24,256 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2020-12-16 22:34:24,466 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 34 hook handlers
2020-12-16 22:34:24,595 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-12-16 22:34:24,603 - octoprint.server - INFO - Intermediary server started
2020-12-16 22:34:24,626 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /mnt/dietpi_userdata/octoprint/.octoprint/uploads...
2020-12-16 22:34:24,627 - octoprint.filemanager.storage - INFO - ... file metadata for /mnt/dietpi_userdata/octoprint/.octoprint/uploads initialized successfully.
2020-12-16 22:34:24,630 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2020-12-16 22:34:24,632 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2020-12-16 22:34:24,633 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2020-12-16 22:34:24,634 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2020-12-16 22:34:24,634 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2020-12-16 22:34:24,634 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2020-12-16 22:34:24,635 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2020-12-16 22:34:24,635 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2020-12-16 22:34:24,635 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2020-12-16 22:34:24,635 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2020-12-16 22:34:24,636 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2020-12-16 22:34:24,636 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2020-12-16 22:34:24,636 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2020-12-16 22:34:24,636 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2020-12-16 22:34:25,272 - octoprint.util.pip - INFO - Version of pip is 20.3.3
2020-12-16 22:34:27,466 - octoprint.util.pip - INFO - pip installs to /usr/local/lib/python3.7/dist-packages/ (writable -> no), --user flag needed -> yes, virtual env -> no
2020-12-16 22:34:27,466 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-12-16 22:34:27,542 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2020-12-16 22:34:27,617 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s)
2020-12-16 22:34:27,626 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Action Commands (0.3) = /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint_actioncommands
|  Announcement Plugin (bundled) = /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages/octoprint/plugins/tracking
[... 54 lines cut ...]
2020-12-16 22:35:30,323 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3/dist-packages (writable -> no), --user flag needed -> yes, virtual env -> no
2020-12-16 22:35:30,323 - octoprint.util.pip - INFO - ==> pip ok -> yes
```
</details>
```
root@VM-Buster:~# grep localPipCommand /mnt/dietpi_userdata/octoprint/.octoprint/config.yaml
        localPipCommand: /usr/local/bin/pip3
```
Just verified by the log output on the web UI.
The plugins manager uses the configured custom pip command:
```
/usr/local/bin/pip3 --disable-pip-version-check install file:///tmp/tmptnyquk_n/OctoPrint-TheSpaghettiDetective-master.zip --no-cache-dir --user
```
But the OctoPrint self-update ignores it and uses the default:
```
/usr/bin/python3 -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.5.2.zip --no-cache-dir --user
```

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-16:
Software update is quite a complex system, while there is reference to collecting the `localPipCommand` in the logic I can't currently tell if it is used correctly.

MichaIng, 2020-12-16:
I also wasn't about to 100% track it down, but the [create_pip_caller](https://github.com/OctoPrint/OctoPrint/commit/4110287349b34b435f46349891d2c5ff2f618fe3#diff-384f811674ec8cb767cd660f7c8a4dc18061bd8bd0a45efd692b7b6015906cccR572) function calls, respectively forgotten cases of direct `LocalPipCaller` function calls it what we need to have a look at. I'll see if I find time to further investigate.

foosel, 2021-01-27:
@MichaIng can you check whether you have an override `pipCommand` entry in your `config.yaml` maybe? It could be that the old default got written to `config.yaml` due to a bug, and that overrides the globally configured local pip command. It would be at `plugins.softwareupdate.checks.octoprint.pip_command` if so.

MichaIng, 2021-01-27:
If there was a `pipCommand` setting present (not added manually then), I guess I would have recognised as I tested and reviewed this implementation quite intensively for a while. But I'll verify.

foosel, 2021-01-27:
Sorry, that should have been `pip_command`, not `pipCommand`, fixed above.

And I'm just asking because looking at the code, I currently don't see any other way for this happen, so I'd like to rule it out first before digging deeper.

MichaIng, 2021-01-27:
```
root@VM-Buster:~# cat /mnt/dietpi_userdata/octoprint/.octoprint/config.yaml
api:
    key: BFA7C862088D40ACBFCBD14BDF772C23
plugins:
    announcements:
        _config_version: 1
    discovery:
        upnpUuid: fce32a7e-d6c6-4bd3-b11e-64d19c8bb068
    errortracking:
        unique_id: 50e52cbd-0b3f-4867-8e66-59546bfdbd6f
    gcodeviewer:
        _config_version: 1
    softwareupdate:
        _config_version: 9
    tracking:
        unique_id: 9b750168-a5b5-48ff-966c-d4583f67463b
    virtual_printer:
        _config_version: 1
printerProfiles:
    default: _default
server:
    commands:
        localPipCommand: /mnt/dietpi_userdata/octoprint/pip.sh
        serverRestartCommand: sudo systemctl restart octoprint
        systemRestartCommand: sudo reboot
        systemShutdownCommand: sudo poweroff
    secretKey: ACLPDjXz5VlSQhffKIO85wlwXCUmQQze
2021-01-27 16:39:03 root@VM-Buster:~# cat /mnt/dietpi_userdata/octoprint/pip.sh
#!/bin/dash
echo 'DEBUG: localPipCommand is used!'
pip3 "$@"
```
Initial pip command check on startup is fine (the additional test output messes with version check, whoops 😅):
```
Jan 27 16:36:26 VM-Buster octoprint[2265]: 2021-01-27 16:36:26,205 - octoprint.util.pip - WARNING - pip command returned unparseable output, can't determine version: DEBUG: localPipCommand is used!
Jan 27 16:36:26 VM-Buster octoprint[2265]: pip 21.0 from /usr/local/lib/python3.7/dist-packages/pip (python 3.7)
Jan 27 16:36:26 VM-Buster octoprint[2265]: 2021-01-27 16:36:26,206 - octoprint.util.pip - INFO - Version of pip is localPipCommand
Jan 27 16:36:27 VM-Buster octoprint[2265]: 2021-01-27 16:36:27,676 - octoprint.util.pip - INFO - pip installs to /usr/local/lib/python3.7/dist-packages/ (writable -> no), --user flag needed -> yes, virtual env -> no
Jan 27 16:36:27 VM-Buster octoprint[2265]: 2021-01-27 16:36:27,676 - octoprint.util.pip - INFO - ==> pip ok -> yes
```
Then I do a software update check, and the default is used:
```
Jan 27 16:44:22 VM-Buster octoprint[2265]: 2021-01-27 16:44:22,478 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip
Jan 27 16:44:22 VM-Buster octoprint[2265]: 2021-01-27 16:44:22,962 - octoprint.util.pip - INFO - Version of pip is 21.0
Jan 27 16:44:22 VM-Buster octoprint[2265]: 2021-01-27 16:44:22,964 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3/dist-packages (writable -> no), --user flag needed -> yes, virtual env -> no
Jan 27 16:44:22 VM-Buster octoprint[2265]: 2021-01-27 16:44:22,964 - octoprint.util.pip - INFO - ==> pip ok -> yes
```
When doing a plugin install, already the web console log shows that the override is used as expected:
```
/mnt/dietpi_userdata/octoprint/pip.sh install file:///tmp/tmpn3a_uaii/OctoPrint-TheSpaghettiDetective-master.zip --user
DEBUG: localPipCommand is used!
Processing /tmp/tmpn3a_uaii/OctoPrint-TheSpaghettiDetective-master.zip
```

foosel, 2021-01-27:
So that was a fun one because I couldn't reproduce in my dev environment. But when I looked closer at the code I understood why. The `localPipCommand` was only being used if OctoPrint was either subscribed to prereleases or from a dev build because I put the code in the wrong location logic wise 🤦‍♀️

Fixed now, live on `maintenance` and ready for 1.6.0. I could no longer reproduce it once I knew what I was looking for, but it would still be good if you could also give it a whirl, during the RC phase the latest. 

Thanks for the heads-up!

MichaIng, 2021-01-27:
Great, I'll test it ASAP. A bit dangerous to have different code running when using development branches indeed 😄. I guess here it is hard to avoid where stable updates are now pulled from PyPI instead of a Git branch 🤔.

foosel, 2021-01-27:
> A bit dangerous to have different code running when using development branches indeed 😄

Necessary, since unstable versions need slightly different software update presets, and I accidentally only modified those. Prereleases and dev builds still support auto updating, but other than stable releases also need to prompt on branch changes and such to offer a downgrade, not just an upgrade.

MichaIng, 2021-01-31:
Works as expected now, with `/usr/local/bin/pip3` as local pip command:
```
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,587 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,751 - octoprint.plugins.softwareupdate - DEBUG - Looks like we are online
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,940 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 49/60, reset at 2021-01-31 00:05
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,941 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: file_check, local: 2020.8.7, remote: 2020.08.07
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,941 - octoprint.util.pip - DEBUG - Going to figure out pip's version
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,941 - octoprint.util.pip - DEBUG - Using cached pip version information for /usr/local/bin/pip3
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,941 - octoprint.util.pip - DEBUG - Using cached pip setup information for /usr/local/bin/pip3
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,948 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 48/60, reset at 2021-01-31 00:05
Jan 31 00:56:18 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,949 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: firmware_check, local: 2020.9.23, remote: 2020.09.23
Jan 31 00:56:19 VM-Buster octoprint[7561]: 2021-01-31 00:56:18,993 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 50/60, reset at 2021-01-31 00:05
Jan 31 00:56:19 VM-Buster octoprint[7561]: 2021-01-31 00:56:19,002 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: octoprint, local: 1.5.3, remote: 1.5.3
Jan 31 00:56:19 VM-Buster octoprint[7561]: 2021-01-31 00:56:19,018 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
```
When removing the setting, I get:
```
Jan 31 00:59:15 VM-Buster octoprint[7620]: 2021-01-31 00:59:15,299 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
Jan 31 00:59:15 VM-Buster octoprint[7620]: 2021-01-31 00:59:15,746 - octoprint.plugins.softwareupdate - DEBUG - Looks like we are online
Jan 31 00:59:15 VM-Buster octoprint[7620]: 2021-01-31 00:59:15,948 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,018 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 46/60, reset at 2021-01-31 00:05
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,018 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: file_check, local: 2020.8.7, remote: 2020.08.07
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,047 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 45/60, reset at 2021-01-31 00:05
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,048 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: firmware_check, local: 2020.9.23, remote: 2020.09.23
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,064 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 47/60, reset at 2021-01-31 00:05
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,087 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: octoprint, local: 1.5.3, remote: 1.5.3
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,610 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,610 - octoprint.util.pip - DEBUG - Going to figure out pip's version
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,610 - octoprint.util.pip - DEBUG - Using cached pip version information for /usr/bin/python3 -m pip
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,611 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3/dist-packages (writable -> no), --user flag needed -> yes, virtual env -> no
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,611 - octoprint.util.pip - INFO - ==> pip ok -> yes
Jan 31 00:59:16 VM-Buster octoprint[7620]: 2021-01-31 00:59:16,633 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
```
> Necessary, since unstable versions need slightly different software update presets

Understood. great that we/you figured it out now 👍.

commit 58259103 (2021-01-27): 🐛 swu: actually use local pip command

We were only using it for prerelease channels or unstable development
versions, not always if set.

Closes #3892
closed 2021-04-27 by foosel