OctoPrint/OctoPrint #3893: [Request] Gcode viewer to report moves outside build volume

ProfEngr, 2020-12-17:
**Is your feature request related to a problem? Please describe.**
Sort of. I lost a 16hr print, only 30% in, when a travel move was requested that went beyond the build volume of the printer. The slicer should have flagged the problem first, but didn't.

**Describe the solution you'd like**
Would it be possible to warn the user before the job starts that it has moves outside the bed size defined in Octoprint?

**Describe alternatives you've considered**
I've submitted a bug report to the slicer developer, but this would be a good sanity check in case the user sends the file to the wrong printer, e.g. a prusa mini when the file needs the volume of a CR-10. 
The analysis shows the part dimensions, but apparently does not include the travel moves.

**Additional context**
There were 4 layers that broke the boundary for this print. Here's the first. It was 0.35mm too far in X. *sadface* 
I could set the firmware bed size 5mm larger in X, but I don't want to. Even though the head can safely move the distance, the nozzle would be off the bed surface.

[image]

cp2004, 2020-12-17:
This is already a feature, if the bed size is set correctly (printer profile) and it is enabled:
[image]

If this did not work, then maybe there is something wrong with it 🤷‍♂️ but in which case you can submit a bug report 🙂. Check you have it enabled, and that your bed size is correctly configured.

ProfEngr, 2020-12-17:
I checked my settings... The model itself does not report as too large for the bed. It's the travel moves from the slicer using *avoid crossing perimeters* that caused the failure.

Should I start a new issue or can this one be changed to a bug report?

[image]

[image]

cp2004, 2020-12-17:
Upload the gcode here, I will take a look and decide. Not sure how the model size detection works but I don't think it'll take me long to find out. Will let you know if it looks like a bug.

ProfEngr, 2020-12-17:
The first offending G1 is on line 138572.

[Chute1-red.zip](https://github.com/OctoPrint/OctoPrint/files/5711021/Chute1-red.zip)

cp2004, 2020-12-17:
Thanks, I dug through the source and it seems model size detection only works for moves *with extrusion*, per:

https://github.com/OctoPrint/OctoPrint/blob/bf8ff6358556aaa073da2d0646f193c3a1fef48f/src/octoprint/util/gcodeInterpreter.py#L389-L393

So this request becomes 'detect movements outside the print volume' in gcode analysis I guess, since it is by design it ignores the travel moves for model size detection. Makes sense for model size, since travel moves at the start/end of a print would always make the size 'bed size' - so this could be decoupled to work out if travel moves are outside the print volume, provide a separate warning.

ProfEngr, 2020-12-17:
What I'll do in the meantime, if I have a print near the extents, is enable travel moves in the preview and/or open the gcode file in Notepad++ and search for values greater than the bed dimension. Maybe the slicer devs will eventually try to account for this on their end too.

foosel, 2022-09-28:
Implemented in #4579 (thanks @kForth!) and now merged, will be in 1.9.0!

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel