OctoPrint/OctoPrint #3894: [Request] API Endpoint for updates

taker218, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
The request is not tied to a problem, just an idea I had.

**Describe the solution you'd like**
I'm currently monitoring my octoprint instances and some other systems in my home network via nagios. I also monitor if there are updates for the OS on my octoprint instances. I just thought it would be cool to have an API endpoint in OctoPrint to check for updates there. I'm not logging in every day to see if there are updates and in the last time everytime I connected to one of my OP instances I had to update some plugins before the print.

**Describe alternatives you've considered**
I don't know if there is an alternative.

**Additional context**
Nothing more to add here.

fieldOfView, 2020-12-18:
Do you mean updates for OctoPrint itself, updates for plugins or system updates?

It could be a bad idea to have instances advertise that they have not had certain security updates.

taker218, 2020-12-18:
Just updates for OctoPrint and the plugins.

The updates for the OS are already taken care of and I don't know if OctoPrint would be able to see if there are OS updates available.

And the endpoint should be with authentication only of course.

cp2004, 2020-12-18:
`http://<your-ip>/plugin/softwareupdate/check` shows all the data you need, I believe - however it is not documented so probably subject to change I guess? Not sure how many of the bundled plugins have API docs.

taker218, 2020-12-18:
@cp2004 oh I didn't know about that, I will have a look at it, but this looks like something I can work with.

Thanks!

closed 2020-12-18 by taker218