OctoPrint/OctoPrint #3896: Gcode from PrusaSlicer 2.2.0 always cause the model size detector to claim that "Object doesn't fit print volume"

antoneliasson, 2020-12-18:
#### What were you doing?

1. Set up a printer with build volume 250x210x210 mm in OctoPrint 1.5.2.
2. Slice 3DHubs Marvin (https://www.thingiverse.com/thing:215703) in PrusaSlicer 2.2.0 (in Linux) for Prusa i3 MK3S with the built-in profile 0.30mm DRAFT. Resulting Gcode:
[20140205_Marvin_KeyChain.1._0.3mm_PLA_MK3S_24m.zip](https://github.com/OctoPrint/OctoPrint/files/5717126/20140205_Marvin_KeyChain.1._0.3mm_PLA_MK3S_24m.zip)
.
3. Upload it to OctoPrint.
4. Wait for analysis to finish (I checked journalctl output)
5. Select it for printing.

#### What did you expect to happen?

The model should pass inspection as it is a tiny piece centered on the build plate.

#### What happened instead?

~~~
Object doesn't fit print volume

Object in 20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode exceeds the print volume of the currently selected printer profile, be careful when printing this.
More

You can disable this check via Settings > Features > "Enable model size detection [...]"
~~~

journalctl shows:

~~~
Dec 18 17:38:13 raspantone octoprint[2405]: 2020-12-18 17:38:13,953 - octoprint.filemanager.analysis - INFO - Starting analysis of local:20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode
Dec 18 17:38:13 raspantone octoprint[2405]: 2020-12-18 17:38:13,953 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/OctoPrint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode
Dec 18 17:38:13 raspantone octoprint[2405]: grep: Invalid content of \{\}
Dec 18 17:38:16 raspantone octoprint[2405]: 2020-12-18 17:38:16,850 - octoprint.filemanager.analysis - INFO - Analysis of entry local:20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode finished, needed 2.90s
Dec 18 17:38:21 raspantone octoprint[2405]: 2020-12-18 17:38:21,128 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: 20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode, owner: antone, user: antone
~~~
Note that the output from grep is not in octoprint.log.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.5.2.

#### Operating System running OctoPrint

Raspbian 10 (32-bit). I've also seen it in OctoPi 0.18 RC1 but this report is for Octoprint 1.5.2 running on plain Raspbian 10.

#### Printer model & used firmware incl. version

Prusa i3 MK3S.

#### Browser and version of browser, operating system running browser

Firefox 83.0, Debian 9.

#### Link to octoprint.log

https://gist.github.com/antoneliasson/1d0e7b0619c64e99e8dbae35e13d4394

#### Link to contents of terminal tab or serial.log

I don't think this is applicable as no print has been started.

#### Link to contents of Javascript console in the browser

~~~
Some cookies are misusing the recommended “SameSite“ attribute 10
Cookie “session_P5000” will be soon treated as cross-site cookie against “https://raspantone/api/files/local” because the scheme does not match. local
Cookie “session_P5000” will be soon treated as cross-site cookie against “https://raspantone/api/files?recursive=true” because the scheme does not match. files
Cookie “session_P5000” will be soon treated as cross-site cookie against “https://raspantone/api/files?recursive=true” because the scheme does not match. files
Cookie “session_P5000” will be soon treated as cross-site cookie against “https://raspantone/api/files/local/20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode” because the scheme does not match. 20140205_Marvin_KeyChain_0.3mm_PLA_MK3S_24m.gcode
Cookie “session_P5000” will be soon treated as cross-site cookie against “https://raspantone/api/files?recursive=true” because the scheme does not match. files
~~~

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

Harry-58, 2020-12-22:
You have to set a Bounding Box in Octoprint printer profile.
Look hier: [https://help.prusa3d.com/en/article/octoprint-configuration-and-install_2182](url)
[image]

cp2004, 2020-12-24:
@antoneliasson just looked at your gcode, there is a movement outside the bed area (-3) at the start of the gcode. Please confirm if following @Harry-58's suggestion fixes the problem.

antoneliasson, 2020-12-24:
Thank you, that is probably it. I think that is the nozzle priming routine. I don't have access to that particular OctoPrint instance at the moment but I will set up another one in the next few days and try to increase the bounding box slightly into the negative Y axis.

antoneliasson, 2020-12-25:
Tried setting a custom bounding box with Ymin = -4 and the warning is gone. Thank you for your help!

closed 2020-12-25 by antoneliasson