OctoPrint/OctoPrint #3900: [Request] Minimum Infill Layers

KenLucke, 2020-12-21:
**Is your feature request related to a problem? Please describe.**
I have had several models of late that end up generating one layer (sometimes even a thin 0.1mm layer) of infill.  This causes the next layer to be a slow "bottom" layer.  It actually wastes more time (it seems) to have that happen than to just print a normal layer, and he amount of filament saved, in most cases, is minuscule.

**Describe the solution you'd like**
A setting under infill that would require a minimum thickness (defaulting to 2*layer height or something) before the program would actually switch to considering the layer to be infill.

**Describe alternatives you've considered**
Support blocker in the layer set to 100% infill, which is a PITA to do and get positioned correctly.

**Additional context**
Add any other context or screenshots about the feature request here.

cp2004, 2020-12-21:
@KenLucke is this relevant to OctoPrint? Sounds more like a slicer problem to me, than something OctoPrint is handling.

KenLucke, 2020-12-21:
Oops, sorry, posted in wrong forum!  Damn.

closed 2020-12-21 by KenLucke