OctoPrint/OctoPrint #3902: Can not connect to my Ender 3 Pro since the last Update

xenon2008, 2020-12-24:
#### What were you doing?
If i want to connect through Octoprint with my Ender 3 Pro i alaways get an Error with:
Error: No more candidates to test, and no working port/baudrate combination detected

I changed nothing on my Printer and nothing on or in Octoprint.
I only made a Update to the latest Version from Octoprint for 2-3 days..

And since these 2-3 Days i have this troubles.
For Example...

Yesterday i get this Error:"Error: No more candidates to test, and no working port/baudrate combination detected"

So i changed the Port from the USB cable on my RPI, i rebootet octoprint, i tried the FailSafe mode... all but any of this works.
Then i unpluged the Power from the Rpi and plugged it back on power again.
I booted the RPI up and than i could connect to my printer as usable.

Today i get the same error (since yesterday.)... Nothing Works, either i unplugged power from the pi and startet it again.
Now i can connect to my Ender 3 Printer, but this is not normal

Sorry hope everybody can unterstand me with my bad english Skills..

Kind Regards

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
A normal Connection between Octoprint and my Ender 3 Pro with SKR 1.3
#### What happened instead?
Get the Error:Error: No more candidates to test, and no working port/baudrate combination detected

#### Did the same happen when running OctoPrint in safe mode?
Yes!

#### Version of OctoPrint

Version 1.5.2

#### Operating System running OctoPrint

Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2

#### Printer model & used firmware incl. version

Ender 3 Pro with SKR 1.3

#### Browser and version of browser, operating system running browser

Google Chrome & Firefox

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5739268/octoprint.log)

#### Link to contents of terminal tab or serial.log

[serial.log](https://github.com/OctoPrint/OctoPrint/files/5739271/serial.log)

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

cp2004, 2020-12-27:
I don't think this is a bug in OctoPrint. Your printer is not being detected by the underlying OS. There is nothing OctoPrint can do to make the printer become visible.
> `octoprint.util.comm - INFO - Serial detection: Performing autodetection with 0 port/baudrate candidates: `

Try different cables or check your board/cable for damage. This issue tracker is for bugs with OctoPrint, for help with other issues (such as this one) please use the [community forums](https://community.octoprint.org) or the [discord server](https://discord.octoprint.org). See also the [contribution guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#issues-tickets-however-you-may-call-them). Thank you.

closed 2020-12-27 by cp2004

ffa95, 2020-12-27:
> #### What were you doing?
> If i want to connect through Octoprint with my Ender 3 Pro i alaways get an Error with:
> Error: No more candidates to test, and no working port/baudrate combination [...]

i have the same problem with my ender 3 pro since 1 month...

marioparanhos, 2021-01-24:
I'm having the exact same problem with my Ender 3 for the last month or so. I know the problem is not OctoPrint. The printer is just not enumerating via usb. It worked reliably for over a year with the exact same cables (5v covered on the USB plug)  and power supply. Now it only works every once in while. 

I've tried different cables and power supplies. Nothing seems to work. But after enough reboots and power cycles I might eventually get lucky and it connects. When it does, it prints reliably until I turn it off again. 

When this first happened, it only took one power cycle to get it to connect. Now it might take dozens. I now have been trying to connect for 2 days with no luck.

I have another OctoPrint next to it for another printer. The Ender doesn't enumerates on that Pi either. The other printer enumerates fine on the Ender's Pi.

All signs point to something going bad on the Ender. Any suggestions?

DaSTIG, 2021-02-08:
Im also having this issue.... I upgraded both my Rpis to the latest Octopi and octoprint.... (havent used them in a week or so) and today, i am unable to connect to either machine. (hypercube & i3 machines) 
sounds like something is up on the latest images.

foosel, 2021-02-09:
And this is why the first step of debugging any kind of issue with your OctoPrint setup should always be running in safe mode.

marioparanhos, 2021-02-10:
I found a way to reliably get my Ender 3 to enumerate again on the Pi. 

Before I used a cable with the 5v disabled to avoid powering the Ender from the Pi. This worked well for a year. Then connection started getting difficult requiring multiple reboots. Now it won't connect at all unless I follow this exact procedure. 

Start with Ender 3 and Pi both off and a regular USB cable between the two (including power pin). 

Power up Pi and let it boot. The Pi will power the Ender board and enumerate as /dev/ttyUSB.

Open up OctoPrint and connect to the printer.

Now power up the printer before doing anything else.

Print away!

This process works 100% of the time for me now. I don't know why and I don't like it, but it works and I can print again.