OctoPrint/OctoPrint #3905: Constant automatic toggling on/off of Terminal Fancy Mode causing the UI to slow down

StefanJanssen95, 2020-12-25:
#### What were you doing
My printer was printing a small model (20 minute print) when I noticed a slow interface. I was looking at a screen which didn't have a terminal (I rarely use the terminal).

#### What did you expect to happen?
I expected the UI to be quick and snappy.

#### What happened instead?
When I looked in my browser console it had it's logs full with
> Terminal: Detected slow client, disabling fancy terminal functionality
> Terminal: Client speed recovered, enabling fancy terminal functionality
[image]
As can be seen in the screenshot with timestamps it was toggling "fancy mode" every 1-5 seconds. The constant toggling fancy mode on and off was causing the slow UI.
When I activated the Terminal view I saw it was constantly toggling the "fancy mode" as well. There was a function to force fancy mode which made the UI more responsive right away. I think the processing should happen on average processing times over 5-10 seconds to avoid constant toggling. 

#### Did the same happen when running OctoPrint in safe mode?
I've not tested it in Safe Mode, but I encountered the issue in both a fresh install of OctoPi (without plugins) and an older installation of OctoPi (with plugins). Currently my printer is not assembled (I'm mid upgrade), but if necessary I will perform a test in Safe Mode as well.
#### Version of OctoPrint
OctoPrint 1.5.2
Python 2.7.13

#### Operating System running OctoPrint
OctoPi 0.15.1 running on a Raspberry Pi 3 Model B

#### Printer model & used firmware incl. version
Creality CR10S Pro

#### Browser and version of browser, operating system running browser
MS Edge Version 87.0.664.66 (64-bits)
Windows 10 Pro 2004 19041.685

#### Link to octoprint.log
https://pastebin.com/4qtZPDJ6

#### Link to contents of terminal tab or serial.log
https://pastebin.com/xBXWkEyd

#### Link to contents of Javascript console in the browser
See above

cp2004, 2020-12-29:
Since the only line missing from this bug report is the 'I have read the FAQ', I'll give you a one-time pass through the bot. It gets upset if you delete that from the template.

cp2004, 2021-01-12:
Just documenting my thoughts with this one, since I've spent quite a bit of time over the last weekend reading the terminal view model.

From what I can see, the fancy terminal and the regular terminal are both in the DOM at the same time, just one of them is visible and the other is not (and not updated). When the UI needs to toggle from fancy to unfancy, (I think) it has to shift all of the log lines into the non-fancy version (or the other way). I haven't looked at this in depth, as I wasn't implementing it in my React experiment (trying to recreate the terminal). I haven't done much looking, but I can see that the toggling of fancy could potentially have a bit of a performance blow. No numbers to put on it, just the *feeling* I got from the code.

foosel, 2021-01-27:
So that might have been down to a simple logic bug, patch see above.

The thing is, it should already only have reenabled the fancy terminal if log push processing times stayed beneath half the threshold for 5s. That is achieved by starting a timer to switch things back. Said timer should have been cleared if log processing took too long, but the logic there was flawed and it never was cleared. Hence the constant back and forth.

The above commit should fix this, however I wasn't able to reliably trigger a slow log so it would be great if y'all could help test that, during the 1.6.0 RC phase the latest.

arueth, 2021-02-08:
Can I just copy the patched file to in /home/pi/oprint/lib/python3.7/site-packages/octoprint/static/js/app/viewmodels/terminal.js or do I also need to repackage?

foosel, 2021-02-08:
I think copying should work. Make a backup just in case.

arueth, 2021-02-08:
Thanks, I did and it seems to have fixed the continuous messages in the console. I'll let it run for a while and see if there is any reoccurrences.

commit de81d91a (2021-01-27): 🐛 Fix fancy/unfancy terminal switching

The switch back was enqueued in a timeout, however that timeout was only
ever cleared if fancy functionality was already enabled when trying to
switch it back on due to a logic error. Now it gets cleared whenever the
time needed for processing the log lines is above the threshold.

Potentially fixes #3905
closed 2021-04-27 by foosel