OctoPrint/OctoPrint #3908: [Request] Separate file views per user

Jnesselr, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
We run OctoPrint and my hackerspace and each member has their own user account. There's a bit of a distaste that every user can just access anyone else's files. Some people make folders for themselves, but others just throw their files on the root directory which clutters everything up. 

**Describe the solution you'd like**
Instead of a global file list, a file list per user. The first time you log in, there should be just a blank list of files.

**Describe alternatives you've considered**
I haven't been able to think of any alternatives that solve the root of the problem which is that user A can access user B's files. The "make a folder to throw your stuff in" helps with organization, but new users keep forgetting to do it.

**Additional context**
I've talked to the main dev about this before, and I couldn't find a quick way to implement it. Still, I figure it's probably good to have it tracked here in the event that someone else can figure it out.

jneilliii, 2020-12-26:
This makes total sense, especially with the requirement of access control and the ability to have granular permissions now.

I believe it will take quite a lot of changes throughout OctoPrint. Looking at the currently stored metadata provided by the files API the addition of a `user` property would be required to be able to filter the file list. The upload/file add API would have to be updated to populate that property. I'm sure there are other places too.

cp2004, 2020-12-27:
This could probably be done quite easily, with minor changes to the API. The API shouldn't return files that aren't owned by the user, so all of this can be done server side. As for uploads, the API key supplied with the upload would determine the user. 
These are the challenges I came up with, up for discussion:

* Who takes ownership of existing files on the server, when they update?
* Can an admin user see all the files? I think yes for this one.

If admins can see all the files, then a `user` field can be added to the GET side of the files API, so that this is visible in the UI. For non-admins, this field would always be the current user, so it always exists to make reading things easier.

mfraser, 2023-06-30:
Did anything ever come of this idea? Thought it would be useful so that other users couldn't accidentally delete any of your files.
Not sure how it would work with plugins such as Obiko.