OctoPrint/OctoPrint #3910: Printing stops with Octoprint shutdown

annovialb, 2020-12-26:
#### What were you doing?
I was running a printing job.
It did stop midway. No more commands were appearing in the terminal tab. 
This happened 3 times over the course of 1 month. This last time, I went checking the log files. More info below.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
The print to complete.

#### What happened instead?
Printing stopped. The printer stopped receiving commands from octoprint.

From the log file "OSError: [Errno 101] Network is unreachable" (see below for the link to it), this seems related to a network problem. I would hope printing would continue also in case the network is temporarily lost.

#### Did the same happen when running OctoPrint in safe mode?
Since this is a rare problem. I did not try to reproduce it in safe mode.

#### Version of OctoPrint
OctoPrint 1.5.2Python 3.7.3

#### Operating System running OctoPrint
Raspbian GNU/Linux 10 (buster) 10.6
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

#### Printer model & used firmware incl. version
Ender 5 Pro - Marlin FW

#### Browser and version of browser, operating system running browser
MacOS Chrome

#### Link to octoprint.log
https://www.dropbox.com/s/3gxwfjbubzvmara/octoprint.log?dl=0

The problem happened at "2020-12-26 23:51:21,093" in the log file

#### Link to contents of terminal tab or serial.log
https://www.dropbox.com/s/sd9xtomoqxb6nwx/Screen%20Shot%202020-12-27%20at%2012.04.25%20AM.png?dl=0

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-12-26:
This doesn't appear to be either related to the network, the OSError you have quoted happens after event loop is already shutdown.

Please try with the timelapse function disabled, or at least fix the settings. It's creating a lot of errors because the image can't be captured from the requested URL.

OctoPrint doesn't shut down when it can't connect to the internet, that is likely some external factor causing the issue here.

tedder, 2020-12-27:
This is on a Pi? It's interesting that `vcgencmd` isn't found. I thought it was included on all Raspbian distros. And you must have a bespoke installation (e.g., not using OctoPi).

It's clear the Pi is getting a shutdown command but not *why*. It appears to be coming external to Octoprint. Feel free to share logs, but this discussion should be on [community.octoprint.org](https://community.octoprint.org/latest) until things are sorted out.

annovialb, 2020-12-27:
Hello,
  thanks for the feedback.
Indeed I found that the rPi did reboot itself at that time, so it is
definitely external to octoprint.
I will investigate why it did reboot.

Cheers

cp2004, 2020-12-27:
Since this is not OctoPrint related, I'll close it. Feel free to request a reopen should you find an issue definitely with OctoPrint.

closed 2020-12-27 by cp2004